MarlinFirmware/Marlin #11091: Auto Bed Leveling not compensating enough

FelixHauser, 2018-06-23:
Auto Bed Leveling not compensating enough

After bed leveling is set up on configuration.h some parts of the bed are not enough compensated for and, the nozzle is hovering the bed at too high distance. The first layer on some parts of the print are not reliable. More specifically the X0Y0 and XmaxYmax corners are well printed and the other two corners are faulty.

I have a a well leveled bed to begin with. As level I I could get it and without springs but fixed. The distance from the bottom of both Z-Carriages is exactly the same, measured with a caliper. I am using a capacitive probe with the correct offsets. I am not using eeprom on any of my printers, so I set everything in the start G-Code of each print. I can see the Z-Motors moving very little as the layer goes on, so I guess that I set everything correctly on the configuration.h file and the ABL is working.

A part from swapping the probe for a new one, I don't know what else to do.

The start G-Code I use in each print is:

```
G28 ; home all axes
G29; Auto Bed Leveling
G1 Z5 F5000; Lift Nozzle
```

I expected for Auto Bed Leveling to be able to compensate for these irregularities on the print bed.

Thanks in advance

[configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/2129995/configuration.files.zip)

thinkyhead, 2018-06-30:
LINEAR bed leveling is unsuitable for a non-flat bed. Try BILINEAR instead.

And, be sure to test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch so we're all on the same page.

FelixHauser, 2018-07-04:
Thanks @thinkyhead I tried with the version of Marlin I have (1.1.8) and the difference with the tests is significant. Much better. I'm using a mirror as a build plate and I was assuming it was perfectly flat...

closed 2018-07-04 by FelixHauser