MarlinFirmware/Marlin #11092: Bad temperature reading LINEAR_ADVANCE

margaale, 2018-06-23:
With latest as bugfix 1.1.x as now(#11084) I'm seeing bad temperature readings about the nozzle. I'm using PID Temp in the nozzles and the bed too. There seems to be high reading that is fooling marlin so its let the temperature so low aborting the printing with a thermal runaway error. As soon as i set K to 0.00 everythings starts working as usual.

margaale, 2018-06-23:
Ramps 1.4.
This is after compiling marlin; 
Sketch uses 220074 bytes (86%) of program storage space. Maximum is 253952 bytes.
Global variables use 5860 bytes (71%) of dynamic memory, leaving 2332 bytes for local variables. Maximum is 8192 bytes.

gloomyandy, 2018-06-23:
That looks like more free memory than the folks that had the other problem, so I guess that it is not that causing your issues.

margaale, 2018-06-23:
It seems that loweing Extruder Acceleration and retraction acceleration helps a little.

Sineos, 2018-06-24:
Duplicate to #10900

thinkyhead, 2018-06-30:
It may be more that the Stepper ISR eats up so much time that the Temperature ISR is getting shorted. One workaround may be to reduce the number of micro-steps on your axes to free up the processor a bit more.

margaale, 2018-07-29:
Just to let you know. After latest bugfix (now) I get no bad readings. I guess that the pr that ensures ADC conversion has everything to do with this.

closed 2018-07-29 by thinkyhead