MarlinFirmware/Marlin #11093: Bad forward extruder moves between retractions

VanessaE, 2018-06-23:
Normal forward extruder moves are being interrupted, or something :smiley:, particularly the first move or two after a retract -- including the corresponding unretract move.  Seems to be a conflict between linear advance, retraction moves, and Slic3r's wipe function.

Listening very closely, I can hear a brief, quiet extruder whine with each failed forward move -- not a typical stall in the middle of a move, more like "tried to go way too fast for the whole move".

The result is the extruder doing more successful backward moves than forward moves, in retract-heavy parts of the print anyway.

`git bisect` process:

```
vanessa@rainbird:~/RepRap/Marlin-bugfix-2.0.x$ git bisect start 
Already on 'bugfix-2.0.x'
Your branch is up-to-date with 'origin/bugfix-2.0.x'.
vanessa@rainbird:~/RepRap/Marlin-bugfix-2.0.x$ git bisect good 2ba198
vanessa@rainbird:~/RepRap/Marlin-bugfix-2.0.x$ git bisect  bad 48a15d1c7e78eaa716bedecb80feeae780c7234a
Bisecting: 23 revisions left to test after this (roughly 5 steps)
[091f74243240992be5f6e2edf7e63a585245e183] Add WIFISUPPORT to example configs
vanessa@rainbird:~/RepRap/Marlin-bugfix-2.0.x$  pushd .; cp ../Configuration* Marlin/; cd /home/vanessa/RepRap/arduino-1.8.5/; /home/vanessa/RepRap/arduino-1.8.5/arduino; popd; git stash

* click the "(➡)" compile/upload button, then close the IDE after it's done

* connect with Pronterface, heat up, print, then stop/disconnect after I get a reliable result

* `git bisect bad` or `good` accordingly, and repeat from `pushd` above
```

The result:

```
vanessa@rainbird:~/RepRap/Marlin-bugfix-2.0.x$ git bisect gooda215725df6a00aa61c5d5a8c5e2d109810f082c9 is the first bad commit
commit a215725df6a00aa61c5d5a8c5e2d109810f082c9
Author: etagle <ejtagle@hotmail.com>
Date:   Tue Jun 12 01:04:26 2018 -0300

    Fix stepper pulse timing
    
    Always honor minimum period on stepper pulse generation, and fix timing calculations
    
    Signed-off-by: etagle <ejtagle@hotmail.com>

:040000 040000 a1a3b76e8e49bdd312c7ca8c2df46badd68c4e23 2599204319092a4343f78ac1d71243c56cb495cf M	Marlin
```

gloomyandy, 2018-06-24:
Are you using JUNCTION_DEVIATION? Does turning that off make any difference? I ask because from this thread https://github.com/MarlinFirmware/Marlin/issues/9917 there are still problems with the calculation of ejerk values. Posting your configuration may help others to investigate further.

VanessaE, 2018-06-24:
No, I have that disabled.

<details><summary>Configs, as a diff</summary>

```diff
commit 7f6dc969078a118d7475ab17a695b28fe7a61168
Author: Vanessa Dannenberg <vanessa.e.dannenberg@gmail.com>
Date:   Sun Jun 24 02:53:47 2018 -0400

    My Settings

diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index 8586c4e54..2da18e54d 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -74,7 +74,7 @@
 // User-specified version info of this build to display in [Pronterface, etc] terminal window during
 // startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
 // build by the user have been successfully uploaded into firmware.
-#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
+#define STRING_CONFIG_H_AUTHOR "bugfix-2.0.x 0d79128c" // Who made the changes.
 #define SHOW_BOOTSCREEN
 #define STRING_SPLASH_LINE1 SHORT_BUILD_VERSION // will be shown during bootup in line 1
 #define STRING_SPLASH_LINE2 WEBSITE_URL         // will be shown during bootup in line 2
@@ -125,7 +125,7 @@
  *
  * :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000]
  */
-#define BAUDRATE 250000
+#define BAUDRATE 115200
 
 // Enable the Bluetooth serial interface on AT90USB devices
 //#define BLUETOOTH
@@ -151,7 +151,7 @@
 #define EXTRUDERS 1
 
 // Generally expected filament diameter (1.75, 2.85, 3.0, ...). Used for Volumetric, Filament Width Sensor, etc.
-#define DEFAULT_NOMINAL_FILAMENT_DIA 3.0
+#define DEFAULT_NOMINAL_FILAMENT_DIA 1.75
 
 // For Cyclops or any "multi-extruder" that shares a single nozzle.
 //#define SINGLENOZZLE
@@ -311,12 +311,12 @@
  *
  * :{ '0': "Not used", '1':"100k / 4.7k - EPCOS", '2':"200k / 4.7k - ATC Semitec 204GT-2", '3':"Mendel-parts / 4.7k", '4':"10k !! do not use for a hotend. Bad resolution at high temp. !!", '5':"100K / 4.7k - ATC Semitec 104GT-2 (Used in ParCan & J-Head)", '6':"100k / 4.7k EPCOS - Not as accurate as Table 1", '7':"100k / 4.7k Honeywell 135-104LAG-J01", '8':"100k / 4.7k 0603 SMD Vishay NTCS0603E3104FXT", '9':"100k / 4.7k GE Sensing AL03006-58.2K-97-G1", '10':"100k / 4.7k RS 198-961", '11':"100k / 4.7k beta 3950 1%", '12':"100k / 4.7k 0603 SMD Vishay NTCS0603E3104FXT (calibrated for Makibox hot bed)", '13':"100k Hisens 3950  1% up to 300°C for hotend 'Simple ONE ' & hotend 'All In ONE'", '20':"PT100 (Ultimainboard V2.x)", '51':"100k / 1k - EPCOS", '52':"200k / 1k - ATC Semitec 204GT-2", '55':"100k / 1k - ATC Semitec 104GT-2 (Used in ParCan & J-Head)", '60':"100k Maker's Tool Works Kapton Bed Thermistor beta=3950", '66':"Dyze Design 4.7M High Temperature thermistor", '70':"the 100K thermistor found in the bq Hephestos 2", '71':"100k / 4.7k Honeywell 135-104LAF-J01", '147':"Pt100 / 4.7k", '1047':"Pt1000 / 4.7k", '110':"Pt100 / 1k (non-standard)", '1010':"Pt1000 / 1k (non standard)", '-4':"Thermocouple + AD8495", '-3':"Thermocouple + MAX31855 (only for sensor 0)", '-2':"Thermocouple + MAX6675 (only for sensor 0)", '-1':"Thermocouple + AD595",'998':"Dummy 1", '999':"Dummy 2" }
  */
-#define TEMP_SENSOR_0 1
+#define TEMP_SENSOR_0 7
 #define TEMP_SENSOR_1 0
 #define TEMP_SENSOR_2 0
 #define TEMP_SENSOR_3 0
 #define TEMP_SENSOR_4 0
-#define TEMP_SENSOR_BED 0
+#define TEMP_SENSOR_BED 13
 #define TEMP_SENSOR_CHAMBER 0
 
 // Dummy thermistor constant temperature readings, for use with 998 and 999
@@ -351,11 +351,11 @@
 // When temperature exceeds max temp, your heater will be switched off.
 // This feature exists to protect your hotend from overheating accidentally, but *NOT* from thermistor short/failure!
 // You should use MINTEMP for thermistor short/failure protection.
-#define HEATER_0_MAXTEMP 275
-#define HEATER_1_MAXTEMP 275
-#define HEATER_2_MAXTEMP 275
-#define HEATER_3_MAXTEMP 275
-#define HEATER_4_MAXTEMP 275
+#define HEATER_0_MAXTEMP 255
+#define HEATER_1_MAXTEMP 255
+#define HEATER_2_MAXTEMP 255
+#define HEATER_3_MAXTEMP 255
[... 465 lines cut ...]
 
-  #define MAX7219_DEBUG_STEPPER_QUEUE 0  // Show the current stepper queue depth on this and the next LED matrix row
+  #define MAX7219_DEBUG_PLANNER_QUEUE 0  // Show the current planner queue depth on this and the next LED matrix row
                                          // If you experience stuttering, reboots, etc. this option can reveal how
                                          // tweaks made to the configuration are affecting the printer in real-time.
 #endif
@@ -1700,4 +1701,13 @@
                               // Default behaviour is limited to Z axis only.
 #endif
 
+/**
+ * WiFi Support (Espressif ESP32 WiFi)
+ */
+//#define WIFISUPPORT
+#if ENABLED(WIFISUPPORT)
+  #define WIFI_SSID "Wifi SSID"
+  #define WIFI_PWD  "Wifi Password"
+#endif
+
 #endif // CONFIGURATION_ADV_H
```
` `
Ignore the stuff after `MAXIMUM_STEPPER_RATE 500000`, I'm not sure why that stuff shows as changed.  Never touched that section.  Probably just the result of copying my configs in from out-of-tree during all the `git bisect` stuff.

</details>

gloomyandy, 2018-06-24:
Things that might be worth trying until someone who really knows this stuff comes along...
* You have a max E acceleration of 10000, which may be too high I'd be tempted to set that to something lower.
* Does lowering the acceleration and retract acceleration help?
* Is the MAXIMUM_STEPPER_RATE figure of 500000 correct? Does lowering it help? 
* There have been a lot of changes in the minimum pulse setting area, so perhaps raising that might help.

I'm sure @ejtagle  will want to take a look at this at some point. What stepper drivers are you using?

gloomyandy, 2018-06-24:
From other discussions of Linear Advance it is clear from the author of that code that the default setting of 10000 is almost certainly not correct for most hardware. When I asked about it on the Junction Deviation thread it was suggested that a value of 2100 was a more realistic starting point. 

My understanding is that the new pulse code is supposed to produce more accurate pulse lengths and frequencies (which are possibly shorter/higher than before), perhaps this means that settings like acceleration, top speed and jerk now need to be more accurately configured?

On other threads related to lost steps with the new pulse code one of the things that @ejtagle  has asked users to try is to adjust the minimum pulse settings upwards (and the max stepper rate downwards) as the current code calculates things differently. There was also a correction to the pulse generation code made two days ago:
https://github.com/MarlinFirmware/Marlin/commit/48a15d1c7e78eaa716bedecb80feeae780c7234a#diff-b1e3f1b10558445963aa5bbf4d20249c
not sure if your tests included that or not?

It may be that none of the above makes any difference, but if you have the time then having the extra data points will probably help determine what the underlying problem is.

VanessaE, 2018-06-24:
> perhaps this means that settings like acceleration, top speed and jerk now need to be more accurately configured?

I don't see how that should matter.  All these step timing changes were supposed to be about jitter, resolution, and the like, if I understand things correctly, and not absolute number of steps per second, so acceleration and peak speeds should be unaffected.

Besides, it only affects the first extruder move or two, after a retract (well, that's as accurate as I can be, it moves too fast for me to be 100% sure), when the stepper has effectively no load, thus extruder acceleration should be able to go much higher than 2000 mm/s².

> not sure if your tests included that or not?

I bisected from 2ba198a624d0bf0b00b044ef0d26514ab54d2a3f to 48a15d1c7e78eaa716bedecb80feeae780c7234a, so yeah, that commit was included.

gloomyandy, 2018-06-24:
This pull request may be of interest:
https://github.com/MarlinFirmware/Marlin/pull/11098
In particular this change:
https://github.com/MarlinFirmware/Marlin/pull/11098/commits/8b1f494b1918ed66056435c2007f0885c1d941c6

which will introduce extra delays for the A4988 if configured. 

What exactly is the visible result of the problem you are seeing? I've been running tests with the latest code and have not seen anything obvious, but I may have missed it.  But I'm probably not running as fast as you especially things like the max Z and E feedrates.

VanessaE, 2018-06-24:
>What exactly is the visible result of the problem you are seeing? 

As I wrote in my opening post, "[t]he result is the extruder doing more successful backward moves than forward moves, in retract-heavy parts of the print anyway."

...which leads to basically zero nozzle pressure.

> But I'm probably not running as fast as you especially things like the max Z and E feedrates.

Z at 30 mm/s and E at 60 mm/s is not fast at all.

VanessaE, 2018-06-25:
This bug report is invalid -- I found the problem.  Somehow its current got turned down from the usual 625mA to a weakling 300 mA.  Probably when I took my E modules off to change their microstep settings.  I must have swapped the spare, which I know was turned down, with the active one.

Not sure why it took until now for this to show up, BUT they've only been in this state for like, a week, so I wonder if I've run into a variant of the "skips steps" issue?  Shouldn't 300 mA on a 5.222:1 geared extruder be enough (mind you, I'm perfectly happy to keep it where it is).

closed 2018-06-25 by VanessaE