MarlinFirmware/Marlin #11094: UBL bed levelling with dual extruders

autonumous, 2018-06-23:
I’ve never really finally tuned my UBL mesh, I thought i would try today

Should UBL bed levelling work with a dual extruder setup? Should a G29 P1/P2 only be performed against Extruder 0 or should it work with either ?

I would assume that if you planned to use Extruder 2 to perform manual probing, you would select extrude 2 (T1)  prior to performing the operation. 

I have a dual extruder, but extruder 1 (T0) is currently out of operation.  I intend to perform the following steps (based on http://marlinfw.org/docs/features/unified_bed_leveling.html):

```
G28
T1
G28
G29 P1
G29 P2
G29 P3
...
```

However during the automatic probing (`G29 P1`) my probe moves beyond the bed and thus the probe fails to trigger, the extruder hits the bed causing a probe failure.

It works if I use Extruder 1 (T0), but I would assume I need to use that nozzle when performing a Manual Probe (G28 P2) ?  Which i can’t as i currently have no Nozzle in the extruder.

my multi extruder head consists of
**IR probe**<--25mm-->**Extruder1** <--38mm-->**Extruder2**

**bed setup**
```
X_BED_SIZE 300
Y_BED_SIZE 200
```

**travel Limits**
```
X_MIN_POS -38
Y_MIN_POS 0
Z_MIN_POS 0
X_MAX_POS 300 
Y_MAX_POS 240 
Z_MAX_POS 300
```

**probe offsets**  
```
X_PROBE_OFFSET_FROM_EXTRUDER -25
Y_PROBE_OFFSET_FROM_EXTRUDER 0
```

**Extruder 1 offsets**
```
HOTEND_OFFSET_X {0.0, 38.0} 
HOTEND_OFFSET_Y {0.0, 0.0}
```

boelle, 2019-02-20:
@autonumous is this still an issue with latest bugfix 2.0?

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead