MarlinFirmware/Marlin #11099: [bugfix-2.0.x] When JUNCTION_DEVIATION is enabled, Ejerk setting value is not displayed.

JXPA, 2018-06-24:
Hello.
E-jerk is not displayed on the line of M205 when M503 is executed with JUNCTION_DEVIATION enabled.

1. Latest Marlin Enable JUNCTION_DEVIATION and flash.
```
#define JUNCTION_DEVIATION
#if ENABLED(JUNCTION_DEVIATION)
  #define JUNCTION_DEVIATION_MM 0.02  // (mm) Distance from real junction edge
  //#define JUNCTION_DEVIATION_INCLUDE_E
#endif
```

2. M502 and M500
3. M503
The setting value of E-jerk is not displayed.
```
14:43:49.757 : echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev> E<max_e_jerk>
14:43:49.761 : echo:  M205 B20000 S0.00 T0.00 J0.02
```

gloomyandy, 2018-06-24:
That is probably correct as e jerk (if it is needed for things like linear advance) is calculated automatically from the junction deviation and E acceleration values.

JXPA, 2018-06-24:
Thank you for your reply.
I forgot to write that I enabled LA1.5.
Does it mean that there is no need to search and set an appropriate E-jerk so that printing does not slow down with LA1.5?

gloomyandy, 2018-06-24:
Good question. Certainly with the current code the ejerk setting is calculated automatically when using a combination of LA and JD. However the ejerk is based upon the JD setting and the E max acceleration setting so you may need to tune that. In most configurations it is set to 10000 which is almost certainly not correct. I asked about this on the JD thread and a value of 2100 was suggested as a good starting point for max E acceleration given a JD value of 0.02 (the default). Note however that again according to that thread that the current calculation for ejerk may not be correct. See:
https://github.com/MarlinFirmware/Marlin/issues/9917
so you may want to wait until that is resolved before spending time trying to tune things.