MarlinFirmware/Marlin #11101: bugfix-2.0.x: Motion control jerky when acceleration is low

nebbian, 2018-06-24:
### Description

When the acceleration is set to low values (I used 100), all motion including travel moves is very jerky.  It appears as if the motion is broken up into segments, and the end of each segment is treated like a stop.

### Steps to Reproduce

1. Configure your marlin for use on a delta (not sure if this is related, but I'm using a delta)
2. In the start GCode, set the acceleration to 100.  eg. M201 X100 Y100 Z100
3. Print the file.

**Expected behavior:** 
The extruder will slow down smoothly at the end of a move, when there is a sharp corner.

**Actual behavior:** [What actually happens]
The extruder moves in a series of small pulses, even when moving in a straight line, and even when travelling.

#### Additional Information

I tried using both bezier curve and jerk, both acceleration schemes provided the same behaviour.

This seems to have been introduced between 8th April 2018, and today, as the previous version of the 2.0 branch didn't exhibit this behaviour.

Configuration is here:
[marlin_configuration_pulsing.zip](https://github.com/MarlinFirmware/Marlin/files/2130568/marlin_configuration_pulsing.zip)

An image showing the issue is here.  Each blob represents where the extruder slows down markedly.
[image]

ejtagle, 2018-06-25:
Could you try PR #11098 and see if it makes a difference ?

nebbian, 2018-06-25:
Hi ejtagle, I can confirm that your branch has fixed this issue, on the exact same file that previously gave issues.  Nice work 👍 
Hopefully your request gets pulled in soon.

thinkyhead, 2018-06-28:
Changes have been merged. Appears to be solving a pile of lingering motion issues.

closed 2018-06-28 by thinkyhead

nebbian, 2018-06-30:
Thanks @thinkyhead, I can confirm that the latest 2.0 branch is working perfectly for me.