MarlinFirmware/Marlin #11102: [FR] at filament runout stop and start inside object

boelle, 2018-06-24:
Not sure if its possible but when filament runs out go to inside the object before lifting nozzle and after new filament has been inserted also start inside the object

my thought behind this is to make it less visible that a change has taken place

GMagician, 2018-06-24:
I think that only slicer could know where "inside of object" is not Marlin that has only "position" to go

GMagician, 2018-06-24:
To inform marlin I think you need some sort of M command with coordinates but what if filament runout after such point? You don't know how big is your layer hence you will not know if you have enough filament after runout. But this is just my opinion

GMagician, 2018-06-24:
> ...lift nozzle

sorry missed

boelle, 2018-06-24:
maybe change my title to: wipe towards point on filament runout?

so if runout happens it will move towards point with special M code and then lift

comps, 2018-06-24:
> i guess the slicer would have to insert a spot on each layer that marlin know is ok to start and stop from
> but how would we do that?

If Marlin had a runtime gcode for `NOZZLE_PARK_POINT`, slicers could set that (on each layer or even more frequently) to a known "safe" point. Whether any slicer actually chooses to implement this, that's another question.

boelle, 2018-06-24:
i will try and pitch the idea to prusa so it maybe could be implemented in their version of Slic3r

GMagician, 2018-06-24:
The only doubt rise in my mind is that G1 commands are read and put into planner queue. Stop will come a little far from real detection point, but I think some margin will be available after runout detection

boelle, 2018-06-24:
yeah, the idea needs some "polishing" i just fired it off before i forgot

same idea at slic3r PE github: https://github.com/prusa3d/Slic3r/issues/1003

ArtieFM, 2018-06-24:
There's a Device that has this function, it's called Resumer 3D

boelle, 2018-06-24:
why another device if it can be done by slicer and firmware?

Roxy-3D, 2018-06-24:
> I think that only slicer could know where "inside of object" is not Marlin that has only "position" to go

Yes...

> To inform marlin I think you need some sort of M command with coordinates but what if filament runout after such point? You don't know how big is your layer hence you will not know if you have enough filament after runout. But this is just my opinion

Most Slic3r's have the concept of 'Special GCode Before Layer Change' and 'Special GCode After Layer Change'.   It would not be difficult to add an option to the filament change code that waited for a special 'Layer Change M (or G) Code'.   And when it got that, it would do the filament change.   And not before.

And at that point, you would just tell it to infill first, and put that special MCode in the 'Special GCode After Layer Change' section.

boelle, 2018-06-24:
Ahh yes, i did not think of that. Just checked that Slic3r PE have "Before Layer Change G-Code" and "After Layer Change G-Code"

My point with it was that Marlin would go to this X, Y spot before it lifted the nozzle up

Problem is that if you have a large object you cant wait much since where would be very limited filament left

Roxy-3D, 2018-06-24:
> Problem is that if you have a large object you cant wait much since where would be very limited filament left

Yes.   But I guess it depends on the placement of your filament sensor.  Mine is several inches away from the nozel's stepper motor...  So I can do a lot of printing before it goes.   Potentially, there could be a 2nd #define that says how much filament is between the sensor and the extruder motor and if a layer change doesn't happen within that much time...   I happens even though the special MCode hasn't happened yet.

boelle, 2018-06-24:
I have my sensor about 100mm before the titan, but everyone will have a different setup
 some might have it butted against the extruder and so only ~20-30mm

Roxy-3D, 2018-06-24:
Well...   Being able to specify the filament length available after the sensor would probably handle this option nicely.   I wonder if anybody wants the feature enough to code it up.    I wouldn't mind doing it, but I don't want to do it for two different code bases.

Roxy-3D, 2018-06-26:
Well....   That would be the purpose of specifying the length of filament.   In my case...  Even with a big print, I can go a full layer before the change is necessary.   But doing it that way lets people decide what to do.   If they want to put the filament sensor right next to the extruder...  That is OK.  But it means you less time until the pause is forced.

mylife4aiurr, 2018-07-03:
This would be much easier for bowden systems because there is often 300 -700mm in the bowden tube after the filament sensor. 
Those on direct feed were there isnt enough time to do the filament change in infill just miss out on new feature but could just do filament change the same way the already do.

Roxy-3D, 2018-07-03:
I'm not sure the bowden systems are better off.   The real question is how much filament length is between the sensor and the drive mechanism.   The long that distance is...   The more printing that can be done without actually running out of filament.

AnHardt, 2018-07-03:
And the more filament is wasted.

Roxy-3D, 2018-07-03:
> I only meant bowden printers should have a much greater chance to do the change in infill because of length of bowden tube to play with.

No...   The whole reason for the bowden tube is so the heavy stepper motor doesn't need to be combined with the nozzle.   Making the hotend assembly lighter makes it easier (and faster) to move it.

The stepper motor driving filament into the bowden tube will be mounted somewhere on the printer's frame.   And immediately before that, will be the filament sensor.   If the filament end gets to the stepper motor, you still have filament all the way down the bowden tube.   But nothing is pushing on it.

mylife4aiurr, 2018-07-04:
Thanks Roxy, I understand your point. Once filament outage is triggered doesnt matter how much filament is in the tube because once its past the extruder motor it cant get pushed to nozzle.

Still hoping for some form of this feature, if not infill change then at worst inner walls???

thinkyhead, 2018-07-04:
Marlin has no concept of "inside of object," or "infill," or "walls." It just sees movement commands from one point to another. Generally-speaking, if you tell the nozzle to "park" in the middle of the bed, most of the time it will be in the middle of the object being printed, though it might not necessarily be touching a solid feature in the current layer.

closed 2018-07-23 by boelle