MarlinFirmware/Marlin #11104: Building Maylan M200 with Marlin 2.0 and PlatformIO

X7JAY7X, 2018-06-24:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

xC0000005, 2018-06-24:
Ok, so to build Marlin for the M200 you need a few more pieces:

From this repo:
https://github.com/xC0000005/M200Toolchain

You get the board .json files (which go in .platformio\platforms\ststm32\boards, and the variant definitions, which go in .platformio\packages\framework-arduinostm32\stm32f1\variants (if you don't see variants under STM32f1, it's probably directly under the arduinostm32 directory).  

Alternately, you can build with STM32GENERIC by doing the following:
Clone from this repo and branch:
https://github.com/xC0000005/Marlin/tree/genericHal
Copy over the M200 configuration files you want to use.
Comment out Serial2 if it's enabled.
In your PlatformIO.ini section for the M200, set platform = https://github.com/xC0000005/platform-ststm32#M200Toolchain

Build as a appropriate. If you build from the generic tree above, it should have board definitions and variants included as it drags down the right builder and so on. 

I haven't done PRs to merge these because:
A. I built the STM32GENERIC package that you're using if you set your platform=
B. I'm not 100% sure these are correct.
C. I'm still waiting for PIO support for ST's official arduino core, and then I'll likely migrate the variants and definitions to it.

That said, you should be able to build by doing so. If your end goal is to enable auto-bed leveling of some sort, there are some hacks you'll need to use to reduce the rom size - you only get 120k to play with after the bootloader and eeporm emulation eats 8k, so whatever you do needs to fit in 112k. It's doable by using dstring & removing the name demangler, or at least it was.

X7JAY7X, 2018-06-25:
Thanks, I will try it tomorrow.  After its built, how does it get on the M200?  Does the M200 have a bootloader?

xC0000005, 2018-06-25:
For a V1, you copy firmware.bin to update.bin and put it in the root of an SD card along with a file named “fcupdate.flg” (I often copy firmware.bin to BOTH filenames). Put the SD card in the mini, turn it off and back on. If it shows a version of V99.<something> it’s alive. Note the first flash can take longer than the UI knows to wait, so be patient. 

The bootloader is ridiculously picky about what SD card it will flash from - the one that came with it is a good choice, formatted for FAT.

X7JAY7X, 2018-06-25:
Hello,

It now sees the board but I am getting some build errors:

Compiling .pioenvs\malyanm200\src\src\HAL\HAL_LPC1768\arduino.cpp.o
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:28:0,
from Marlin\src\HAL\HAL_LPC1768\LPC1768_Servo.cpp:63:
Marlin\src\HAL\HAL_LPC1768\../../inc/../HAL/HAL_STM32F1/HAL.h:44:10: fatal error: Arduino.h: No such file or directory

Compiling .pioenvs\malyanm200\src\src\HAL\HAL_LPC1768\include\Wire.cpp.o
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:28:0,
from Marlin\src\HAL\HAL_LPC1768\SoftwareSPI.cpp:32:
Marlin\src\HAL\HAL_LPC1768\../../inc/../HAL/HAL_STM32F1/HAL.h:44:10: fatal error: Arduino.h: No such file or directory

xC0000005, 2018-06-25:
What’s your platformio.ini section for the m200 look like?

X7JAY7X, 2018-06-25:
[env:malyanm200]
platform    = ststm32
framework   = arduino
board       = malyanM200
build_flags = !python Marlin/src/HAL/HAL_STM32F1/stm32f1_flag_script.py -DMCU_STM32F103CB -D __STM32F1__=1 -std=c++1y -D MOTHERBOARD="BOARD_MALYAN_M200" -DSERIAL_USB -ffunction-sections -fdata-sections -Wl,--gc-sections
src_filter  = ${common.default_src_filter}
#-<frameworks>
lib_ignore  =
  U8glib
  LiquidCrystal_I2C
  LiquidCrystal
  NewliquidCrystal
  LiquidTWI2
  Adafruit NeoPixel
  TMC2130Stepper
  Servo(STM32F1)
  TMC26XStepper
  U8glib-HAL
  TMC2208Stepper
  c1921b4

xC0000005, 2018-06-25:
It’s highly likely that PlatformIO has broken variants building once again. I’ll look at the PIO builder tonight, but my guess is that if they’ve done what they usually do, there’s some code that assumes that the variant will have the CPU in the name, and the LD script can be derived from that, and so on (instead of reading from the board’s json file like it used to).

[Builder.zip](https://github.com/MarlinFirmware/Marlin/files/2134559/Builder.zip)

I've attached a zip with a couple of files that overwrite their counterparts under .platformio\platforms\ststm32\builder\frameworks\arduino\ (and one that goes in the subfolder maple, overwriting the F1 builder). It's also a hack but I needed the builder running again. Overwrite your builder files with the right versions from this zip and see if it doesn't come closer.

X7JAY7X, 2018-06-26:
Alot of warning but not there is only one error:

Compiling .pioenvs\malyanm200\src\src\HAL\HAL_STM32F4\HAL.cpp.o

Marlin\src\HAL\HAL_STM32F1\persistent_store_flash.cpp: In function 'void HAL::PersistentStore::read_data(int&, uint8_t*, uint16_t, uint16_t*, bool)':

Marlin\src\HAL\HAL_STM32F1\persistent_store_flash.cpp:96:6: error: ambiguating new declaration of 'void HAL::PersistentStore::read_data(int&, uint8_t*, uint16_t, uint16_t*, bool)'

xC0000005, 2018-06-26:
There’s a PR open to fix this, I believe - 
https://github.com/MarlinFirmware/Marlin/pull/11090/files <https://github.com/MarlinFirmware/Marlin/pull/11090/files>

You can apply those changes and I think it will work (I am using STM32GENERIC, so it’s different).

X7JAY7X, 2018-06-26:
That took care of that but now 3 others popped up:

Marlin\src\module\../inc/../HAL/HAL_STM32F1/HAL.h:126:24: error: '__disable_irq' was not declared in this scope

Marlin\src\module\../inc/../HAL/HAL_STM32F1/HAL.h:125:24: error: '__enable_irq' was not declared in this scope

Marlin\src\module\../inc/../HAL/HAL_STM32F1/HAL.h:125:24: error: '__enable_irq' was not declared in this scope

xC0000005, 2018-06-26:
Those are ones I’m not familiar with, I’ll look at them soon.

xC0000005, 2018-06-26:
> Ok, thanks a ton!

I suspect whatever this is is related to https://github.com/MarlinFirmware/Marlin/pull/10999/files <https://github.com/MarlinFirmware/Marlin/pull/10999/files> - you could try reverting that and seeing if it changes anything.

X7JAY7X, 2018-06-26:
I didn't realize that I had an old copy of Marlin.  I pulled the latest, made the changes to persistent_store_flash.cpp, and it now builds successfully.  I will try to load it onto my M200 tonight.  Thanks.

xC0000005, 2018-06-26:
This is more likely to be PIO screwing up variant builds again. There's a rework in progress of the builders and they make some bad assumptions about how to find variant directories, names, include builders, set linker scripts, vector tables, and a few other things.

X7JAY7X, 2018-06-27:
Thanks again for all your help.  Much appreciated.  I loaded the firmware.  Everything works but the fan control.  Do have to change something somewhere?

xC0000005, 2018-06-27:
The M200 config has the fan set to auto - does it not turn on when the temp is > 50 degrees celsius?

X7JAY7X, 2018-06-27:
It does not.  It was working with the original firmware right before I loaded Marlin.  I will check the hardware tomorrow and see if its trying to turn the fan on.  Maybe something happened to my board or fan but doubtful.

Also, can I disable the auto and have GCode control?  I have a hardwired hotend cooling fan to 12v and I also have a layer fan on the mainboard fan plug.   I want the layer fan to be controllable through Gcode.

xC0000005, 2018-06-27:
Absolutely - in advanced configuration.h look for PB8, which is currently assigned to E0_AUTO_FAN somthing. Comment that out and set FAN_PIN in pins.h to PB8 (it’s commented out). The sample configs are from my printer where fan 1 (the default one) is the auto fan) and fan 2 (the unpopulated header on V1) is the layer fan.  I’m not sure what would be stopping it, to be honest, but you can check and see if manual control works. Also, if the LCD shows the right temps, the ADC is reading the data correctly, so it must be something about the config.

> On Jun 26, 2018, at 8:33 PM, X7JAY7X <notifications@github.com> wrote:
> 
> GCode

X7JAY7X, 2018-06-27:
So here was the problem.  #define E0_AUTO_FAN_PIN -1.  I didn't start with your config apparently.
I changed the Maylan pin.h.  It turns on and off with GCode now but PWM isnt working.  Its either 0 or 100% no matter what speed I try to set it to (ex M106 P0 S125 - 50%).

X7JAY7X, 2018-06-27:
#ifndef FAN_PIN
  #define FAN_PIN        PB8   // FAN1 header on board - PRINT FAN
#endif
#define FAN1_PIN           PB3   // FAN2 header on board - CONTROLLER FAN
#define FAN2_PIN           -1    // FAN3 header on board - EXTRUDER0 FAN

xC0000005, 2018-06-27:
Enable soft pwm in the configuration.

xC0000005, 2018-06-27:
#define FAN_SOFT_PWM in configuration.h should give you fan speed control

X7JAY7X, 2018-06-27:
Trying this now.  Is your config in the examples folder?  If not, can you send it to me?  I want to compare settings.  Thanks

xC0000005, 2018-06-27:
Its the one in the examples folder.

X7JAY7X, 2018-06-27:
Thanks, that worked for PWM.  I never had to change any of that before.  Wondering what config files I started with.

xC0000005, 2018-06-27:
The M200 doesn’t have hardware PWM, and even if it did the fan isn’t hooked to the right pins for it. Make sure steps per mm is right for your variant of the M200 and the endstops are inverted correctly for your setup.

xC0000005, 2018-06-27:
That reply made no sense. It doesn’t have hardware PWM set up. The fan pins are PB3 and PB8, so software is the only choice (I hope that makes more sense.)

X7JAY7X, 2018-06-27:
Ok, that makes more sense.  I was going to say that I thought the STM32F103 had hardware PWM.  Everything looks good so far.  I am going to try printing tonight so I will see how that goes.

X7JAY7X, 2018-06-28:
I am having issues with the LCD.  It is very flaky.  It doesn't always update the correctly and get stuck alot.  For example, I am running a simple GCODE file for bed leveling that goes to each corner and when it completes the progress bar is stuck at 94%.  Also, if I change  either temp while its in print mode it locks up.  Do you think I need to update the LCD?  I have never done that.  Its a V1, if I need to update it can you send me the LCD update?

xC0000005, 2018-06-28:
The LCD code is pretty much my own. I will see if I can figure out what keeps the completion screen from popping. The LCD is picky about getting exactly the right messages in the right order. Temperature wise I haven’t seen the printer lock up but I will see if the right command is being sent. If it enqueues a wait, it would wait.

X7JAY7X, 2018-06-28:
Thanks.  One more thing, can I specify how many characters I want to show up of the filenames on the SD card. The original firmware showed alot more than Marlin does.

fiveangle, 2018-06-28:
#define SCROLL_LONG_FILENAMES

xC0000005, 2018-06-28:
That, too, is all my code. I will see.

xC0000005, 2018-06-28:
Try this - in malyanlcd.cpp, line 345, change card.filename to:
card.longFilename[0] ? card.longFilename : card.filename

and see if it shows the long filenames.

xC0000005, 2018-06-28:
Ok, I know the fix for the long file names. I’ll create a PR. Both instances of card.filename need to be switched, and it works fine.

> On Jun 27, 2018, at 9:53 PM, JC Nelson <authorjcnelson@gmail.com> wrote:
> 
> card.longFilename[0] ? card.longFilename : card.filename

fiveangle, 2018-06-28:
If the board json needs updating too, please do the needful

X7JAY7X, 2018-06-28:
My line 345 in maylanlcd.cpp does not contain card.filename.  My line 345 is "write_to_lcd_P(PSTR("{SYS:OK}"));"

[image]

X7JAY7X, 2018-06-28:
I have also noticed that it takes a very long time to cancel a print as well.  Any thoughts on the buggyness of the LCD?

xC0000005, 2018-06-28:
I’m waiting on answers as to how to lay out the variant and adjust the names. Once I have that, I’ll renamed everything and make the changes, but stm32duino and the new ST arduino core are both having similar issues in not working with variants the same way they once did. stm32generic does not, but that’s because I modified the builder in my repo to cope with it. Once PIO settles on a solution, I’ll use it.

xC0000005, 2018-06-28:
First off, let me settle some terms so we’re talking about the same thing. The LCD UI (running on an ESP8266 MCU) is not the problem here. Malyan’s code is directly integrated with the LCD reader, parsing commands over USART1 and directly mucking with the printer’s internal objects in response to commands.

When I set out to build an implementation for Marlin, I wanted to make it work with as few changes as possible to the core of the code. I knew from watching the debug output from Malyan’s motion controller code (which you can do if you hook a USB->TTL converter to the display port at 500,000, 7, N, 1) that some of their operations used enqueue_and_echo. Why do I know it? Because it enqueues and echos that it’s doing so.  

So my solution was to make all the operations inject gcode using the enqueue and echo call. Gcode becomes the “interface” by which the LCD effects changes, if that makes sense, BUT there are some downsides to this approach. 
For instance, if a gcode file sets temp using the “wait” parameter and you adjust it via the LCD, this just doesn’t work. Why? Because the LCD code enqueues a temperature change, but that won’t be executed until after the current one completes. 

I may actually change that to not use enqueue specifically for that reason. 
Cancelling a print is literally no different than the cancel from other LCDs. I mean that because I copied the cancel code at the time I wrote it. (I should have refactored to make a “cancel print” function.)

Anyway, the buggyness of the LCD is pretty much directly related to me implementing the functionality in a way that impacted Marlin’s core code in the least way possible. I do think it makes sense to move to using the thermal manager’s setTargetHotend function to avoid the delay with setting temperatures, and I’m 100% aware that feedrate support doesn’t work.  Long file names I have a fix for, so it’s slowly getting better.

xC0000005, 2018-06-28:
[malyanlcd.cpp.zip](https://github.com/MarlinFirmware/Marlin/files/2146464/malyanlcd.cpp.zip)
This malyanlcd.cpp has the long file name and target temp fixes (note I've only had time to test LFNs). See if it behaves better. Print cancel is no different, feedrate still doesn't work (and isn't even documented in their MMS wiki). I'll fix feedrate next if we can get temp and filenames behaving correctly.

X7JAY7X, 2018-06-28:
I will try it later tonight and let you know how it works.  Any plans to change the Cancel?

X7JAY7X, 2018-06-29:
Long filenmaes are fixed.  I cannot verify the temps.  It immediately is going to the print finished screen when clicking on a file.

X7JAY7X, 2018-06-29:
I commented out line 416 //write_to_lcd_P(PSTR("{TQ:100}")); to get it to print.  Good news is that it looks like the Temp is fixed.  I can change it while its heating up and while its printing.  The display echos back the correct temp as well.

X7JAY7X, 2018-06-29:
I think I see the issue with the finished.  card.sdprinting is one of the flags used to signal that a print has finished for the UI.  card.openAndPrintFile is used to start the print but it doesn't set sdprinting to true though so its always false.

xC0000005, 2018-06-29:
Yes - that’s probably what’s wrong. I’m working on getting my printer back in one piece so I can look at it (I like the M200 but the bed wire design is terrible).

xC0000005, 2018-06-29:
OpenAndPrintFile enqueues an M23, and M23 calls card.startFilePrint(), which will set it to true, but this allows for a window that I think I understand. Once I test something, I'll create a PR.

xC0000005, 2018-06-29:
I understand a great deal more about what's going on. There are some subtle problems and some not so subtle ones.
First off, the display code is way, way too chatty. It's emulating the stock MC firmware but the current code sends way too much data. Secondly, there's a command missing after the {TQ:100}, I'm guessing it's either {SYS:DONE} or {SYS:OK}, but after sending the {TQ:100} it needs one of those (I'm out of town until Monday but then I'll listen in on a stock MC firmware print and figure out which one it is). I've reduced the updates to be only when the print percentage changes, or the UI asks for temperature/percent status, and that cuts out 99% of the serial traffic.

X7JAY7X, 2018-06-29:
Can you send your latest code and I will test?  Thanks

xC0000005, 2018-06-29:
https://raw.githubusercontent.com/xC0000005/Marlin/genericHal/Marlin/src/lcd/malyanlcd.cpp <https://raw.githubusercontent.com/xC0000005/Marlin/genericHal/Marlin/src/lcd/malyanlcd.cpp>

Should be functional except for not showing the “print completed” screen at 100%.

X7JAY7X, 2018-06-30:
This is alot better.  It seems to fix almost all of the issues I was seeing.  It did show the print finished as well for the first print but not the second.  The second one went right back to the file menu but it actually started the print.  When I went to start another print, it stopped printing completely.  Also, before, moving any of the axis was very buggy but now it seems really smooth and responsive.  I will use it more and let you know if I see any more issues.

Should setting the speed work?

thinkyhead, 2018-06-30:
> Should setting the speed work?

If you mean, adjusting the feedrate during a print, then yes, it should work, as long as it's still under the maximum speed and acceleration limits.

xC0000005, 2018-06-30:
I don’t believe I implemented the feedrate menu command yet. I will fix that soon once I get print completion showinfnxoeexflu.

X7JAY7X, 2018-07-01:
After using it a bit more there are other issues.  Sometimes the temps do not display.  They stay at zero.

xC0000005, 2018-07-01:
Does the printer heat up and function otherwise?  I suspect I may need a maximum update time to force a refresh. If it isn’t heating that’s one problem. If the LCD isn’t being told of temp changes that’s another.

xC0000005, 2018-07-02:
I’ve updated the code on github (same location) with support for feedrate, but I’ll need to add diagnostics for the zero temp issue - I can’t repro it if I manually issue a status comand, it always sends the temp status, and {B:0} is resulting in temperature as well. (If you preheat, the LCD starts an S:I loop that continues, and if you don’t, it’s B:0). Can you tell me how you are hitting the zero temp readings?

Eno423, 2018-07-03:
Ill check my printer out in a bit as well as I have been running it on my controller a bit, though I have only ran a handful of prints so far, 

only issues I have are those that we discussed in the other thread

X7JAY7X, 2018-07-04:
I cant seem to reproduce the temps going to 0.  I am using it now.  The only thing I noticed so far was that the SD card is a little buggy.  Sometimes it shows the files and sometimes is says "Please wait" or "SD Card Empty".  I will use it more and report back.  Thanks

jannau, 2018-08-19:
The include <libmaple/atomic.h> error is caused by [moving the header in Arduino_STM32](https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/3f7db854e4072c98fe435126e49f52e62e816256). With the include fixed to <util/atomic.h> compiling works but linking fails with "cannot open linker script file stm32f103.ld: No such file or directory".

vsavic, 2018-08-21:
@jannau did you change stm32f1.py which comes with installation, or version from here: https://github.com/MarlinFirmware/Marlin/issues/11104#issuecomment-400078289

Can you post your modified file?

[38 more replies]

closed 2018-11-07 by thinkyhead