MarlinFirmware/Marlin #11105: Kossel mini Marlin Firmware Calibration problems. ((First time 3D printing))

Beinlaus, 2018-06-24:
### Description

Hello 3D printers. so im new to this game and have had a pretty steep learning curve. Bought a used 3d printer online that suposedly worked and only used for about 30 houers or so he said :P 

I have now installed the newest firmware from Marlin and gone through some relly good youtube videoes on calibrating and adjustments. It seems i am missing somthing as i keep failing to calibrate.
I am a big fan of manual calibration so i am using line 693:#define PROBE_MANUALLY. when i go in to me LCD display and try to run it absolutly nothing happens. i have tried adjusting X Y Z cordinates but it seems the adjustments does not save as when i run a print it digs the nozzel in the plate. 

i have on purpose set the print height larger then the printer is as this is suposed to be calibrated but noting ever calibrates so im kinda stomped right now. 

### Steps to Reproduce

1. [First Step]
LCD: Prepare
2. [Second Step]
LCD:Prepare
3. [and so on...]
LCD: AUTO Calibrate

**Expected behavior:** [What you expect to happen]
Auto Calibration
**Actual behavior:** [What actually happens]
Absolutly nothing
#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

tuskiomi, 2018-06-28:
Do you have your power on for the stepper motor?

thinkyhead, 2018-06-28:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch so we're all on the same page. Once you have given the latest code a test-run, we can move forward and gather more information.

Then, ZIP up your `Configuration.h` and `Configuration_adv.h` files (as requested in the Issue template) and drop them into your next reply so that we can check for errors and build against your configurations, if needed.

boelle, 2019-02-20:
@Beinlaus did you what thinky suggested?

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead