MarlinFirmware/Marlin #11108: How to make UBL work properly

ryan34side, 2018-06-25:
Hello

Is it possible to have precise orders to operate the UBL?
1) What should we put to turn it on the first time once the z OFFSET and done ?

2) And what should we put in the gcode not to redo the leveling of the bed and especially when you turn off and restart the printer ?

thank you very much because with all the posts I see a little of everything and I can not know what to do as command

I have a cr10s with a bl touch and the UBL

thank you very much

Nicolas

sorry for English I'm french!

th33xitus, 2018-06-25:
Did you take a look into the documentation?
http://marlinfw.org/docs/features/unified_bed_leveling.html

ryan34side, 2018-06-25:
yes I saw but I do not know what to put in the boot script if I want to do other print but without redo g29 p1 leveling

forkoz, 2018-06-25:
G29 T - show points
G420 L0 S1; << inside start script to activate bed leveling mesh. You can also have it turn on all the time after G28 from configuration.h and then just to G29 A once.

ryan34side, 2018-06-25:
@forkoz 
For levelling : 
G28
G29 P1
G29 A
G29 S1

for de startup script /

G28
G29t
G420 L0 S1
G28 A

it's exact ?

Roxy-3D, 2018-06-25:
If you set it up so UBL is active at power up (or reset)....   You don't even need a G29 in your start up script.   It will load the mesh and UBL will be active.    Or...  you can do it the other way and do a 
- G28
- G29 L 1
- G29 A
in your start up script.

ryan34side, 2018-06-25:
thank you very much i note it and realized a good leveling once i have inscirt in the config.h my z off extrude i have to do

G28
G29 P1
G29 A
G29 S1

good ?

@Roxy-3D

ryan34side, 2018-06-25:
ok vut we agree to do the leveling I use it well 
G28
G29 P1
G29 A
G29 S1

?

Roxy-3D, 2018-06-25:
To build the mesh and save it...  Use the 'Quick Start' instructions located towards the top of:

http://marlinfw.org/docs/features/unified_bed_leveling.html

ryan34side, 2018-06-25:
thank you for your help I will test tonight !! @Roxy-3D

forkoz, 2018-06-25:
I don't really even need fine tuning on my printer. I just do
G29 P1
G29 P3
G29 S
M500

This is good because I get different mesh values with different bed temperature. Maybe I should save them all to different positions like S0, S1, S2, etc. I only generate a new mesh when I mess with carriage or Z screws or change materials that use different bed temperature.

ryan34side, 2018-06-26:
I tested everything works apparently the only problem is when I turn off or I turn on the printer or when I do a print justa after a leveling the height of the z is not good. after for the print after the height returns to the correct height. I do not understand

thanks for help

@Roxy-3D

Roxy-3D, 2018-06-26:
Can you re-write that with different words.   I don't understand what you are trying to communicate.
Also...    Please consider turning on the Z-BabyStepping in Configuration_adv.h.   And also the 'Double Click' option for it.    That makes it much easier to get the first layer going down right.

boelle, 2019-02-20:
@ryan34side is this still an issue?

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead