MarlinFirmware/Marlin #11111: STOP due to BLTouch probe .. really stupid catch-22

AndKe, 2018-06-25:
### Description  
and
### Steps to Reproduce

if, on connection/boot of the printer, the BLTouch will touch the printbed, then user gets:

```
:STOP called because of BLTouch error - restart with M999
Changing monitoring state from "Connecting" to "Error: STOP called because of BLTouch error - restart with M999"
Changing monitoring state from "Error: STOP called because of BLTouch error - restart with M999" to "Offline (Error: STOP called because of BLTouch error - restart with M999)"
Connection closed, closing down monitor
```
now connection is closed, unable to lift Z axis, and/or perform M999
Physical visit to the printer is required..

**Expected behavior:** [What you expect to happen]

As no printing (levelling) is going on, not even were the BLTouch motion called for, I would expect NOT to do a STOP.

Maybe except while levelling or using BLtouch as Z endstop, an BLTouch error should never cause STOP.

GMagician, 2018-06-25:
On boot BLtouch is tested, even if not used, if something goes wrong you get the error.

thinkyhead, 2018-06-27:
I'm not sure how to work around it. The BLTouch is tested to see if it's "triggered" before even being deployed, and AFAIK it shouldn't ever show this state unless it has been turned on for probing.

The patch was originally made by @Bob-the-Kuhn at 0369f97ec11b98ef20445bdda2b7f63dcb3b63d7 so perhaps he can provide some explanation why it might be giving this unusual result.

AndKe, 2018-06-27:
more or less the same goes for 

```
Changing monitoring state from "Printing" to "Error: Probing failed"
Changing monitoring state from "Error: Probing failed" to "Offline (Error: Probing failed)"
Connection closed, closing down monitor
```

printer closes connection, says nothing about "STOP" - and even leaves heaters on.
basically sabotaging further remote control before reboot.

A much better solution would be to do a STOP, ignoring further gcode until M999

GMagician, 2018-06-27:
Error is raised when probe is too lower and can't extend completely because it touch the bed.
One solution can be to remove bltouch init call and replace it with same code present now in G28 (it reset bltouch without alarm). Current 'Init' code should be moved to 'probe down' request and called just once (storing status after first valid test)

I may try to address it if this solution is considered as valid

thinkyhead, 2018-06-28:
It might just be OctoPrint (or other host software) closing the connection. It shouldn't do that on STOP, but I believe if it sees "Error:" then it takes that as a cue to disconnect.

closed 2018-06-29 by thinkyhead