MarlinFirmware/Marlin #11112: [bugfix-2.0.x] homing continues

hamster65, 2018-06-25:
### Description
My x and y axis do not work (I think I killed both of my TMC2130, maybe it was the power sequence, still investigating). So those 2 axes do not move at all for now. 
When I home the printer, nothing moves for a couple of seconds while Marlin tries to move x and y. After some seconds, Marlin lowers the Z axis. 

Safe homing and QUICK_HOME is enabled. 

### Steps to Reproduce
1.) Disable x and y axis by removing your stepper drivers or detaching belts or fry your motor drivers :)
2.) Do a G28

**Expected behavior:**
- Marlin cancels homing and goes into an error state because end stops are not being reached and thus the x and y position is unknown, meaning that z homing cannot be performed safely.

**Actual behavior:**
- Marlin continues homing, my bltouch probe misses the bed and part of the print head crashes into the bed.
(Marlin 2.0-x compiled the day before)

thinkyhead, 2018-06-30:
Notes on implementation:

The only way to effectively detect failure of an endstop/probe move is to add new logic to the low-level stepper code, asserting that the current move *must* be interrupted by an endstop or probe trigger, and if the move completes without triggering an endstop then call "kill" to halt the machine.

I'm not sure how this will interact with the situation where endstops are enabled during printing, something which is allowed, but which doesn't use a separate flag. We would need to extend the endstop enabled flags so that they can contain a value other than `true`/`false` for the case where we want to allow endstops being enabled, but we don't want completed moves to be considered an error.

We'll also need to check that there are no situations where homing or probe moves are segmented, since that would prevent applying the proposed logic.

thinkyhead, 2018-07-01:
Merged changes to the bugfix branches to call `kill` on a failed homing move.

closed 2018-07-01 by thinkyhead