MarlinFirmware/Marlin #11114: [Feature Request] add RADDS2LCD functionality

freakezoid1604, 2018-06-25:
Hi
is it possilble to add the RADDS2LCD it to Marlin/Marlin2.0?
Here the link to the RADDS2LCD Project https://www.thingiverse.com/thing:1740725
I try to add it, but i was not able to do it.
Thanks

smoki3, 2018-06-26:
So it's easy... 

Enable the Full Graphic Reprap Discount Display first.

Then open pins_RADDS.h
Scroll down to 

'  #elif ENABLED(REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER)

    #define LCD_PINS_RS     46
    #define LCD_PINS_ENABLE 47
    #define LCD_PINS_D4     44'

Change the lines to:

'#define LCD_PINS_RS     42
    #define LCD_PINS_ENABLE 43
    #define LCD_PINS_D4     44'

freakezoid1604, 2018-06-26:
I did that but it didn't work, but i will try it again in the evening.
But thanks for your answer. I will give you an feedback.

smoki3, 2018-06-26:
Its working using this for more than a year now

freakezoid1604, 2018-06-26:
[image]
On my side it didn’t work

smoki3, 2018-06-26:
Does it work on repetier for you? otherwise I think its an adapter/lcd error

freakezoid1604, 2018-06-26:
Yes it works
[image]

smoki3, 2018-06-26:
can you upload your config?

freakezoid1604, 2018-06-26:
Here my config.
For testing i use a blank marlinfw down load. i only adujst the board and the pin plus the full graphic display
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2137747/Configuration.zip)

smoki3, 2018-06-26:
You dont use RADDS as board.

freakezoid1604, 2018-06-26:
Arg mix a folder and the project in arduino
give me some seconds i try again

freakezoid1604, 2018-06-26:
but the same situation
Aber das gleiche Fehlerbild
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2137807/Configuration.zip)

smoki3, 2018-06-26:
So have you tried to connect a host? Or press the reset button on the RADDS. It doesnt boot with the display at the first time. 
Also please connect the thermistors. 
Because Marlin doesn't start as far as I know without thermistors.

freakezoid1604, 2018-06-26:
i have aready thermistors on the board. Also i connect the powersupply. i press the button for the reset many times (ond RADDS and on the due)

smoki3, 2018-06-26:
Are you able to control your printer with a host?

freakezoid1604, 2018-06-26:
Yes I can connect and operate the board. So it seem it run

smoki3, 2018-06-26:
So really wired. I use the same adapter for over one year without any problems

freakezoid1604, 2018-06-26:
Version 3 yes with the 1kohm resistors
[image]

smoki3, 2018-06-26:
Ok my versions doesn't have the resistors

freakezoid1604, 2018-06-26:
Maybe it could be the problem but on repetier it works fine so hm.

smoki3, 2018-06-26:
I think that the schematic of you adapter PCB.

freakezoid1604, 2018-07-17:
[image]
That’s the schematics

smoki3, 2018-07-17:
yes here its working fine. On all firmwares i have ever tested

freakezoid1604, 2018-07-18:
with that change?

`So it's easy...

Enable the Full Graphic Reprap Discount Display first.

Then open pins_RADDS.h
Scroll down to

  #elif ENABLED(REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER)

    #define LCD_PINS_RS     46
    #define LCD_PINS_ENABLE 47
    #define LCD_PINS_D4     44

Change the lines to:

    #define LCD_PINS_RS     42
    #define LCD_PINS_ENABLE 43
    #define LCD_PINS_D4     44`

smoki3, 2018-07-18:
Yes exactly

thinkyhead, 2018-11-05:
Marlin now has `RADDS_DISPLAY` as an LCD option.

closed 2018-11-05 by thinkyhead