MarlinFirmware/Marlin #11117: [Feature Request] galvo support

Blacksheep70, 2018-06-25:
I already tried to get contact via Facebook but there was no response yet.

I know that therre was some attempts to integrate galvo support in Marlin but this seems to stuck for a while. I just want to point out that there exist a (5 to 6 year old) fork of Marlin where it seems there is already the neccessary done:

[beardface/Marlin-OpenSL](https://github.com/beardface/Marlin-OpenSL)

As my build comes close to the end I would like to push things forward. So I forked this projekt and was able to modify that old code that it could be compiled with an actual Arduino IDE. This code can be pulled from 

[Blacksheep70/Marlin-OpenSLA](https://github.com/Blacksheep70/Marlin-OpenSLA )

As I am not a professional developer I would like to make an offer: I would spend a complete set with

- Galvos
- Driver cards
- PCB with a DAC and two OpAmps to amplify the ILDA signal
- a RAMPS 1.4 Set

for the project and send it to the developer who take responsibility of this task. The set is completely wired and ready to use. In all hardware questions I would assist of course. The hardware works and a demo program shows that the galvos are able to draw animations / pictures. So "all" to be done is to integrate the possibility to run this galvos with Marlin using a DAC. (via SPI).

Furthermore I would pay a bounty of 200 euros for the Marlin project. I have no commercial interests, I am just a guy who wants to see this feature integrated and get the own SLA printer completed.

thinkyhead, 2018-06-27:
The PR #6977 is on hold, as the 2.0.x branch work is focused on the HAL at this time, and it's a massive job simply to get a version of Marlin that can handle `EXTRUDERS 0`.

nesquik011, 2018-08-14:
i am lost so no one ever got it to work ? 
i am really really lost so does it support Galvanometer or no ! ?  and if not  may you check that system and try to put it in  or work on it   

https://www.youtube.com/watch?v=hbFObc3SvLU

thinkyhead, 2018-08-21:
You may be able to hire a developer for $$$ to implement the feature for you. Everyone here is too busy to attend to that project, which is quite complex due to the need to first support zero `EXTRUDERS`. Post a "Help Wanted" in the RepRap forums and other 3D printing communities with a good cash offer, and someone will likely respond.

RavenWarrior, 2018-09-28:
Does the marlin for laser cutter have 0 extruders?
Just a thought

RavenWarrior, 2018-09-28:
I best start learning marling code writing then. See you in 10years 🤣

thinkyhead, 2018-09-29:
That would be great! The more volunteer developers who can join in to help, the more that we can move Marlin forward. We expect to still be here in 10 years, if you're still interested at that time.

Phr3d13, 2018-09-29:
A lot of people (myself included) start small. Helping with docs, helping with configuration issues, etc, and then progress from there. I still barely know what I'm doing but still manage to help out (I've helped add 2 boards and a few example configs).

thinkyhead, 2018-09-29:
We’ll try to put more “good first issue” tags on issues that beginners should be able to tackle. Those can then be searched by tag in the issue queue.

RavenWarrior, 2018-09-29:
Quool, I hope I can get my head round it all and help you guys out.

Blacksheep70, 2018-10-01:
Hereby I am revoking my offer to pay a 200 euro bounty. Meanwhile I am working on an own solution.

closed 2018-10-01 by Blacksheep70

thinkyhead, 2018-10-02:
Good luck! If you need guidance from some smart cookies who may not have visited this page, a good place to hang out is the Marlin Firmware Discord server. One of the admins would be happy to add you to the "developers" group for extra access. https://discord.gg/n5NJ59y