MarlinFirmware/Marlin #11124: Manual 'Auto' leveling does not adjust to Delta printer's inherent parabola.

tuskiomi, 2018-06-27:
### Description

Delta printers have an inherent parabolic that is built into the marlin firmware, and I would assume would be taken out by adjusting the mesh leveling. This parabola manifests on the XY plane where the top is the center, and the bottom is around the edges, and is proportional to the radius from the center. This parabola is not fixed by the manual Bi-linear bed leveling process.

Here is a picture of the delta printer at X0 Y0 Z1: https://i.imgur.com/oeludpC.jpg

And here is a picture of the delta printer at X50 Y0 Z1: https://i.imgur.com/vcORQWF.jpg

As you can see, with the Z coordinate being the same, The nozzle will touch in some places, while not others, even after calibrating the bed. This leads me to believe that the manual calibration does little to actually adjust the height of the nozzle in a grid formation

### Steps to Reproduce

1. configure delta kossel with this config: https://pastebin.com/Hr0epKWJ, and this adv_config: https://pastebin.com/ZTeZvKJD.  Upload. 
2. Center using `G28`. home to Z1 using `G0 Z1`
3.  Observe that the head is 1mm above the plate. 
4.  Move to  X50 Y0 Z1 using `G0 X50`
5. observe that the nozzle touches the plate
6. Enter `G28` Then `G29` to level the bed.
7. Once done leveling the bed, repeat steps 2 - 5 and observe no change, where there should be.

#### Additional Information

I'll have an inductive  sensor to test this automatically later.

thinkyhead, 2018-06-28:
This is more what `G33` is intended to do. First, use that command to get the geometry sorted out, and only then will `G29` produce reasonable results.

closed 2018-06-28 by tuskiomi

thinkyhead, 2018-06-30:
We hope to produce more step-by-step instructional videos on various features as time allows.