MarlinFirmware/Marlin #11127: Hardcode an entire print?

iadams7, 2018-06-27:
I defined a list of commands such as seen at the bottom, with 

#define calibrate_pla_1st_layer_script "string of Gcode commands" 

Then created a new menu item to run the script. 

void calibrate_pla_1st_layer_script_function(){
       enqueue_and_echo_commands_P(PSTR(calibrate_pla_1st_layer_script));
       lcd_return_to_status();
     }

MENU_ITEM(function, MSG_CALIBRATE_PLA, calibrate_pla_1st_layer_script_function);

When I run the line on the printer nothing will happen, and when I observe the run on Printrun I will eventually get the line 
echo:Unknown command: "" 
and it fails. 

I have triple checked the gcode string and have never found any typos or trailing whitespaces.

Is there a way to hardcode a print to run from a menu line?

#define calibrate_pla_1st_layer_script "G21 \nG90 \nM82 \nM107 \nM117 Preparing \nM140 S45 \nM109 S180 \nG28 \nM203 Z4 \nG29 \nM203 Z7 \nG28 X \nM420 S1 \nG0 Y0 F5000 \nG0 Z15 F3000 \nM190 S45 \nM109 S210 \nM300 S440 P100 \nG4 P200 \nM300 S440 P100 \nG4 P200 \nM300 S440 P800 \nM0 Live adjust Z height!\nM420 S1 \nM117 Print starting \nG90 \nG92 E0 \nM420 S1 \nG0 Z0 \nM83 \nG1 E22 F500 \nG1 E18 F400 \nG1 E10 F400 \nG4 S2 \nM400 \nM420 S1 \nG0 F500 X3 Y0 Z0.6\nG1 E2 F300 \nG1 F1000 X104 E10 \nM82 \nG92 E0 \nM107\nG0 F3600 X40.275 Y32.461 Z0.3\nG1 F840 X40.546 Y32.22 E0.01809\nG1 X40.863 Y32.053 E0.03597\nG1 X41.22 Y31.966 E0.0543\nG1 X41.58 Y31.966 E0.07226\nG1 X41.937 Y32.053 E0.09059\nG1 X42.254 Y32.22 E0.10847\nG1 X42.525 Y32.461 E0.12656\nG1 X42.736 Y32.768 E0.14515\nG1 X42.862 Y33.097 E0.16272\nG1 X42.91 Y33.493 E0.18262\nG1 X42.91 Y58.487 E1.42958\nG1 X42.957 Y58.848 E1.44774\nG1 X43.106 Y59.209 E1.46723\nG1 X43.343 Y59.519 E1.48669\nG1 X43.653 Y59.756 E1.50616\nG1 X44.014 Y59.904 E1.52563\nG1 X44.4 Y59.956 E1.54506\nG1 X44.786 Y59.904 E1.56449\nG1 X45.147 Y59.756 E1.58396\nG1 X45.457 Y59.519 E1.60342\nG1 X45.694 Y59.209 E1.62289\nG1 X45.843 Y58.848 E1.64238\nG1 X45.89 Y58.487 E1.66054\nG1 X45.89 Y33.493 E2.90749\nG1 X45.938 Y33.097 E2.9274\nG1 X46.064 Y32.768 E2.94497\nG1 X46.275 Y32.461 E2.96356\nG1 X46.546 Y32.22 E2.98165\nG1 X46.863 Y32.053 E2.99953\nG1 X47.22 Y31.966 E3.01786\nG1 X47.58 Y31.966 E3.03582\nG1 X47.937 Y32.053 E3.05415\nG1 X48.254 Y32.22 E3.07203\nG1 X48.525 Y32.461 E3.09012\nG1 X48.736 Y32.768 E3.1087\nG1 X48.862 Y33.097 E3.12628\nG1 X48.91 Y33.493 E3.14618\nG1 X48.91 Y58.487 E4.39314\nG1 X48.957 Y58.848 E4.4113\nG1 X49.106 Y59.209 E4.43078\nG1 X49.343 Y59.519 E4.45025\nG1 X49.653 Y59.756 E4.46972\nG1 X50.014 Y59.904 E4.48919\nG1 X50.4 Y59.956 E4.50862\nG1 X50.786 Y59.904 E4.52805\nG1 X51.147 Y59.756 E4.54751\nG1 X51.457 Y59.519 E4.56698\nG1 X51.694 Y59.209 E4.58645\nG1 X51.843 Y58.848 E4.60593\nG1 X51.89 Y58.487 E4.6241\nG1 X51.89 Y33.493 E5.87105\nG1 X51.938 Y33.097 E5.89095\nG1 X52.064 Y32.768 E5.90853\nG1 X52.275 Y32.461 E5.92711\nG1 X52.546 Y32.22 E5.94521\nG1 X52.863 Y32.053 E5.96308\nG1 X53.22 Y31.966 E5.98142\nG1 X53.58 Y31.966 E5.99938\nG1 X53.937 Y32.053 E6.01771\nG1 X54.254 Y32.22 E6.03558\nG1 X54.525 Y32.461 E6.05368\nG1 X54.736 Y32.768 E6.07226\nG1 X54.862 Y33.097 E6.08984\nG1 X54.91 Y33.493 E6.10974\nG1 X54.91 Y58.487 E7.35669\nG1 X54.957 Y58.848 E7.37486\nG1 X55.106 Y59.209 E7.39434\nG1 X55.343 Y59.519 E7.41381\nG1 X55.653 Y59.756 E7.43328\nG1 X56.014 Y59.904 E7.45274\nG1 X56.4 Y59.956 E7.47217\nG1 X56.786 Y59.904 E7.49161\nG1 X57.147 Y59.756 E7.51107\nG1 X57.457 Y59.519 E7.53054\nG1 X57.694 Y59.209 E7.55001\nG1 X57.843 Y58.848 E7.56949\nG1 X57.89 Y58.487 E7.58765\nG1 X57.89 Y33.493 E8.83461\nG1 X57.938 Y33.097 E8.85451\nG1 X58.064 Y32.768 E8.87209\nG1 X58.275 Y32.461 E8.89067\nG1 X58.546 Y32.22 E8.90876\nG1 X58.863 Y32.053 E8.92664\nG1 X59.22 Y31.966 E8.94497\nG1 X59.58 Y31.966 E8.96293\nG1 X59.937 Y32.053 E8.98126\nG1 X60.254 Y32.22 E8.99914\nG1 X60.525 Y32.461 E9.01723\nG1 X60.736 Y32.768 E9.03582\nG1 X60.862 Y33.097 E9.0534\nG1 X60.91 Y33.493 E9.0733\nG1 X60.91 Y58.487 E10.32025\nG1 X60.957 Y58.848 E10.33841\nG1 X61.106 Y59.209 E10.3579\nG1 X61.343 Y59.519 E10.37737\nG1 X61.653 Y59.756 E10.39683\nG1 X62.014 Y59.904 E10.4163\nG1 X62.4 Y59.956 E10.43573\nG1 X62.786 Y59.904 E10.45516\nG1 X63.147 Y59.756 E10.47463\nG1 X63.457 Y59.519 E10.4941\nG1 X63.694 Y59.209 E10.51356\nG1 X63.843 Y58.848 E10.53305\nG1 X63.89 Y58.487 E10.55121\nG1 X63.89 Y33.493 E11.79817\nG1 X63.938 Y33.097 E11.81807\nG1 X64.064 Y32.768 E11.83564\nG1 X64.275 Y32.461 E11.85423\nG1 X64.546 Y32.22 E11.87232\nG1 X64.863 Y32.053 E11.8902\nG1 X65.22 Y31.966 E11.90853\nG1 X65.58 Y31.966 E11.92649\nG1 X65.937 Y32.053 E11.94482\nG1 X66.254 Y32.22 E11.9627\nG1 X66.525 Y32.461 E11.98079\nG1 X66.736 Y32.768 E11.99938\nG1 X66.862 Y33.097 E12.01695\nG1 X66.91 Y33.493 E12.03685\nG1 X66.91 Y58.487 E13.28381\nG1 X66.957 Y58.848 E13.30197\nG1 X67.106 Y59.209 E13.32146\nG1 X67.343 Y59.519 E13.34092\nG1 X67.653 Y59.756 E13.36039\nG1 X68.014 Y59.904 E13.37986\nG1 X68.4 Y59.956 E13.39929\nG1 X68.786 Y59.904 E13.41872\nG1 X69.147 Y59.756 E13.43819\nG1 X69.457 Y59.519 E13.45765\nG1 X69.694 Y59.209 E13.47712\nG1 X69.843 Y58.848 E13.49661\nG1 X69.89 Y58.487 E13.51477\nG1 X69.89 Y33.493 E14.76172\nG1 X69.938 Y33.097 E14.78162\nG1 X70.064 Y32.768 E14.7992\nG1 X70.275 Y32.461 E14.81779\nG1 X70.546 Y32.22 E14.83588\nG1 X70.863 Y32.053 E14.85375\nG1 X71.22 Y31.966 E14.87209\nG1 X71.58 Y31.966 E14.89005\nG1 X71.937 Y32.053 E14.90838\nG1 X72.254 Y32.22 E14.92625\nG1 X72.525 Y32.461 E14.94435\nG1 X72.736 Y32.768 E14.96293\nG1 X72.862 Y33.097 E14.98051\nG1 X72.91 Y33.493 E15.00041\nG1 X72.91 Y58.487 E16.24737\nG1 X72.957 Y58.848 E16.26553\nG1 X73.106 Y59.209 E16.28501\nG1 X73.343 Y59.519 E16.30448\nG1 X73.653 Y59.756 E16.32395\nG1 X74.014 Y59.904 E16.34341\nG1 X74.4 Y59.956 E16.36285\nG1 X74.786 Y59.904 E16.38228\nG1 X75.147 Y59.756 E16.40174\nG1 X75.457 Y59.519 E16.42121\nG1 X75.694 Y59.209 E16.44068\nG1 X75.843 Y58.848 E16.46016\nG1 X75.89 Y58.487 E16.47832\nG1 X75.89 Y33.493 E17.72528\nG1 X75.938 Y33.097 E17.74518\nG1 X76.064 Y32.768 E17.76276\nG1 X76.275 Y32.461 E17.78134\nG1 X76.546 Y32.22 E17.79944\nG1 X76.863 Y32.053 E17.81731\nG1 X77.22 Y31.966 E17.83564\nG1 X77.58 Y31.966 E17.8536\nG1 X77.937 Y32.053 E17.87194\nG1 X78.254 Y32.22 E17.88981\nG1 X78.525 Y32.461 E17.90791\nG1 X78.736 Y32.768 E17.92649\nG1 X78.862 Y33.097 E17.94407\nG1 X78.91 Y33.493 E17.96397\nG1 X78.91 Y58.487 E19.21092\nG1 X78.957 Y58.848 E19.22909\nG1 X79.106 Y59.209 E19.24857\nG1 X79.343 Y59.519 E19.26804\nG1 X79.653 Y59.756 E19.28751\nG1 X80.014 Y59.904 E19.30697\nG1 X80.4 Y59.956 E19.3264\nG1 X80.786 Y59.904 E19.34583\nG1 X81.147 Y59.756 E19.3653\nG1 X81.457 Y59.519 E19.38477\nG1 X81.694 Y59.209 E19.40424\nG1 X81.843 Y58.848 E19.42372\nG1 X81.89 Y58.487 E19.44188\nG1 X81.89 Y33.493 E20.68884\nG1 X81.938 Y33.097 E20.70874\nG1 X82.064 Y32.768 E20.72632\nG1 X82.275 Y32.461 E20.7449\nG1 X82.546 Y32.22 E20.76299\nG1 X82.863 Y32.053 E20.78087\nG1 X83.22 Y31.966 E20.7992\nG1 X83.58 Y31.966 E20.81716\nG1 X83.937 Y32.053 E20.83549\nG1 X84.254 Y32.22 E20.85337\nG1 X84.525 Y32.461 E20.87146\nG1 X84.736 Y32.768 E20.89005\nG1 X84.862 Y33.097 E20.90762\nG1 X84.91 Y33.493 E20.92753\nG1 X84.91 Y58.487 E22.17448\nG1 X84.957 Y58.848 E22.19264\nG1 X85.106 Y59.209 E22.21213\nG1 X85.343 Y59.519 E22.2316\nG1 X85.653 Y59.756 E22.25106\nG1 X86.014 Y59.904 E22.27053\nG1 X86.4 Y59.956 E22.28996\nG1 X86.786 Y59.904 E22.30939\nG1 X87.147 Y59.756 E22.32886\nG1 X87.457 Y59.519 E22.34832\nG1 X87.694 Y59.209 E22.36779\nG1 X87.843 Y58.848 E22.38728\nG1 X87.89 Y58.487 E22.40544\nG1 X87.89 Y33.493 E23.65239\nG1 X87.938 Y33.097 E23.6723\nG1 X88.064 Y32.768 E23.68987\nG1 X88.275 Y32.461 E23.70846\nG1 X88.546 Y32.22 E23.72655\nG1 X88.863 Y32.053 E23.74443\nG1 X89.22 Y31.966 E23.76276\nG1 X89.58 Y31.966 E23.78072\nG1 X89.937 Y32.053 E23.79905\nG1 X90.254 Y32.22 E23.81693\nG1 X90.525 Y32.461 E23.83502\nG1 X90.736 Y32.768 E23.8536\nG1 X90.862 Y33.097 E23.87118\nG1 X90.91 Y33.493 E23.89108\nG1 X90.91 Y58.487 E25.13804\nG1 X90.957 Y58.848 E25.1562\nG1 X91.106 Y59.209 E25.17568\nG1 X91.343 Y59.519 E25.19515\nG1 X91.653 Y59.756 E25.21462\nG1 X92.014 Y59.904 E25.23409\nG1 X92.4 Y59.956 E25.25352\nG1 X92.786 Y59.904 E25.27295\nG1 X93.147 Y59.756 E25.29241\nG1 X93.457 Y59.519 E25.31188\nG1 X93.694 Y59.209 E25.33135\nG1 X93.843 Y58.848 E25.35083\nG1 X93.89 Y58.487 E25.369\nG1 X93.89 Y33.493 E26.61595\nG1 X93.938 Y33.097 E26.63585\nG1 X94.064 Y32.768 E26.65343\nG1 X94.275 Y32.461 E26.67201\nG1 X94.546 Y32.22 E26.69011\nG1 X94.863 Y32.053 E26.70798\nG1 X95.22 Y31.966 E26.72632\nG1 X95.58 Y31.966 E26.74 [... cut ...] \nG1 X127.343 Y59.519 E42.9765\nG1 X127.653 Y59.756 E42.99596\nG1 X128.014 Y59.904 E43.01543\nG1 X128.4 Y59.956 E43.03486\nG1 X128.786 Y59.904 E43.05429\nG1 X129.147 Y59.756 E43.07376\nG1 X129.457 Y59.519 E43.09323\nG1 X129.694 Y59.209 E43.11269\nG1 X129.843 Y58.848 E43.13218\nG1 X129.89 Y58.487 E43.15034\nG1 X129.895 Y34.283 E44.35788\nM117 Finishing Up \nM104 S0 \nM140 S0 \nG91 \nG1 E-1 F300 \nG1 Z+0.5 E-10 X-20 Y-20 F9000 \nG1 E-15 \nG90 \nG28 X \nG1 Y140 \nM84 \nM117 Print finished \nM82 \n"

crzcrz, 2018-06-27:
This is the source of `enqueue_and_echo_commands_P`:

```c++
/**
 * Inject the next "immediate" command, when possible, onto the front of the queue.
 * Return true if any immediate commands remain to inject.
 */
static bool drain_injected_commands_P() {
  if (injected_commands_P != NULL) {
    size_t i = 0;
    char c, cmd[30];
    strncpy_P(cmd, injected_commands_P, sizeof(cmd) - 1);
    cmd[sizeof(cmd) - 1] = '\0';
    while ((c = cmd[i]) && c != '\n') i++; // find the end of this gcode command
    cmd[i] = '\0';
    if (enqueue_and_echo_command(cmd))     // success?
      injected_commands_P = c ? injected_commands_P + i + 1 : NULL; // next command or done
  }
  return (injected_commands_P != NULL);    // return whether any more remain
}

/**
 * Record one or many commands to run from program memory.
 * Aborts the current queue, if any.
 * Note: drain_injected_commands_P() must be called repeatedly to drain the commands afterwards
 */
void enqueue_and_echo_commands_P(const char * const pgcode) {
  injected_commands_P = pgcode;
  (void)drain_injected_commands_P(); // first command executed asap (when possible)
}
```

So it will only execute 1 command. It appears that you have to do something like:

```c++
enqueue_and_echo_commands_P(PSTR(calibrate_pla_1st_layer_script));
while(drain_injected_commands_P());
```

iosonopersia, 2018-06-27:
Well, that will consume a LOT of program memory...

MasterPIC, 2018-06-28:
I think it's possible to add a menu entry to print a dedicated file stored on SD

thinkyhead, 2018-06-28:
While `enqueue_and_echo_commands_P` _can_ run multiple commands, the more commands you try to run through it, the more likely it becomes that some internal function will replace the string when it wants to use `enqueue_and_echo_commands_P` for its own purposes. So basically: you can't do that. Instead, as suggested by @crzcrz, you will need to create a custom G-code that, when executed, processes the whole string in a contained loop, sending one command at a time to `enqueue_and_echo_command`.

marcio-ao, 2018-06-29:
Is this maybe a job for `process_subcommands_now_P`? We have used it for executing long macros.

marcio-ao, 2018-06-29:
For reference [#10450](https://github.com/MarlinFirmware/Marlin/pull/10450)

marcio-ao, 2018-06-29:
Oh, right. This is for LCD code. Generally it is not a good idea to call `process_subcommands_now_P` from the LCD code due to blocking. One alternative would be to create a new GCODE, have *it* call `process_subcommands_now_P`, and then use `enqueue_and_echo_commands_P` to have the LCD code enqueue that new GCODE.

I'm not sure how @thinkyhead feels about adding new GCODEs for custom things like that. Is there maybe a reserved set of GCODE for use-defined features? If not, maybe use something gnarly like G347429...

thinkyhead, 2018-06-29:
> I'm not sure how @thinkyhead feels about adding new GCODEs for custom things like that.

Users are free to do what they like in their personal copy of Marlin. It's highly unlikely this would ever be a general feature for all users.

Roxy-3D, 2018-06-29:
Also...  Why would you want the printer to be able to print an entire object from its firmware?  An SD Memory card can hold what ever you want.   Can't you just print from that?

If you just want to check out the printer's ability to put down a first layer....   Have you checked out the G26 Mesh Validation Pattern?

filipgoc, 2018-06-30:
@Roxy-3D Good idea. Yet, one of the main reasons we did not consider your G26 is the inability to couple it with live z babystepping (as we discussed in another thread at some point). 

That being said, we not only want a custom pattern, but I'd prefer this one to come from the slicer, as that guarantees the same kind of flow management as the actual prints.

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead