MarlinFirmware/Marlin #11128: [FR] M code to reset/reboot Marlin

alexxy, 2018-06-27:
It will be good to have M code that will reboot or reset marlin fw.

For example Smoothieware uses reset command to reboot board.

p3p, 2018-06-27:
I was pondering the same thing, useful for flashing the LPC176x boards remotely, I have mine setup to software reset on the break character detected with the Emergency Parser feature, the same feature could be useful for other real-time control characters but I'm not sure @thinkyhead would like me adding grbl like commands so perhaps an M-code would be preferred

Roxy-3D, 2018-06-27:
I think this would be a good idea.   But, is it possible to actually do a real reset?   What I'm concerned about is at reset, the hardware starts off in a known, clean state.   If we just start re-initializing hardware pieces, do we really come up clean?

alexxy, 2018-06-27:
For LPC we can call 

```
NVIC_SystemReset();
```

Grogyan, 2018-06-27:
The Host service should already be able to issue a reset after programming.
Not sure this is needed for firmware programming when there are plugins such as [this one] (https://github.com/OctoPrint/OctoPrint-FirmwareUpdater)

alexxy, 2018-06-27:
But not for LPC

Grogyan, 2018-06-27:
Write a plugin that supports the LPC and others.

p3p, 2018-06-27:
@Roxy-3D Software resets tend to just set the program counter back to 0, (call the reset interrupt, will be very different between mcus) so hardware state could be preserved if it not reinitialised by the software, RAM state may not be cleaned etc, although firmware may set it to zero in the framework initialisation, so no I doubt the state would be as clean as a hard reset, but it does work well in my experience.

@alexxy indeed that is how I'm currently doing it with mine, it's trivial to implement, but if you want an M code rather than just the break character as I'm currently doing there will need to be some consensus.

xC0000005, 2018-06-28:
NVIC_SystemReset would work on STM32 systems, but unfortunately leave battery backed ram untouched (even on non battery backed systems). Since that's a common place to stash status information for flashing and so on, a reset on some STM32 boards would not have the same effect it does on others.

zachwelch, 2018-07-26:
If NVIC_SystemReset fails to clear state, then that is 100% a bug in the Marlin bootcode. The firmware startup process should reinitialize every single piece of hardware and memory that it touches for this exact reason, because otherwise that call does not do what it claims it should do. That goes for _any_ firmware, not just Marlin.

TwatBurglar, 2018-10-02:
+1 on what @zachwelch  said.   

And also +1 on having an M command issue NVIC_SystemReset()  on LPC !   Would be quite useful.

Andrei-Pozolotin, 2019-02-28:
@p3p Chris:
can you please share your code for  `software reset on the break character detected with the Emergency Parser`?

Andrei-Pozolotin, 2019-03-05:
Here is a way to reflash/reboot LPC1768: relies on remote ssh via [OctoPi](https://octoprint.org).
Note: currently Marlin always mounts sd card internally on boot #13274 #13288.

Configuration:
1. Enable sd card support in `OctoPrint`. This will also issue `M21` on connect (sd card mount).
2. Configure `OctoPrint->Settings->Gcode Script->After connect`  command `M22`  (sd card unmount).
3. Create `OctoPrint` [custom control](https://plugins.octoprint.org/plugins/customControl/) to issue reset command [M997](https://github.com/MarlinFirmware/Marlin/pull/13281) from user interface.
4. Provide simple ssh upload/mount/copy script, i.e. [update.sh](https://github.com/random-builder/Marlin/blob/wanhao-d6-mks-sbase/buildroot/update.sh)

The workflow becomes then:
1. Build `firmware.bin` locally on developer workstation. This gives a new image.
2. Click `OctoPrint->Connection->Connect`. This exposes sd card to the `OctoPi` host.
3. Invoke `update.sh`. This will upload image, mount sd card, copy image from `OctoPi` to sd card.
4. Click `OctoPrint->Control->Reboot`. This will issue `M997`, and will leave `OctoPrint` disconnected.

Powerprobot, 2019-07-07:
I use Marlin from 2019-07-6 and M997 on a SKR 1.3 Board would not work.
What work is by enabled AUTO_BED_LEVELING_BILINEAR
G28
G29 S0
after entering these commands, the Board will do a full reset!
Is this a Bug or a Feature?

blueeagle69, 2019-07-14:
@Powerprobot  
I just tried this for the hell of it.
I ran G28 followed by G9 S0, and my printer went into a loop of rehoming Z (BLTouch v3.1).
The only way out of it was to reset my board (SKR 1.3)
Dn't think I'll dso that again!

Vertabreak, 2020-01-19:
@boelle looks like a patch was merged for this.

closed 2020-01-24 by boelle

alexxy, 2020-01-30:
It was merged only for stm32 but not for lpc and others

zakdma, 2020-02-23:
@boelle please reopen this. PR specified added support only for STM32 not for LPC1768 on SKR 1.3 as described in ticket