MarlinFirmware/Marlin #11134: How to set probe expected vs end stop distance

Cyberwizzard, 2018-06-27:
### Description

I have a Z end-stop as well as a newly added Z probe. The end stop triggers 2 mm above the surface of the bed (depending on where Z is homed). The probe triggers with the hot end 0.5 mm above the bed.

Every now and then the auto bed levelling works (G29, 3-point), but most of the time it measures one or 2 corners and then suddenly prints "Probing failed".

If I move the Z end-stop to hit the bed (so essentially have a negative offset) and then do G29, it works.

I conclude the probing logic expects the probe to lower to a specific point (above the current 0 point on the Z axis perhaps?) and no further before giving up. Since I can not increase the range (or height) of the probe, can I tell Marlin its okay to go down to -1.0 mm on the Z axis to find the bed surface?

#### Additional Information

* This is not an M851 issue (which is set to -0.5 via Z_PROBE_OFFSET_FROM_EXTRUDER on bootup), nor is it an issue with Z_PROBE_OFFSET_RANGE_MIN (set to -6) or Z_PROBE_OFFSET_RANGE_MAX (set to 6), unless I misunderstood those settings.
* Since having both an end-stop and a probe seems a common configuration, am I simply missing an obvious setting in the Configuration.h?

Cyberwizzard, 2018-06-28:
To answer my own question: there is no such facility as it seems that the probing mechanism uses a standard move which is aborted once an end-stop triggers. This is fine when the probe fires before the Z-min end-stop is reached but in my case the aluminum bed combined with 2 protuding screw heads prevents me from lowering the probe to the point where this assumption is true.

For anyone who has this issue like me, I wrote the following hack to work around it:
```diff
diff --git a/Marlin/endstops.cpp b/Marlin/endstops.cpp
index 1b1cab0..40a1c80 100644
--- a/Marlin/endstops.cpp
+++ b/Marlin/endstops.cpp
@@ -496,7 +496,13 @@ void Endstops::update() {
           #if ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
             if (z_probe_enabled) UPDATE_ENDSTOP(Z, MIN);
           #else
-            UPDATE_ENDSTOP(Z, MIN);
+            // Hack Cyberwizzard: only update the Z-min end-stop when not using the Z probe (prevents the machine from stopping when hitting the endstop before the probe hits the bed)
+            #if ENABLED(Z_MIN_PROBE_ENDSTOP)
+              // This is the actual hack, only enabled when a probe is available
+              if(!z_probe_enabled) UPDATE_ENDSTOP(Z, MIN);
+            #else
+              UPDATE_ENDSTOP(Z, MIN);
+            #endif
           #endif
         #endif
       #endif
```

This essentially disables the Z-min end-stop when using the Z probe. But since probing is only allowed inside the bed area anyway, I reckon this poses little risk to my machine.

thinkyhead, 2018-06-28:
There might be a glitch in Marlin (`bugfix-x.x.x` branches) that stops the movement early when the probe hasn't triggered, but I need to look further. I've only seen it on a half-built SCARA, and I didn't complete the investigation. The solution that works for your setup is fine, and may be a clue to a more general solution.

Anyway, you should also give the bugfix branches a test-run to see if we already solved the issue for your case. We added the setting `Z_PROBE_LOW_POINT` to allow tuning where the probe should stop before it gives up and throws an error:
```cpp
#define Z_PROBE_LOW_POINT          -2 // Farthest distance below the trigger-point to go before stopping
```

Cyberwizzard, 2018-06-28:
The decision if it's a glitch or not depends on the desired behavior I guess. The current code (including the bugfix branch) seems to treat the Z-min end-stop as an absolute barrier; if the probe did not fire before hitting it, something went wrong. My change alters this: the end-stop is a method to determine the rough offset from the bed (few mm's at most) and then the probe is required (by means of an G29) to find the bed exactly.

An "G28 Z0" is then still very useful for things like start-up scripts which park the toolhead somewhere which requires a known Z position (even if its not enough to start printing right away).

Z_PROBE_LOW_POINT is also very nice to have, but I think disabling the Z-min end-stop during probing is required for it to be useful. Without this change, one could also just use the Z-min end-stop to limit the travel during probing using the code as it is right now :)

thinkyhead, 2018-06-28:
Ultimately, it's best not to have the Z endstop at a higher point than the probe trigger position. If possible, see if you can drop the endstop down to a height where the nozzle is just touching the bed.

Cyberwizzard, 2018-06-28:
Yes, like I explained, without a different probe, a different bed or different screws, that's not possible without causing a collision somewhere... so I figured a software fix is easier :)

thinkyhead, 2018-06-28:
Well, the Z endstop isn't _specifically_ or _intentionally_ being used as a safety stop. It's just a side-effect of the fact that it's still being checked during the probe (any towards Z-min) move. We might be amenable to applying your solution to remove the ambiguity, and if anyone really wants it, we could add (yet another) sub-option to also check the Z min endstop as a safety.

thinkyhead, 2018-06-28:
What the heck. Applied and merged.

closed 2018-06-28 by thinkyhead

AnHardt, 2018-06-28:
While having a "homing switch" above the level we want to "probe" makes perfect sense, a "endstop" does not.

thinkyhead, 2018-06-30:
I guess all we _really_ have is "homing switches," since that's the only function we've assigned to the switches mounted on the axes. But we still call them "endstops" because it's shorter.