MarlinFirmware/Marlin #11138: Layer height changing during print - Dual extruder

ramireseverton, 2018-06-28:
### Description

I am facing a sudden layer height change during a print, it starts printing normally then after some layers, every layer seems to be double the define in slicer.

### Steps to Reproduce

-Dual extruder head (E3DV6 Bowden, only one carriage)
-Using Cura 3.3.1
-Marlin bugfix 1.1
-Using Lin Advance
-Using UBL bed leveling
-CoreXY kinematics

-Extruders are not grinding neither missing steps.
-Only happens when using both extruders, machine works pretty fine when only one of them is used during the entire print.

[image]

thinkyhead, 2018-06-28:
This is a duplicate of an earlier issue. We've been trying to figure out the cause for some time now, but since none of us have the hardware to test on, it has not been solved yet.

To help narrow down the issue, try disabling the sub-options of UBL such as Z Fade (and so on) to see if there is anything that eliminates (or exacerbates) the bad behavior.

margaale, 2018-06-28:
I've been experiencing the same issue only with higher distance between layers. If you deactivate UBL all goes back to normal.

GypsyRobot, 2018-06-28:
Ok. I had some extra time to look for other reports: #7469 #7469 #10130 #10088 
All of them seems to be related with bed leveling correction. specially UBL. Some says forcing set a Z offset between extruders like M218 T1 Z-0.01 would be a workaround for the problem.
@Roxy-3D also says if I lift Extruder 0, change tools to Extruder 1 and them put it back down, it would also kind of solve the issue for now. I think this way is easier since I can set a Z-hop when retracting and force retracting at every tool change using my slicer (Cura).

I will try disabling z fade as well as the other options and then report here if at least one of them works.

Thanks guys!

thinkyhead, 2018-06-30:
The problem is probably related to the way that UBL handles updates to the current position internally. As I understand it, you have to issue some kind of move after any change to the position so that UBL gets the new position copied into its internal shadow registers.

boelle, 2019-02-20:
@GypsyRobot this is still an issue right?

GypsyRobot, 2019-02-20:
> @GypsyRobot this is still an issue right?

Unless this was addressed in version 1.1.9 (didn't have the time to test yet), it is still an issue.

boelle, 2019-02-20:
@margaale do you still see the problem?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

margaale, 2019-03-06:
Sorry, after this i disabled UBL and never used it again. So i can give you any feedback.

boelle, 2019-03-06:
@margaale we need @GypsyRobot to reply since he created this issue

margaale, 2019-03-06:
Sure, no problem, but i only answered your question...

closed 2019-05-16 by thinkyhead