MarlinFirmware/Marlin #11139: Q: adjust first layer height without re-doing G29

boelle, 2018-06-28:
So i got my Z travel spot on for the center of the bed

Then i did 10 G30's and made an Avg. probe offset in Z direction

I did the complete G29 and removed a few bad spots

now with heated nozzle its 0.3mm to high up in the center

are there a way to correct this without having to redo the G29?

i assume that my probe offset was wrong even thou i spend the extra time to get it right

forkoz, 2018-06-28:
Depends on the bed leveling you use, I change offset all the time on UBL and don't have to redo it at all. What will change it is heating the bed and heating nozzle if it also warms the sensor.

boelle, 2018-06-28:
i use UBL

so i just change Z_PROBE_OFFSET_FROM_EXTRUDER ?

i know heat changes things, but nozzle is still 0.3mm to high up

boelle, 2018-06-28:
i hope i'm correct, please check my logic:

#define Z_PROBE_OFFSET_FROM_EXTRUDER -6.106

is what i have now, my brain says that i should make that

#define Z_PROBE_OFFSET_FROM_EXTRUDER -5.806 

is that correct?

of course i can redo the G29 P1 routine but i want to avoid that since it takes forever to do

forkoz, 2018-06-28:
don't ever put z offset into the config file. set it with the LCD or M851. Hard coding z offset is a recipe for pain.

forkoz, 2018-06-28:
my UBL points are different at 60c, 70c, 80c, etc. not by .3 but by enough where the layer might not be squished. if you enable baby stepping in config_advanced you would just move z offset by .3 when printing. then if I like that offset I save it with M500 and that's my new offset. I only get it with a feeler gauge once or twice and then adjust from there. UBL points don't appear to be affected by Z itself though so I can take them at Z0 and Z1.x, etc... btw 5mm is quite a lot. your sensor must be pretty high above the nozzle.

boelle, 2018-06-28:
the probe just sits in an a bracket i made quickly and i did not adjust it

i know baby stepping, but i wanted to figure what number i got wrong since its 0.3 to high up

using baby stepping does not learn my head much, so if you can tell my what number i got wrong i would appreciate that very much

i might print a totally different mount for my e3d v6 (no clone) that has a integrated holder for the inductive probe, and at that time i know there is a small tool on thingiverse that i can use to adjust the probe in relation to the nozzle

but for now i just want to get the printer going again

boelle, 2018-06-28:
but you gave me an idea

i want to get the firmware settings right for PLA

and then for ABS i will use baby stepping. i hardly use ABS so it makes sense at least for me to have as many settings in the firmware so i dont forget them (i have a memory like a colander: full of small holes)

boelle, 2018-06-28:
but i assume that i simply got #define Z_PROBE_OFFSET_FROM_EXTRUDER wrong and have now adjusted it

time to test if it made the difference needed

boelle, 2018-06-28:
i tried both -6.406 and -5.806, it did not change anything, and yes i did also change it in eeprom

so i guess it only comes in to effect after another G29 run and the only temp fix is to use baby stepping, but that means i have to print and cant check with a feeler gaguge first

AnHardt, 2018-06-28:
Z-probe-offset is the distance between nozzle tip and trigger point of the probe. It should be used while probing, not when the probe data is used. It should also be used when the probe is used for homing. When homing to max, it should be logical, the Z-probe-offset, is not involved. You should change the length of the z-axis, or if available the correction for both of your z-max-endstops (homing switches) .

gloomyandy, 2018-06-28:
You can use G29 J to quickly reprobe the bed and adjust the mesh offset and any tilt. I tend to use G29 J3 (9 point probe) rather than the default 3 point. However if heating has changed things by 0.3mm then it may also have changed the shape of the bed significantly so your existing mesh may not be so good if it was created when the bed was cold.

boelle, 2018-06-28:
heating has not changed things, i simply got my probe offset wrong by 0.3mm

so i think i will adjust probe offset and do another G29 after dinner

boelle, 2018-06-28:
just so i dont do it wrong

#define Z_PROBE_OFFSET_FROM_EXTRUDER -6.106

to bring the bed 0.3mm closer to nozzle i would change it to 

#define Z_PROBE_OFFSET_FROM_EXTRUDER -5.806

right?

i know that my lenght of Z is correct because i printed a small 20x200 square but with UBL turned off

boelle, 2018-06-28:
think i will print that mount where the probe holder is integrated

my head cooked up an crazy idea, can you tell marlin what direction you want to home in?

my idea was to home to max to get the 2 screws in sync, and then use to probe to get down to the bed?

just an crazy thought, anyways i'm off to dinner but will close this one since i'm so close to getting it right and i dont want to take more space on the issue list than i have to

closed 2018-06-28 by boelle

boelle, 2018-06-28:
and i found some help here: https://www.3d-diy.de/tutorials/marlin-auto-bed-leveling-mit-ubl/

if i find that the probe is not calibrated right i can just shift the entire mesh up and down with  G29 P6 C  followed by either a positive value or a negative one

but i have of course corrected my probe offset so its near correct next time i do a full probe of the bed

and i also learned how to remove bad probe points, the mk42 bed is nice but it has 6 screws to hold it and 9 probe points. if a probe happens on top of a screw it will trigger before everything else since there is more inductance in it. the probe points have more copper in them than the rest of the bed but less inductance than the screws.

but there is a plugin for octoprint that can show it as a 3d height map and i just hover the mouse over the spikes and i get X and Y and can quickly invalidate those points. and then i use G29 P3.13 to find a reasonble avg. value for those spots. i have been suggested to just reprobe again but it would just get me the same bad results

what would have been super nice was if marlin could do just like prusa... search for the 9 spots and compare where it finds them to where they should be. where they should be is a known and all the user needs to make sure that the nozzle lands precise on top of 0.0 if told to do so

of course i could also just get a bl touch since that would not depend on inductance, but i have to make do with what i have.

Roxy-3D, 2018-06-29:
> Z-probe-offset is the distance between nozzle tip and trigger point of the probe. It should be used while probing, not when the probe data is used. 

Yes!   And this isn't very scientific.   But this works...     I set Z_PROBE_OFFSET_FROM_EXTRUDER as accurately as I can.  And then I probe the bed.   And using that information I do a very small print in the center of the bed (where I probed).    Depending on how the first layer is going down...   I adjust the Z_PROBE_OFFSET_FROM_EXTRUDER value.

I get that number right!!!   It usually takes 2 or 3 tweaks to get it correct.

Once that is correct....  And I can print something small where the Z-Axis is homed....   Then I do all the bed leveling setup.

But trying to bring up bed leveling with inaccuracies in the initial setup is doomed to failure.

boelle, 2018-06-29:
hehe, its getting the probe offset that caused me trouble, i had to "spank" my brain a bit to understand it

the offset was wrong by 0.3mm with a feeler gauge and i wanted to get it more correct before i moved on

so i lowered the offset and it became 0.6mm off, so i discovered that i had to increse the number which makes sense now... firmware would think that the probe is further away and move the nozzle closer

boelle, 2018-06-29:
oh, forgot to ask

it was suggested that i could use G29 J to tilt the mesh, is that a bad or good thing?

Roxy-3D, 2018-06-29:
Get your mesh fine tuned and correct...    If you have your mesh accurate, most likely you will decide you don't even need to mess around the J tilt options.

But J tilt does serve a purpose.   If you have a mesh that accurately describes your piece of glass, and you end up with a small piece of plastic under the glass...   The J tilt will handle that.

thinkyhead, 2018-06-29:
You can also try `M420 Cn` (same as `G29 P5 Cn`) to center the mesh on a given "mean" value.

yhaiovyi, 2018-12-27:
All right, so after reading this thread, I still don't understand a thing. So do you suggest that we use re-flashing firmware with different `Z_PROBE_OFFSET_FROM_EXTRUDER` value every time we want to use different first layer height? Because I tried to apply an offset `G29 T` shows it correct, but when I start printing it still uses the original value.