MarlinFirmware/Marlin #11140: echo:EEPROM version mismatch (EEPROM=V27 Marlin=V55) echo:Hardcoded Default Settings Loaded

bfrakes256, 2018-06-28:
Marlin bugfix2.0.x  using Arduino beta 1.9.0

Compiles fine and uploads to Mega 2560 board.

Pronterface shows:

echo:EEPROM version mismatch (EEPROM=V27 Marlin=V55)
echo:Hardcoded Default Settings Loaded

M501 repeats above

Using the LCD shows the correct values I've loaded into the configuration.h 

Please advise...

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

gloomyandy, 2018-06-28:
The message is simply telling you that the contents of your eeprom are from an earlier version of Marlin and that as a result it has loaded the default settings instead of data from the eeprom. The eeprom settings will not have been touched. However you will not be able to load them with the newer version of Marlin. Instead you need to write new settings to eeprom (using m500). After you have stored the new settings the version number of the eeprom will be updated and you will no longer get this message.

bfrakes256, 2018-06-28:
m501 only gave me the current.  So M500 will write what is in the firmware and then m501 will show the new setting?

Cool...

bfrakes256, 2018-06-28:
It has been many years since I've done much C++ programming and never to anything like these new devices.  I did much of my C++ on IBM mini's many moons ago.

closed 2018-06-28 by bfrakes256