MarlinFirmware/Marlin #11141: Compilation error if #define EXTRUDER_AUTO_FAN_SPEED not equal 255

svs072477, 2018-06-28:
Marlin 2.0
RURAMPS4D + Arduino DUE
if 
#define E0_AUTO_FAN_PIN 8
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED   255
compilation OK
if
#define EXTRUDER_AUTO_FAN_SPEED   100

compilation error
n file included from sketch\src\inc/MarlinConfig.h:37:0,

                 from sketch\src\Marlin.h:25,

                 from sketch\src\Marlin.cpp:31:

sketch\src\inc/SanityCheck.h:1368:45: error: expected primary-expression before ',' token

     static_assert(GET_TIMER(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);

                                             ^

thinkyhead, 2018-06-30:
`GET_TIMER` isn't yet implemented for DUE, so for now you'll have to leave the speed set to 255.

@p3p, @Bob-the-Kuhn — I'm surprised these are still dangling at this point in development. What should we do for these in `fastio_Due.h`?
```cpp
// check if pin is an input
#define _GET_INPUT(IO)
// check if pin is an output
#define _GET_OUTPUT(IO)
// check if pin is a timer
#define _GET_TIMER(IO)
```

p3p, 2018-07-01:
I'm not really familiar with that platform, I think DUE is a cortex-m3 so all gpio pins should be input/output capable, `_GET_INPUT`,` _GET_OUTPUT` could return true, `_GET_TIMER` I'm unsure.

ejtagle, 2018-07-01:
@p3p: Those macros are used to get the  direction the IO pin is currently set to, and they are used mostly for debugging purposes, except GET_TIMER(), that is used only for the Extruder fan...

thinkyhead, 2018-07-01:
`GET_TIMER` is just `bool` so it could simply return `true` if all pins are timer-capable.

thinkyhead, 2018-07-01:
I found this example to simply get the pin mode from the DDR…
```cpp
//
// Function to read state of mode bit in DDR register.
//
// If the pin number is invalid, return -1
//
// For valid pins:
//   return 0 if the pin is an input and 1 if it is an output.
//
// This corresponds to direction bits in the DDR.
// That is, INPUT is defined as 0 and OUTPUT is 1.
//
// Note that the digitalPinToBitMask and digitalPinToPort
// do not test for invalid pin numbers, so I compare with a
// constant defined by a conditional compile directive.
//
#include <pins_arduino.h>
int getPinMode(const int p) {
  constexpr int MAX_DIGITAL_PIN_NUMBER =
    #if defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
      69
    #else
      19
    #endif
  ;

   // Check valid pin number
   if (p > MAX_DIGITAL_PIN_NUMBER) return -1;

   // Convert designated Arduino pin to ATMega port and pin
  uint8_t pbit  = digitalPinToBitMask(p), pport = digitalPinToPort(p);

   // Read the ATmega DDR for this port
   volatile uint8_t *pmodereg = portModeRegister(pport);

   // Test the value of the bit for this pin, return TRUE for 1
   return (*pmodereg & pbit) ?  1 : 0;
}

//
// Test function to display DDR register bits for Arduino pins
//
void setup() {
   Serial.begin(9600);

   // Set up various pins and test I/O mode of all pins
   pinMode(13, OUTPUT);
   pinMode(3, OUTPUT);

   //
   // Now get mode values for the pins
   // If it's for an ATmega1280 or '2560, pins go up to 69
   //
   for (int pin = 0; pin < 21; pin++) {
       const int mode = getPinMode(pin);
       if (mode < 0) {
           Serial.print("Pin number "); Serial.print(pin); Serial.println(" is invalid.");
       }
       else {
           Serial.print("Mode of Pin "); Serial.print(pin); Serial.print(" is ");
           Serial.println(mode == INPUT ? "INPUT" : "OUTPUT");
       }
   }
}

void loop() { }
```

ejtagle, 2018-07-01:
For AVR, we are already accessing the DDR register.
For SAM, it should go something like this:

```
// check if pin is an input
#define _GET_INPUT(IO) ((g_APinDescription[IO].pPort->PIO_OSR & g_APinDescription[IO].ulPin) == 0)

// check if pin is an output
#define _GET_INPUT(IO) ((g_APinDescription[IO].pPort->PIO_OSR & g_APinDescription[IO].ulPin) != 0)
```

closed 2018-07-01 by thinkyhead