MarlinFirmware/Marlin #11143: Save last line/layer number in case of printer failure

ksya, 2018-06-29:
Last night I was printing from SD card. When I came back to check on the print, a thermal runaway has occured and the printer had stopped mid-print. Yesterday I had a similar issue, when I rebooted my PC, the USB was not powered anymore and as a result, the power supply turned off (this is a setting I think).

I have googled how to find the last location of the gcode, but I could not find it. The only way to resume the print is by counting the layers and editing the gcode, removing all code that is already processed.

To save me some time in the future, it would be great if Marlin would save the last line of the gcode it processed in the EEPROM. This makes resuming prints much easier.

marcio-ao, 2018-06-29:
@ksya: This has been discussed in several other threads. The problem is that EEPROM only has 100,000 write cycles before it is damaged and the board needs to be replaced. If you saved the location after each GCODE, you'ld need a new board after not so many prints.

There is some discussion [here](https://github.com/MarlinFirmware/Marlin/pull/10479)

thinkyhead, 2018-06-29:
If Marlin dies in the middle of a print, the host software should have some idea of the last line it was able to send successfully. In the case of an SD print, Marlin can use the `POWER_LOSS_RECOVERY` feature to resume a print that failed for unknown reasons. This feature is not yet released, but is in the "bugfix" branches and will be included in Marlin 1.1.9.

closed 2018-06-29 by thinkyhead

5a2v0, 2018-07-02:
thinkyhead i know that the "POWER_LOSS_RECOVERY" feature is not released yet, but i got marlin 2.0 bugfix on my clome prusa i3 and i tried it.

after a fail, on restart the printer notify me that there is an interrupted print and i can choose if resume or not.

if I choose the resume option, the printer warm up the hotbed and then the extruder.. after that it goes home with all axis (2 times x and y) and say "continuing" on the display, but it Will stay parked on x0 y0 forever ...
do you know why?

thinkyhead, 2018-07-02:
> do you know why?

There might still be a bug. The feature was patched a couple of days ago so maybe it's already fixed? We'll keep refining and testing if needed.