MarlinFirmware/Marlin #11149: Skew compensation and lcd error

Matias001998, 2018-06-29:
I have been trying to use Skew compensation in marlin with the lcd but i keep getting this message when compiling

`Arduino:1.8.5 (Windows 10), Tarjeta:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\ultralcd.cpp: In function 'void _lcd_toggle_bed_leveling()':

ultralcd.cpp:2560: error: 'set_bed_leveling_enabled' was not declared in this scope

     void _lcd_toggle_bed_leveling() { set_bed_leveling_enabled(!planner.leveling_active); }

                                                                                        ^

sketch\ultralcd.cpp: In function 'void lcd_prepare_menu()':

ultralcd.cpp:2688: error: 'leveling_is_valid' was not declared in this scope

       if (leveling_is_valid()) {

                             ^

exit status 1
'set_bed_leveling_enabled' was not declared in this scope
`

It looks like i need to have bed leveling activated to use this feature, but i dont have auto bed leveling in my machine, so i dont know if it is a bug or it has a simple fix.
(sorry if my english is not very good but its not my native language)

thinkyhead, 2018-06-30:
Try this change to `ultralcd.cpp`:
```diff
- #if ENABLED(LCD_BED_LEVELING) || (PLANNER_LEVELING && DISABLED(SLIM_LCD_MENUS))
+ #if ENABLED(LCD_BED_LEVELING) || (HAS_LEVELING && DISABLED(SLIM_LCD_MENUS))
```

Matias001998, 2018-06-30:
Note: i went back and get
 `#if ENABLED(LCD_BED_LEVELING) || (PLANNER_LEVELING && DISABLED(SLIM_LCD_MENUS))`
back, and enabled SLIM_LCD_MENUS and it compiled

thinkyhead, 2018-06-30:
Try both items from my comment above. `SLIM_LCD_MENUS` only conceals the error.

thinkyhead, 2018-06-30:
Now patched in both bugfix branches.

closed 2018-06-30 by thinkyhead