MarlinFirmware/Marlin #11157: Auto Temp

3Dmaniac, 2018-06-30:
Hi guys,

In the display options (Control>temperature), i see an option called Auto temp.
In it i can see a max temp value, min temp value and a factor value.
Turning it on i see no changes. Starting a print i see no changes.

What does this feature do?

Thanks

3Dmaniac, 2018-06-30:
Hi AnHardt,

thanks for the answer.

So, if i turn the auto temp feature on, it will override the PID routine?
And this will work without any modification on the print gcode as long as the print gcode temperature is set between min temp and max temp?

Thanks

thinkyhead, 2018-06-30:
The PID still runs, but the temperature target is adjusted dynamically also.

thinkyhead, 2018-06-30:
Did you also add `M109 F<factor> S<min> B<max>` to your startup G-code? It has to come after all other `M109` commands or auto-temp will be canceled.

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead