MarlinFirmware/Marlin #11158: Hardware Y endstops opposite homing direction not respected

hexane360, 2018-06-30:
### Description

My printer has two hardware Y endstops, and it homes to `Y_MAX`. However, the planner ignores any `Y_MIN` triggers. This is *not* a pin or electrical issue, as M119 shows all endstops working correctly.

I believe the problem is in `Endstops::update()`.
[This](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/endstops.cpp#L434-L458) is the code for X endstops:
```
  #if HAS_X_MAX
    #if ENABLED(X_DUAL_ENDSTOPS) && X_HOME_DIR > 0
      UPDATE_ENDSTOP_BIT(X, MAX);
      #if HAS_X2_MAX
        UPDATE_ENDSTOP_BIT(X2, MAX);
      #else
        COPY_LIVE_STATE(X_MAX, X2_MAX);
      #endif
    #else
      UPDATE_ENDSTOP_BIT(X, MAX);
    #endif
  #endif
```

However, [this](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/endstops.cpp#L460-L484) is how Y endstops are handled:
```
  #if HAS_Y_MAX && Y_HOME_DIR > 0
    #if ENABLED(Y_DUAL_ENDSTOPS)
      UPDATE_ENDSTOP_BIT(Y, MAX);
      #if HAS_Y2_MAX
        UPDATE_ENDSTOP_BIT(Y2, MAX);
      #else
        COPY_LIVE_STATE(Y_MAX, Y2_MAX);
      #endif
    #else
      UPDATE_ENDSTOP_BIT(Y, MAX);
    #endif
#endif
```

This includes an extra check, only checking `Y_MAX` if that's the homing direction.

### Steps to Reproduce

1. Enable two probes on the Y axis
2. Enable hardware endstops (M120)
3. Disable software endstops (M211 S0)
4. Move into endstop opposite homing direction

**Expected behavior:** Movement stops and trigger is reported

**Actual behavior:** Planner continues to step into endstop

#### Additional Information
The following are some relevant configuration settings:
Configuration.h:
```
#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
//#define USE_XMAX_PLUG
#define USE_YMAX_PLUG
#define USE_ZMAX_PLUG
```
```
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
```
```
#define Z_MIN_PROBE_ENDSTOP
#define Z_MIN_PROBE_PIN 22
```
```
#define X_HOME_DIR -1
#define Y_HOME_DIR  1
#define Z_HOME_DIR -1
```
```
// Min software endstops constrain movement within minimum coordinate bounds
#define MIN_SOFTWARE_ENDSTOPS
#if ENABLED(MIN_SOFTWARE_ENDSTOPS)
  //#define MIN_SOFTWARE_ENDSTOP_X
  //#define MIN_SOFTWARE_ENDSTOP_Y
  #define MIN_SOFTWARE_ENDSTOP_Z
#endif

// Max software endstops constrain movement within maximum coordinate bounds
#define MAX_SOFTWARE_ENDSTOPS
#if ENABLED(MAX_SOFTWARE_ENDSTOPS)
  #define MAX_SOFTWARE_ENDSTOP_X
  //#define MAX_SOFTWARE_ENDSTOP_Y
  #define MAX_SOFTWARE_ENDSTOP_Z
#endif
```
Configuration_adv.h:
```
#define ENDSTOPS_ALWAYS_ON_DEFAULT
```
```
//#define X_DUAL_STEPPER_DRIVERS
//#define Y_DUAL_STEPPER_DRIVERS
//#define Z_DUAL_STEPPER_DRIVERS
```

hexane360, 2018-06-30:
The same problem exists for Z opposite the homing direction.

thinkyhead, 2018-06-30:
For endstop checking during moves, Marlin has only ever had (intentional) consideration for a single end of the axis, as defined by the HOME_DIR options. It should only be updating and responding to endstops when the axis is moving in that direction.

thinkyhead, 2018-06-30:
@ejtagle — Do we want to extend Marlin to be able to handle endstops at both ends of an axis? It would require a bit of work to do so, and I suppose would have to be added as a 2.0-only feature at this point.

hexane360, 2018-06-30:
That's interesting. Either way, the code needs to be changed. If the expected behavior is to only respect the 3 endstops used for homing, then the current X-axis code needs to be changed. If all configured endstops are to be respected while moving in that direction, then the current Y-axis and Z-axis code needs to be changed.

It's not clear to me why homing direction should impact endstop behavior during regular print moves. However, if that is the expected behavior, then `USE_*_PLUG` is redundant and can be inferred from `*_HOME_DIR` instead. If 3 endstops will always be ignored, why is there even support for them?

ejtagle, 2018-06-30:
@thinkyhead : Even if it could initially be thought it makes sense to support this, if you think carefully it does not:

Let´s assume we make all endstops be respected at all times, ignoring the axis movement direction: Ok, as soon as an endstop triggers, then the motor will stop (that is not a problem)
But, you must somehow determine the direction of movement that will "untrigger" the triggered endstop. So, Marlin must know the association between endstops and directions, otherwise, it is not possible to be sure if a given movement will force against the endstop or will untrigger it.

You can surely implement some kind of autodetection, move an axis, for example, when homing, and just assume the triggered endstop is the MIN or the MAX of that axis. But eventually you risk swapping the maximum and the minimum points of that axis, if endstops are not properly configured or connected.

I am not a fan of this. User must properly wire endstops and must properly configure them. Endstops are there to avoid Marlin driving into the associated direction, not to prevent bad configurations. 

The best we could do is to kill() with a message "Badly configured/wired endstops" as message, because from the firmware perspective, you simply don´t know if it is a bad configuration, bad wiring, or desired feature to "swap axis directions" if an improper limit swith is triggered.

1)

thinkyhead, 2018-06-30:
> Endstops are there to avoid Marlin driving into the associated direction, not to prevent bad configurations.

Well, the one place where it might matter is with the `ABORT_ON_ENDSTOP_HIT_FEATURE_ENABLED`, which will kill an SD print whenever an endstop collision is detected. But I agree, in all cases the endstop switches should only cause move interruption when the axis is moving _towards_ the given endstop.

> …if that is the expected behavior, then USE_*_PLUG is redundant…

Those options were added to be able to specify which endstop plug pins are in use for any switch-like purpose, not specifically for use as endstops or homing. At the time they were added, the mere existence of an endstop pin would cause certain parts of Marlin to be compiled in. They were added as a way to say "don't bother with some of those pins."

hexane360, 2018-06-30:
Wait, now I'm confused.

The Y_MIN endstop is not being respected while moving *towards* the Y_MIN endstop. It is currently doing *nothing*, and the printer is fine ramming the axis into the frame.

The question is "should more than 3 endstops ever be checked", not "should all endstops always be checked regardless of movement direction".

ejtagle, 2018-06-30:
@hexane360 : Your printer homes to Y_MAX... Why should motors go and trigger Y_MIN when homing to Y_MAX ?

hexane360, 2018-06-30:
I'm not concerned about behavior during homing, I'm concerned about behavior during regular print moves. I home Y (to max), disable software endstops, then move `G1 Y-10`. The printer rams into the frame and the triggered endstop is ignored.

This behavior seemed to change with commit #11013.

ejtagle, 2018-06-30:
I think the

thinkyhead, 2018-06-30:
I merged #11013 with full cognizance that it might come up later. Now the question is, given the latest changes to the endstop code, can it be reverted without causing unwanted side-effects?

ejtagle, 2018-06-30:
The UPDATE_ part of the endstops::update() needs to continuously poll, without regarding homing directions, or ```ENDSTOPS_ALWAYS_ON_DEFAULT``` will never work as expected (should kill the print if an endstop is triggered, and the homing direction must not be taken to disregard that decision)

thinkyhead, 2018-06-30:
Let the chips fall where they may, I've reverted #11013 (both branches).

thinkyhead, 2018-06-30:
Another issue was pointed out today. If an endstop move completes successfully without ever triggering an endstop, it indicates a problem, so we should call `kill` in that case. It would require an extra mode of `endstops.enabled` so it can distinguish between moves that should throw an error when they _**don't**_ trigger an endstop, and moves that should _**only**_ abort when they _**do**_ hit an endstop.

ejtagle, 2018-06-30:
Yes, in that case, the motion block completed without interruption. You can easily detect that by inspection of endstops.trigger_state()

thinkyhead, 2018-06-30:
Because of the way sensorless homing works, this feature would need to be implemented as part of the stepper ISR. Basically, at the end of every block check if the move was supposed to trigger an endstop, and if it was then we know (1) the move wasn't aborted so (2) it should call `kill` from there.

ejtagle, 2018-06-30:
As stated previously, you already have that information on endstops.hit_state(), even for sensorless homing

thinkyhead, 2018-06-30:
If the move wasn't aborted, we already know it failed to trigger an endstop, so we don't actually need to examine the trigger-state.

ejtagle, 2018-06-30:
The condition can be added to 
```do_probe_move()```
and 
`` do_homing_move()```

just before the call to endstops.hit_on_purpose()

ejtagle, 2018-06-30:
We are talking about the same, but there is no need to add an extra check to the stepper. If the movement was not aborted, endstops.state() will be 0
Otherwise, it will contain the endstop or probe that aborted the movement

thinkyhead, 2018-06-30:
(A single back-tick is sufficient to specify `inline code`.)

thinkyhead, 2018-06-30:
I guess my question is, whether to add the check to all the functions used by `G28` and `G29` or whether to just add a single check to the stepper ISR and add a new mode to `endstops.enabled`. If I can rely on the high-level `hit_state` then it would be simpler (though probably more code) to add the checks to the functions used by `G28` and `G29`.

ejtagle, 2018-06-30:
I would not add a call to kill() from the ISR: 
1) That forces the compiler to save the whole runtime environment (including floating point) even if kill() is mostly never called
2) There is no warranty from the ISR that the LCD is usable at all. It depends on if the ISR was called when the LCD was updated or not (very likely situation with the ST7290... ;( )

All GCODES related to homing end calling do_probe_move() or do_homing_move(), so those are the only functions to modify. And you could factor out a function that does exactly that check and then calls endstops.hit_on_purpose()

thinkyhead, 2018-06-30:
PRs posted: #11160, #11161

closed 2018-06-30 by thinkyhead