MarlinFirmware/Marlin #11159: bilinear bed leveling bugfix-2.0.x

KA8888, 2018-06-30:
Hello,
yesterday I configured and uploaded the latest 2.0 bugfix on my Anycubic Kossel Pulley. I have enabled bilinear bed leveling and since then, the grid looks weird. I recorded a video to show what I mean.
https://youtu.be/-a3l5GjMWoE
I also drew the leveling points on a sheet of paper.
[image]
There are some missing points on the outside.

Delta auto calibration does not work either. It is messuring only 5 points in the middle and two points outside. After that, the hotend rises about 150mm up and stops completely without any message on the LCD.

Both worked for me on 1.1.8

[MarlinConfig.zip](https://github.com/MarlinFirmware/Marlin/files/2152143/MarlinConfig.zip)

thinkyhead, 2018-06-30:
Try playing with the `MIN_PROBE_EDGE` to see how it affects probing.

This seems unusual:
```cpp
#define X_PROBE_OFFSET_FROM_EXTRUDER   0.0   // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER   0.0   // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -17.20  // Z offset: -below +above  [the nozzle]
```
When the nozzle itself is the probe, usually the `Z_PROBE_OFFSET_FROM_EXTRUDER` is a very small positive number. A large negative number implies that there is a mechanical or inductive probe, and it would be very strange if that had a 0,0 XY offset.

KA8888, 2018-06-30:
I set MIN_PROBE_EDGE to 5 and 10, it hat no effect on it.

Anycubic uses a microswitch with magnetic connection.
[image]
Its centered under the nozzle. My vid wasnt that good, sorry for that.

thinkyhead, 2018-07-01:
If the microswitch is triggered when the probe touches the bed then your `Z_PROBE_OFFSET_FROM_EXTRUDER` should be set to a small positive number.

KA8888, 2018-07-01:
The probe is below the nozzle, if it triggers, the nozzle is at Z17.2. This means it has to be a negative number, because it gets subtracted from the `position.
`#define Z_PROBE_OFFSET_FROM_EXTRUDER -17.20  // Z offset: -below +above  [the nozzle]`
Negative if the triggerpoint of the probe is below and positive if its above the nozzle.
[image]

The problem with the gridpoints doesnt seem to be because of my configuration. I configured everythink centered and MIN_PROBE_EDGE on all sides the same. It should at least probe the same amount on points at every side. And if i set GRID_MAX_POINTS_X to 7 it shouldnt matter what MIN_PROBE_EDGE i choose. Its always has to probe the 7 points (except on the corners), only the distance between them is changing depending on the MIN_PROBE_EDGE value.

I think its a bug in the leveling code.

thinkyhead, 2018-07-03:
Let's do some logging:

- Download Marlin `bugfix-1.1.x` or `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Copy the log output into a **.TXT file** and attach it to your next reply.

KA8888, 2018-07-03:
I just downloaded the latest 2.0 bugfix and noticed, theres a change in the pins_TRIGORILLA_14.h file.
In line 36 there is this code:
`#include "pins_RAMPS_14.h"`

As soon as i try to compile, im getting this error:
```
In file included from sketch\src\HAL\HAL_AVR\../../inc/../pins/pins.h:142:0,

                 from sketch\src\HAL\HAL_AVR\../../inc/MarlinConfig.h:30,

                 from sketch\src\HAL\HAL_AVR\HAL.cpp:29:

sketch\src\HAL\HAL_AVR\../../inc/../pins/pins_TRIGORILLA_14.h:36:27: fatal error: pins_RAMPS_14.h: No such file or directory

 #include "pins_RAMPS_14.h"

                           ^

compilation terminated.

exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino Mega or Mega 2560.
```

Anyway, i did that debuging stuff with the bugfix 2.0 from 2 days ago:
[bed leveling.txt](https://github.com/MarlinFirmware/Marlin/files/2158524/bed.leveling.txt)

One final question:
My Z Probe End Script is `G1 X0 Y0 Z200/nM500`. Is it storing the settings if i do it like this? Or do I still have to execute the M500 G-code manually?

Thank you in advance.

KA8888, 2018-07-03:
Now I tried the bugfix1.1 with the config files from the bugfix2.0 and got the same results. There are still to much probing points missing.

At the next step i compiled my old 1.1.8 version. Its working like it should, heres a picture of my probing points and the txt file:
[image]
[bed leveling 1.1.txt](https://github.com/MarlinFirmware/Marlin/files/2158934/bed.leveling.1.1.txt)

The last think I tried is my old 1.1.8 Version with the config files of my bugfix2.0. The only changes i done was to deactivate the nozzle park and the filament change feature. These caused an error while compiling.
The result: bed leveling worked as it should

**This means, that my configuration is fine. There must be a change between 1.1.8 and the bugfix1.1/bugfix2.0 that causing these issues.
[1.1.8 with bugfix2.0 config files.txt](https://github.com/MarlinFirmware/Marlin/files/2158996/1.1.8.with.bugfix2.0.config.files.txt)**

thinkyhead, 2018-07-24:
Here's your issue:
```diff
- #define LEFT_PROBE_BED_POSITION -(DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE))
+ #define LEFT_PROBE_BED_POSITION -(DELTA_PRINTABLE_RADIUS + MIN_PROBE_EDGE)
  #define RIGHT_PROBE_BED_POSITION DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE)
  #define LEFT_PROBE_BED_POSITION -(DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE))
  #define RIGHT_PROBE_BED_POSITION DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE)
- #define FRONT_PROBE_BED_POSITION -(DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE))
+ #define FRONT_PROBE_BED_POSITION -(DELTA_PRINTABLE_RADIUS + MIN_PROBE_EDGE)
  #define BACK_PROBE_BED_POSITION DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE)
```

thinkyhead, 2018-07-24:
Here's your issue:
```diff
- #define LEFT_PROBE_BED_POSITION -(DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE))
+ #define LEFT_PROBE_BED_POSITION -(DELTA_PRINTABLE_RADIUS + MIN_PROBE_EDGE)
  #define RIGHT_PROBE_BED_POSITION DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE)
- #define FRONT_PROBE_BED_POSITION -(DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE))
+ #define FRONT_PROBE_BED_POSITION -(DELTA_PRINTABLE_RADIUS + MIN_PROBE_EDGE)
  #define BACK_PROBE_BED_POSITION DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE)
```

thinkyhead, 2018-07-24:
Testing… should have a solution soon…

thinkyhead, 2018-07-24:
I was just doing a compile with your 2.0.x configuration and I can't reproduce the issue. The grid correctly goes from -80 up to +80 in X and Y.

For some reason your logs show strange grids.
```
For 1.1.8 the grid is: -75,-50,-25, 0,25,50,75    (25 apart centered on  0)
For 2.0.x the grid is: -80,-54,-28,-2,24,50,76    (26 apart centered on -2)
```
I don't get these results when I compile with the 2.0.x configs you posted. I'm not sure why the 1.1.8 build is choosing to use a larger (15mm) margin, or why your 2.0.x build is only going as far as 76 in the positive X direction.

Try the latest `bugfix-2.0.x` code and see if the issue still exists for you.

closed 2019-01-11 by KA8888