MarlinFirmware/Marlin #11162: Mesh bed leveling way off

airexes, 2018-07-01:
Using marlin 1.1.8 and a cr10s. Having issues with the manual mesh bed leveling on my bed with a mirror tile attached. I had the printer working fantastic before I changed the hotend mount and made a few firmware changes to compensate. I have made sure the the hotend is level with the crossbar.

I can repeatedly level the corners without issue, however when I start to perform the manual mesh bed leveling everything goes downhill. The first position already needs about -.05 adjustment. The right corner needs about +.2, the middle about -.2 meaning that is a 0.4mm difference . The most deviation I normally saw was 0.075.

So basically when leveling, the left of the bed now has the nozzle placed too high and the right way too low. 

When I actually go to print the left of the bed prints a lot lower then the right. The right side the filament barely sticks. I can feel the z axis compensating like always. 

I then go to check that the corners are still level and they check out perfectly.

forkoz, 2018-07-01:
Print your mesh. Maybe you have some mechanical issue now. Nothing about 1.8.8 has changed in a while.

thinkyhead, 2018-07-01:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

airexes, 2018-07-07:
I have tested with the 1.1.x bugfix branch and there was no difference. I went through and only chose a few basic settings I needed to perform the test. I really didn't feel like it was a software issue but didn't know where else to turn.

I've just verified it can't be the z axis by both replacing the z switch and parking the nozzle at a corner and verifying it's level. I raised the z axis 200mm and back down again and it was still perfectly level on that corner.

airexes, 2018-07-07:
Okay, I replaced the motherboard with a stock Cr10S one. On the stock firmware for some reason I couldn't find mesh bed leveling, but it did have corner leveling utility. I ran it, leveled the corners then homed the machine.
Everything was as it should be for the first time. I unplugged the printer to see if it would stay level and sure enough it did.

Opened marlin 1.1.8, flashed it. Back to the same problem in exactly the same way. When running the corner leveling utility Left is too high right is too low.