MarlinFirmware/Marlin #11164: TMC2130 + MKS Gen 1.4: No stepper movement from marlin

JakeBoothy, 2018-07-01:
Hey there, 

I'm having an issue getting 2 TMC2130's working on my X and Y axis. I have a Folgertech FT-5 R1.

Right now everything seems to be configured and wired correctly but there is no movement from the stepper motors from marlin. The motors are hissing very faintly when I give them commands, I only realized this when I took off the belt, as the sound was so faint that I couldn't hear it. 

When I run the Simple example in the TMC2130 library the motors move correctly, and I've even run the Software_SPI example and the calibrateSpreadCycle example and they both work fine. There are a few similar issues (https://github.com/MarlinFirmware/Marlin/issues/10320 and https://github.com/MarlinFirmware/Marlin/issues/9058), although I should mention that I am running the default power supply on the FT-5 and I don't think I am able to control it, it isn't configured to anyway. 

I'm pretty much out of ideas at this point. Could it be the same power supply issues? I wanted to add a delay before initializing the tmc2130 drivers but I could not find where to put it. I'll add my M122 output and my config_adv.h at the end. I have my CS pins set to pin 57 and 58 which I don't think are used anywhere else. I've tried a few different configurations and versions of marlin, right now I made a new config on the latest bugfix version. Thank you for your help.

M122 output is:

		X	Y
		Enabled		false	false
		Set current	1000	1000
		RMS current	994	994
		MAX current	1402	1402
		Run current	17/31	17/31
		Hold current	8/31	8/31
		CS actual		10/31	0/31
		PWM scale		0	0
		vsense		0=.325	0=.325
		stealthChop	false	false
		msteps		256	256
		tstep		1048575	1048575
		pwm
		threshold		0	0
		[mm/s]		-	-
		OT prewarn	false	false
		OT prewarn has
		been triggered	false	false
		off time		3	0
		blank time	24	16
		hysteresis
		-end		-2	-3
		-start		1	1
		Stallguard thrs	0	0
		DRVSTATUS	X	Y
		stallguard		X
		sg_result		282	0
		fsactive
		stst		X	X
		olb			X
		ola			X
		s2gb
		s2ga
		otpw
		ot
		Driver registers:	X = 0x81:0A:01:1A
			Y = 0xE0:00:00:00

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/2152352/Config.zip)

MikeLud, 2018-07-01:
@JakeBoothy 
One think I notice is the vsense looks off in M122 output. It looks correct in your Configuration_adv.h file. Maybe try M502 to restore your settings
M122 
vsense 0=.325	0=.325
Configuration_adv.h
#define R_SENSE           0.11  // R_sense resistor for SilentStepStick2130

teemuatlut, 2018-07-01:
What's your library version?
Either downgrade to 2.3.0 or upgrade to 2.4.3.

closed 2018-07-01 by JakeBoothy