MarlinFirmware/Marlin #11166: [2.0.x] custom startup code/ pin initialisation

TheLongAndOnly, 2018-07-01:
I need to set some pins permanently to high to configure my RADDS Extension Board properly. Is there any recommended way to do this in Marlin?

thinkyhead, 2018-07-01:
Add to `setup()` in `Marlin.cpp`…
```cpp
OUT_WRITE(MY_PIN, HIGH);
```

TheLongAndOnly, 2018-07-01:
Wouldn't this be a kind of "brute force" solution?
What about this option: Add the option to create a board_init() function and a define BOARD_HAS_INIT. When enabled, the board_init() function is called from setup()...

iosonopersia, 2018-07-01:
But you still would have to manually edit Marlin code to insert those 'OUT_WRITE(MY_PIN, HIGH);' lines... I don't see the point...

TheLongAndOnly, 2018-07-01:
Sure, but it could be moved to the correct context. Personally I don't think hat a core file should be edited for configuration/setup purposes. I came to the conclusion, that a function would not work, als the boards have no cpp yet. But it could be done through a BOARD_INIT macro.

xC0000005, 2018-07-02:
Do these need to initialize after Marlin is running? In the malyan boards with pullup transistors for USB, I turn them on in a variant initi function in the board definition.

thinkyhead, 2018-07-03:
> Personally I don't think that a core file should be edited for configuration/setup purposes.

If it's just for your own setup then it's fine. If it's for everyone, then #11170 is better.

closed 2018-07-03 by thinkyhead