MarlinFirmware/Marlin #11167: Amax value

3Dmaniac, 2018-07-01:
Hi guys,

So, checking the acceleration menu (Control>Motion>Acceleration) i can see several acceleration settings.

Acceleration
A-travel
Amax X,Y,Z,E

Reading the configuration.h it says:

Acceleration=X, Y, Z and E acceleration for printing moves.
A-travel= X, Y, Z acceleration for travel (non printing) moves.
Amax=Maximum start speed for accelerated moves.

It's not very clear for me what Amax is.
So...Amax value is a velocity speed or acceleration?
And it's for printing moves or non printing moves?

Thanks

tig33r, 2018-07-01:
This is probably jerk value...

3Dmaniac, 2018-07-01:
Thanks for the reply.
Actually it's not jerk.
Jerk menu is separate from the acceleration menu with different values.

gloomyandy, 2018-07-01:
I think that Amax is what it says, the maximum acceleration allowed by the machine for the particular axis. The other values are the "default" acceleration used for the various types of move (and should probably always be <= the corresponding Amax value). .

hexane360, 2018-07-02:
Basically, if you use M204 or the LCD to set an acceleration higher than Amax, the planner will cap it to Amax. Note that this is done on a per-axis basis while acceleration is set based on move type.

The relevant code starts about [here](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/planner.cpp#L2078) in `planner.cpp`.

closed 2018-07-03 by thinkyhead