MarlinFirmware/Marlin #11173: TMC2130 Sensorless Homing Trouble

rodrigosclosa, 2018-07-01:
How are you guys, all right?

I'm in a dilemma for setting up the sensorless homing with the TMC2130 drivers, with Ramps 1.4 and Arduino Mega. I bought these from china already in the SPI-enabled version in the hope of being easy to configure the sensorless homing.

I already used a TMC2130 before I bought it without being SPI and had to make the modifications to make it work with SPI (I followed the tips of topics here).
I bought these for not having been able to enable the sensorless homing in the old as well. This version is sold on Aliexpress:

https://en.aliexpress.com/item/4X-MKS-TMC2130-V1-1-To-Stepstick-Stepper-Motor-Driver-de-Fun-o-SPI-with-Dissipador/32850172695.html?spm= a2g0s.9042311.0.0.38b2b90a7GC1In

They already have the CFG4 and CFG5 with a small resistor, without the SPI resistor and already with the SPI pins welded up. I just added a pin to DIAG1 (which is the pad closest to the potentiometer).

Everything works great, Stealthchop, SpreadCycle, Interpolation .. But the sensorless homing I can not activate at all!

I tested it with Marlin versions 1.1.8, bugfix-1.1.x and bugfix-2.0.x. In none of them can I activate sensorless homing detection.

My M122 comes out as follows:

```
		X	Y	Z	E0
Enabled		false	false	false	false
Set current	500	400	650	500
RMS current	489	397	642	489
MAX current	689	560	905	689
Run current	15/31	12/31	20/31	15/31
Hold current	7/31	6/31	10/31	7/31
CS actual		7/31	6/31	10/31	7/31
PWM scale	255	114	48	7
vsense		1=.18	1=.18	1=.18	1=.18
stealthChop	true	true	true	true
msteps		16	16	16	16
tstep		1048575	1048575	1048575	1048575
pwm
threshold		79	98	164	173
[mm/s]		100.08	100.85	3.01	30.07
OT prewarn	false	false	false	false
OT prewarn has
been triggered	false	false	false	false
off time		5	5	5	5
blank time	24	24	24	24
hysterisis
-end		2	2	2	2
-start		3	3	3	3
Stallguard thrs	8	8	0	0
DRVSTATUS	X	Y	Z	E0
stallguard					
sg_result		0	0	0	0
fsactive					
stst		X	X	X	X
olb		X			
ola		X			
s2gb					
s2ga					
otpw					
ot					
Driver registers:
	X = 0xE0:07:00:00
	Y = 0x80:06:00:00
	Z = 0x80:0A:00:00
	E0 = 0x80:07:00:00

ok
```

My endstops and TMC configuration looks like this:

```
echo:Stepper driver current:
echo:  M906 X 500 Y 400 Z 650 E0 500
echo:Sensorless homing threshold:
echo:  M914 X8 Y8
ok

#define X_MIN_ENDSTOP_INVERTING true
#define Y_MIN_ENDSTOP_INVERTING true
```

I did a test running the StallGuard TMC2130 Sketch and I connected an LED to the DIAG1 and the GND of the Ramps, but when the motor starts to turn the LED is already on and the terminal does not change any value even if I put some pressure on the motor shaft. Even if I change the STALL_VALUE to 63 the same happens. Any value I put in STALL_VALUE does not affect the behavior at all.

I have now updated the Arduino lib from the TMC2130 to version 2.4.3 and also had the same behavior.

Does anyone have any light or tips to give me what else can I test?

Thanks in advance for your help.

[Configuration bugfix-1.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/2153052/Configuration.bugfix-1.1.x.zip)

Thank you,
Rodrigo Sclosa.

govido, 2018-07-01:
Hi, I run my steppers at STALL_VALUE 1. If you go higher than 8 it will be less likely to trigger. So you should test values like 1, 0 or even -10 or so.
The rest sounds ok to me. As long as the Diag1 pin is connected to X-min endstop pin on the board (and Y) it should work.

rodrigosclosa, 2018-07-01:
@govido tks for your reply. 

I've tried to tune up rising from 0 to +63 and from 0 to -64. But either no luck.
I'll try to put just 1 to test.

One strange thing I noticed in the bugfix-1.1.x branch is that the direction of rotation of the motor is changed if I use negative value. Without touching anything else on the firmware configuration, only the sensitivity of the sensorless homing. How can this be possible?

govido, 2018-07-02:
That's interesting... From my own experience with the latest bugfix versions of Marlin and Tmc2130 I would try your config first with the proven 1.1.8 version. That supports all you need for tmc drivers and works stable.

wbarber69, 2018-07-02:
You’re lucky I had to revert to an April build of bugfix as I tried to install today’s bugfix and I got absolutely no working tmc2130s. And it definitely has to do with sh as the only axis that will move for me is the one that doesn’t use sh, my z axis.  Also I noticed that you said it would go the wrong way with negative values, that to me points to an issue where if you use the dish pins to the endstops but don’t declare sensorless homing in the firmware it will sometimes do that on homing moves.  Make sure you have sensorless_homing defined not just the values for the sensitivity.

teemuatlut, 2018-07-02:
First try to get the stallGuard example working.
I just tried it with a FYSETC v1.1 driver and the example does work for sure.
What does the serial output look like?

rodrigosclosa, 2018-07-02:
@teemuatlut Tks for your reply.

I could not make the example work properly. I have set up the SPI pins correctly and my engine only makes a noise and does not spin, even if I type + several times in the terminal to increase the speed.

Simple example of the library I have been able to perform, the motor turns from side to side normally.

To make the StallGuard instance work, I copied the part from rotating the Simple Example engine.
But any value I put in STALL_VALUE does not affect anything. The Led is lit all the time.

What did you need to do to make it work?

teemuatlut, 2018-07-02:
I didn't _need_ to do anything but I did change `STALL_VALUE` to 10 so that I could see coolStep working properly. Essentially the example works as is.

rodrigosclosa, 2018-07-02:
Strange, because my example does not work. But the TMC drivers are working perfectly on the printer, I already printed several parts with them .. I just wanted to activate even Sensorless Homing ...

teemuatlut, 2018-07-02:
Did you change _anything_ in the example before uploading?

rodrigosclosa, 2018-07-02:
Only SPI pins because i'm using a custom CS PIN in the AUX2 interface...

rodrigosclosa, 2018-07-03:
Hi guys! 

I tried again to open the StallGuard example, just changed the SPI pins to hit my Y-axis pins on my Ramps (see attached).

The motor only locks the shaft and does not turn to either side. Not even if I type the + in the terminal or -, if I type 0 it releases the axis and with 1 it locks again. If I push the shaft, the connected LED of the DIAG1 does not change, it remains off.

The output of the terminal is this:

Start ...
0 1023 979
0 1023 979
0 1023 979
0 1023 979
0 1023 979
0 1023 979
0 1023 979
0 1023 979
0 1023 979
0 1023 979
0 1023 979
0 1023 979

And keeps going....

Does anyone know what could be wrong?
The drivers work perfectly with Marlin, I'm printing with them. But Sensorless Homing can not enable ..

Attached are also my Marlin Configuration and Configuration_adv files bugfix_2.0.x.

Tks,
Rodrigo.
[Files.zip](https://github.com/MarlinFirmware/Marlin/files/2161316/Files.zip)

HWiese1980, 2018-07-04:
I've had the same issue, by the way. Just sayin', because, unfortunately, I couldn't find a solution either. The examples also don't work. The output does not change when I put load on the motors. 

I bought my TMC2130 from Watterott which I assume to be a reliable distributor, not selling cheap Chinese clones (or at least somehow denote it).

rodrigosclosa, 2018-07-04:
@teemuatlut How can I do that? The examples should not work directly without having to change anything? I followed this guide too, but without success: http://marlinfw.org/docs/hardware/tmc_drivers.html

teemuatlut, 2018-07-04:
The example assumes X axis is used on a ATMega2560 and RAMPS1.4.
If you wish you to use another axis, you need to change the pin definitions accordingly, along with the step port (*PORTF* as default) and pin bit position (*0* as default).

On a RAMPS board the Y step pin is `D60` which is in `PORTF` as pin number `6`.

rodrigosclosa, 2018-07-04:
@teemuatlut 
Right. I've changed the SPI pin settings for my setup. I only changed the Y_CS, the others are unchanged.

Where do I need to change the PORTF and the pin position? Could you provide an example of that?

I'll try the example with the X axis too ...

Tks for your partience ...
Rodrigo.

teemuatlut, 2018-07-04:
```cpp
ISR(TIMER1_COMPA_vect){
  PORTF |= 1 << 0;
  PORTF &= ~(1 << 0);
}
```

rodrigosclosa, 2018-07-04:
OK, I'll try it. Tks for the sketch.
But do you have any tips on what could be wrong and I can not enable the Sensorless homing in Marlin?

matthew73210, 2018-07-09:
Hello,

I too am having kinda the same problem with the tmc2130. Steppers work fine under Marlin, direction and all. The problem arises with "stallguard".

If I use the stallguard sketch that comes in the library, it all works fine. i.e. if I hold the spindle the diag pin goes high and the serial out gives me the same results. I can also change the sensitivity.
But I had to define the spi pins for it to work.

For info, I'm using an Arduino mega 2560 with a ramps 1.5.

Here is the sketch i  used : 

`
/**
 * Author Teemu Mäntykallio
 * 
 * Plot TMC2130 motor load using the stallGuard value.
 * You can finetune the reading by changing the STALL_VALUE.
 * This will let you control at which load the value will read 0
 * and the stall flag will be triggered. This will also set pin DIAG1 high.
 * A higher STALL_VALUE will make the reading less sensitive and
 * a lower STALL_VALUE will make it more sensitive.
 * 
 * You can control the rotation speed with
 * 0 Stop
 * 1 Resume
 * + Speed up
 * - Slow down
 */
#define MAX_SPEED  40 // In timer value
#define MIN_SPEED  1000

#define STALL_VALUE -1 // [-64..63]

// Note: You also have to connect GND, 5V and VM.
//       A connection diagram can be found in the schematics.
#define EN_PIN    
#define DIR_PIN   
#define STEP_PIN  
#define CS_PIN    53 
#define MOSI_PIN  51
#define MISO_PIN 50
#define SCK_PIN  52

#include <SPI.h>
#include <TMC2130Stepper.h>
#include <TMC2130Stepper_REGDEFS.h>
TMC2130Stepper driver = TMC2130Stepper(EN_PIN, DIR_PIN, STEP_PIN, CS_PIN);

bool vsense;

uint16_t rms_current(uint8_t CS, float Rsense = 0.11) {
  return (float)(CS+1)/32.0 * (vsense?0.180:0.325)/(Rsense+0.02) / 1.41421 * 1000;
}

void setup() {
  //init serial port
  {
    Serial.begin(250000); //init serial port and set baudrate
    while(!Serial); //wait for serial port to connect (needed for Leonardo only)
    Serial.println("\nStart...");
    driver.begin();
  }

  //set TMC2130 config
  {
    driver.rms_current(600); // mA
    driver.microsteps(16);
    driver.diag1_stall(1);
    driver.diag1_active_high(1);
    driver.coolstep_min_speed(0xFFFFF); // 20bit max
    driver.THIGH(0);
    driver.semin(5);
    driver.semax(2);
    driver.sedn(0b01);
    driver.sg_stall_value(STALL_VALUE);
  }

  // Set stepper interrupt
  {
    cli();//stop interrupts
    TCCR1A = 0;// set entire TCCR1A register to 0
    TCCR1B = 0;// same for TCCR1B
    TCNT1  = 0;//initialize counter value to 0
    OCR1A = 256;// = (16*10^6) / (1*1024) - 1 (must be <65536)
    // turn on CTC mode
    TCCR1B |= (1 << WGM12);
    // Set CS11 bits for 8 prescaler
    TCCR1B |= (1 << CS11);// | (1 << CS10);  
    // enable timer compare interrupt
    TIMSK1 |= (1 << OCIE1A);
    sei();//allow interrupts
  }

  //TMC2130 outputs on (LOW active)
  digitalWrite(EN_PIN, LOW);

  vsense = driver.vsense();
}

ISR(TIMER1_COMPA_vect){
  PORTF |= 1 << 0;
  PORTF &= ~(1 << 0);
}

void loop()
{
  static uint32_t last_time=0;
  uint32_t ms = millis();

  while(Serial.available() > 0) {
    int8_t read_byte = Serial.read();
    if (read_byte == '0')      { TIMSK1 &= ~(1 << OCIE1A); digitalWrite( EN_PIN, HIGH ); }
    else if (read_byte == '1') { TIMSK1 |=  (1 << OCIE1A); digitalWrite( EN_PIN,  LOW ); }
    else if (read_byte == '+') if (OCR1A > MAX_SPEED) OCR1A -= 20;
    else if (read_byte == '-') if (OCR1A < MIN_SPEED) OCR1A += 20;
  }
    
  if((ms-last_time) > 100) //run every 0.1s
  {
    last_time = ms;
    uint32_t drv_status = driver.DRV_STATUS();
    Serial.print("0 ");
    Serial.print((drv_status & SG_RESULT_bm)>>SG_RESULT_bp , DEC);
    Serial.print(" ");
    Serial.println(rms_current((drv_status & CS_ACTUAL_bm)>>CS_ACTUAL_bp), DEC);
  }
}
`
But when i set stallguard in marlin on the M122 command  i get the same printout as the first comment. If i use the M914 Xxx command i can change the stallvalue on the fly, and this reflects with the M122 command. Yet when hold the spindle the diag1 does not go high, and this for any value of M914.

Interestingly when i set this command : 

`
#define SENSORLESS_HOMING // TMC2130 only

  #if ENABLED(SENSORLESS_HOMING)
    #define X_HOMING_SENSITIVITY  -1
    #define Y_HOMING_SENSITIVITY  -1
  #endif
`
The value is not reflected in the M112 command nor the initial echo out when i connect to the board. 

I have the latest version of Marlin 1.1.8.

My plan was to issue a command in the adv_config with this command : 

`
#define  TMC_ADV() {  }
`

I'll have to try later on this week when I have time.

Apart from that, all works fine.

Any ideas?

Matthew

matthew73210, 2018-07-09:
Sorry about the code not parsing, I don't know why it only worked on the last section.

rodrigosclosa, 2018-07-10:
@teemuatlut 

Sorry for the delay, but I tested the example sketch with the X-axis.
It also does not work for me .. The engine just makes noise, it does not spin anywhere, even though I type "+" in Terminal. And the output is the same as I sent above, same values ​​...

You commented about my engines needing a positive signal .. My engines are simple Nema17, worked with the A4988 and worked first with the TMC2130 except for Sensorless Homing.

For Marlin, I have not been able to make it work either. I tried again by Marlin 1.1.8 this weekend, then the Marlin bugfix-1.1.x and now again the Marlin bugfix-2.0.x ...

I'm about to change my set for a MKS Rumba + .. Does it help?

Tks,
Rodrigo.

comps, 2018-07-10:
Are you sure you're not hitting https://github.com/MarlinFirmware/Marlin/issues/10629#issuecomment-393244481 ? .. The symptoms seem similar enough, though if it works on earlier Marlin versions, then it's something different.

rodrigosclosa, 2018-07-10:
@comps 
To use sensorless homing, can not I leave stealthChop enabled? I did not find any instruction on this .. Just saying that Home is always done using spreadCycle ..
I left my Hybrid Threshold enabled too .. At speed above 100mm, I can see it working when I'm printing.

My Step Drivers are from a new version of Fystec that already comes with 0ohm resistors welded on the CFG4 and CFG5, and without the SPI resistor. They are already standard for use with SPI.

Tks,
Rodrigo.

mbuc, 2018-07-16:
@rodrigosclosa Have you tried reducing HOMING_FEEDRATE_XY?

rodrigosclosa, 2018-07-17:
@mbuc My HOMING_FEEDRATE_XY was 3000, very slow.. How it could affect the working of sensorless homing?

thinkyhead, 2018-09-01:
Any better with Marlin 1.1.9?

boelle, 2019-02-20:
@rodrigosclosa tried latest bugfix 2.0?

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead

safaenet, 2019-12-03:
whats the result ? Issue was not solved and the topic is closed ! i have exactly the same problem and dont know what to do

lucassing, 2019-12-09:
> whats the result ? Issue was not solved and the topic is closed ! i have exactly the same problem and dont know what to do

Same here, why do you close it @boelle? Is already any solution that you forgot to post it?

boelle, 2019-12-09:
[image]

safaenet, 2019-12-09:
@lucassing  i finally solved my own problem by soldering small tiny jumpers to enable sensor-less homing. actually i didn't care about it in the first place but it was important. See [THIS ](http://marlinfw.org/assets/images/features/FYSETC_tmc2130._SPI.jpg)image.
S CURVE ACCELERATION should be disabled. MAX  FEED-RATE and ACCELERATIONS should be reduced in order to not to lose steps and avoid layer shifting.