MarlinFirmware/Marlin #11176: Noise/communication problem tmc2130

DiogoSlb, 2018-07-01:
Good night. Im having some problems with my Gt 2560 board and tmc 2130 drivers.
These are my configs.
1.
https://ibb.co/jPgM4y
2.
https://ibb.co/hCkg4y
3.
https://ibb.co/nNwojy
4.
https://ibb.co/e7Nm4y
5.
https://ibb.co/mx564y

i double checked my connection.

i have. 
ALL SCK driver pins connected to 52 pin (SCK) (SD CARD) (img6)
ALL SDI driver pins connected to 51 pin (MOSI) (SD CARD) (img6)
ALL SDO driver pins connected to 50 pin (MISO) (SD CARD) (img6)

6.
https://ibb.co/k2HB4y
 
X_CS_PIN pin 38 (sd card)
Y_CS_PIN pin 24 (max limit switch X)
Z_CS_PIN pin 28 (max limit switch Y)
E0_CS_PIN pin 32 (max limit switch Z)

This is my debug.
Connecting...
start
Printer is now online.
echo: External Reset
Marlin 1.1.8
echo: Last Updated: 2017-12-25 12:00 | Author: (none, default config)
echo:Compiled: Jul  1 2018
echo: Free Memory: 4825  PlannerBufferBytes: 1232
echo:Hardcoded Default Settings Loaded
echo:  G21    ; Units in mm
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z4000.00 E500.00
echo:Maximum feedrates (units/s):
echo:  M203 X200.00 Y200.00 Z2.00 E20.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.30 E2.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:Stepper driver current:
echo:  M906 X 800 Y 800 Z 800 E0 800
echo:Sensorless homing threshold:
echo:  M914 X8 Y8
Y driver error detected:
overtemperature
short to ground (coil B)
		X	Y	Z	E0
Enabled		false	false	false	false
Set current	800	800	800	800
RMS current	1436	1436	1436	1436
MAX current	2025	2025	1121	1121
Run current	25/31	25/31	25/31	25/31
Hold current	5/31	5/31	5/31	5/31
CS actual		0/31	0/31	31/31	31/31
PWM scale	0	255	255	0
vsense		1=.18	0=.325	0=.325	1=.18
stealthChop	true	true	false	true
msteps		0	256	256	256
tstep		134217727	0	4294901760	0
pwm
threshold		0	0	0	0
[mm/s]		-	-	-	-
OT prewarn	false	true	false	true
OT prewarn has
been triggered	false	false	false	false
off time		15	0	15	0
blank time	16	54	16	16
hysterisis
-end		-3	-3	12	-3
-start		8	8	1	1
Stallguard thrs	8	8	0	0
DRVSTATUS	X	Y	Z	E0
stallguard		X			X
sg_result		768	1023	0	1023
fsactive		X	X
stst				X
olb		X			X
ola		X		X
s2gb			X		X
s2ga		X		X
otpw			X
ot			X
Driver registers:
	X = 0x00:00:00:00
	Y = 0x00:00:00:00
	Z = 0x00:00:00:00
	E0 = 0xFF:FC:00:00
Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

What did i miss? can someone help me ? Ty

govido, 2018-07-02:
Hi, it looks like you got the Chinese drivers. Please check if the solder bridge is on there.
http://marlinfw.org/docs/hardware/tmc_drivers.html (search for fystec) 
If the two contacts are soldered together the drivers won't work with spi and will report rubbish like short to ground.

MikeLud, 2018-07-02:
@DiogoSlb 
See the below issue, your issue might be the same, related to the TMC library version
#11164

DiogoSlb, 2018-07-02:
I Will try both things after work. Ty guys.

DiogoSlb, 2018-07-04:
Worked ty very much govido.

closed 2018-09-01 by thinkyhead