MarlinFirmware/Marlin #11179: latest bugfix branch breaks tmc 2130

wbarber69, 2018-07-02:
I recently downloaded the latest bugfix branch today.  I changed all settings to mimic what i was already using.. I had tmc 2130 drivers working perfectly with sensorless homing and everything.  Now when i try to compile i get this error:

SENSORLESS_HOMING requires X_MIN_ENDSTOP_INVERTING and ENDSTOPPULLUP_XMIN when homing to X_MIN

I understand why its saying this, but i never needed to invert my endstops before, nor did I ever change these settings in any previous version of bugfix.  I changed the settings it says to until the error went away but now i cannot get my x or y axis to move at all.  I really dont know where to start here, as i have made this configuration setting for setting exactly as my old firmware, and now nothing works.

wbarber69, 2018-07-02:
I added my two builds, one works the other does not.  remove the .txt and unrar

[error-6-1-18.rar.txt](https://github.com/MarlinFirmware/Marlin/files/2153851/error-6-1-18.rar.txt)
[working-4-21-18.rar.txt](https://github.com/MarlinFirmware/Marlin/files/2153852/working-4-21-18.rar.txt)

wbarber69, 2018-07-02:
sorry i uploaded the wrong previous version.  but i am still having these issues.  apparently along the way marlin now requires endstop pullups to be enebled for this feature.  but even after i get the errors to go away i still cannot get any motion out of the steppers.  reverting to a previous version that still requires pullups works as it should, and the latest has no motion.

wbarber69, 2018-07-02:
[4-21-18.rar.txt](https://github.com/MarlinFirmware/Marlin/files/2153965/4-21-18.rar.txt) older bugfix with everything working as it should.  latest bugfix has no motion.  and no errors as i followed the rules in sanity check

thinkyhead, 2018-07-03:
- Either enable `ENDSTOPPULLUPS` or the individual axis pullups, but not both.
- Don't forget to use `M502`, `M500` after flashing the firmware.
- If you're not using the X Max endstop for something, disable `USE_XMAX_PLUG`.
- If you're not using the Z Max endstop for something, disable `USE_ZMAX_PLUG`.
- Be sure to use the latest `bugfix-1.1.x` or `bugfix-2.0.x` branch so we're all on the same page.

wbarber69, 2018-07-03:
Reverted back to my April build and no issues.  This latest build however will not move on any axis that has sensorless homing.  I was running on a build previous to my April build and it didn’t require endstop pull-ups.  My April build does.  It could be that I was using a pre-configured example before April.  And I missed something endstop related.  But endstop pull-ups are not my issue.  I fixed the pull-up issue and the latest build still does not move on x or y axis.  Also from April build to current something else has changed.  I was always able to enable sensorless homing and still use my z endstop like normal.  Now it almost requires that z be sensorless as well.  Even though I don’t it on z I now have to put my z axis sensitivity at around 14 to get it to not falsely trigger for whatever reason, even though I don’t even have it hooked up. If I set z sensitivity to 0 it considers it triggered already and so nothing moves.  And I don’t know why because as I stated my z diag pin is unpopulated currently. I haven’t seen anywhere where anything has changed with sensorless homing so I am stumped on this one.  Perhaps I’ll let another commit or two before I try again.  Could just be something on my end.  I’ve been digging and I can’t find any where where I messed up or any stray unintentional characters or anything.  I did notice that another user was having issues with sensorless homing but after reading through their issues I couldn’t find anything relevant to my case.

marcio-ao, 2018-07-03:
@wbarber69: Could it be related to [this change](https://github.com/MarlinFirmware/Marlin/issues/10294). That change was done because the default configuration was dangerous to some boards.

joseandres42, 2018-07-13:
Hi, I'm having the same issues. I didn't need to invert the endstops logic before. My previous build was the bugfix-2.0.x from March: 
[Marlin-bugfix-2.0.x-2018-03-14.tar.gz](https://github.com/MarlinFirmware/Marlin/files/2194148/Marlin-bugfix-2.0.x-2018-03-14.tar.gz)
This is my new build:
[Marlin-bugfix-2.0.x-2018-07-13.tar.gz](https://github.com/MarlinFirmware/Marlin/files/2194158/Marlin-bugfix-2.0.x-2018-07-13.tar.gz)
Removing the SENSORLESS_HOMING section from SanityCheck.h will let it compile, but one axis gets stuck in TRIGGERED state. Also, I'm using a modified RAMPS 1.4 board so I edited the pins_RAMPS.h file. Thank you :)

wbarber69, 2018-07-13:
Yeah as was pointed out before.  There was a safety issue or something with the way all this was handled before.  You now need to invert whatever you had before.

joseandres42, 2018-07-14:
That was the first thing I tried, but if I invert what I had I get the X axis stuck in triggered:
Send: M119
Recv: Reporting endstop status
Recv: x_min: TRIGGERED
Recv: y_min: open
Recv: z_min: open

wbarber69, 2018-07-14:
Are you using sensorless homing on the x axis?  And do you have endstop pull-ups enabled?

joseandres42, 2018-07-15:
Yes, I'm using sensorless homing on both axes. I also have the pull-ups enabled. I tried enabling them individually and globally and it doesn't make a difference.

wbarber69, 2018-07-15:
Does the x axis and y axis have the same setting?

joseandres42, 2018-07-15:
Yes, except the current which is a little different. I also tried different values for the homing sensitivity. The M119 gcode reports the X endstop as TRIGGERED but the M122 says the TMCs didn't trigger:
Recv: been triggered   false false false false

wbarber69, 2018-07-15:
Did the wire come loose?

thinkyhead, 2018-07-15:
That `M122` output refers to the Overtemp Pre-Warn:
```
OT prewarn has
been triggered  false false false false
```
The `M119` command does report the current status of stallguard-based sensorless homing, but they will rarely show "triggered" because stallguard-based endstops don't stay triggered for very long — they're only triggered when a motor is actively hitting a barrier.

wbarber69, 2018-07-16:
Which pretty much means it’s an electrical issue, or you’ve got something inverted the wrong way in firmware

joseandres42, 2018-07-16:
It's not an electrical issue because my March build works just fine, and I'm using the same configuration. Both builds are uploaded in a previous comment.

seandop, 2018-07-24:
I'm experiencing a similar issue, and I'm still actively trying to nail it down. My build from 6/28 on the bugfix 1.1.x branch homes fine using sensorless homing on both X and Y. Using files from 7/19, my X stepper tries to home on the first G28 issued but it does so in StealthChop, rather than SpreadCycle. Obviously, stallguard won't work in StealthChop, so I shut off the machine before it rams the carriage into the endstop. After restarting the printer, the gantry assembly will raise on a G28, but the X and Y motors do not respond and eventually the printer is halted. X and Y will not attempt to home again unless I reflash the firmware. Re-flashing my build from 6/28 and again everything works fine...

joseandres42, 2018-08-06:
I'm still having this issue, but I'm using a modified RAMPS board. I'm stuck my March build, and would love to update it.

GibsDev, 2018-08-28:
I'm having the same exact issue. Could it be that the conditions for the warning are not correct?

closed 2018-09-06 by thinkyhead

wbarber69, 2019-05-03:
I finally went ahead and recently tried to upgrade to the latest bugfix branch again, and low and behold i had the exact same issue.  I finally figured it out though and its pretty stupid.  Im using the mks gen-l board and when i define that board in my configuration it then uses the pins.h for that specific board.  well after hours of trying to finally nail this problem down i had a eureka moment.  in the past the gen-l pins.h would just forward to the generic ramps pins.h but somewhere along the line there was some changes made to the gen-l pins.h  and there are actually two lines in there calling up the cs pins for the x and y.  and since i use software spi for my trinamic drivers and those cs pins are declared in the standard pins.h thery are overridden by the gen-l pins.h.. after changing the necessary cs pin settings in the gen-l pins.h file im back up and running on the latest bugfix branch.

RiRilot, 2019-06-23:
I am having the exact same issue but I'm using an SKR V1.3. With sensorless homing enabled I have to invert the X and Y endstops and configure pullups for both X and Y. When asking X or Y to home the endstops trigger immediately no matter what the value of M914 sensitivity.
I'm using the latest build of Marlin 2.0. Board is correctly defined as BIGTREE_V1_3

seandop, 2019-06-23:
Been there. Whoever had the brilliant idea to declare those two pins in the
MKS gen-l pins file broke my Trinamic setup several months ago and it took
me a while to figure it out. Note that in the “newer” 2.0.x firmware
versions, you can declare your cs pins in either configuration.h or
configuration_adv.h (I forget which) for Trinamic software spi which will
override the pins listed in your pins.h file. Just to make things more
confusing lol.