MarlinFirmware/Marlin #11180: BLTouch not triggering after the first touch and crashing the bed

bfrakes256, 2018-07-02:
Installing a BLTouch in Marlin and been having issues with it probing the first time, not reversing the z motor and crashing into the bed.

I've tested the BLTouch with a voltmeter and it does trigger on the correct pins.  As the pin hits the sensor the voltage drops then comes back to the original voltage as it goes into the flashing mode which indicates a problem.

The distance from the nozzle to the bottom of the housing on the BLTouch is set to 8.3mm as specified.  Tried reducing this distance to see if it will trigger earlier with not joy.

Marlin seems to be configured correctly but I may have missed something.

Have not tried changing the BLTOUCH_DELAY value...

Ideas?

Help appreciated...

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

bfrakes256, 2018-07-02:
It is now working

closed 2018-07-02 by bfrakes256

CWSpear, 2019-03-20:
@bfrakes256 what did you do to get it working?

bfrakes256, 2019-03-20:
Are you referring to the btouch?

Fiddled and changed to the beta version of the Arduino software if so...

Lots if fiddling with Marlin...

;-)