MarlinFirmware/Marlin #11181: [2.0.x] Answer : Z_max and Z_min

J-charles-C, 2018-07-02:
I,

Little question.
I have plugged my Z_max optical Endstop.
With M119 command All is ok. Triggered when is close and open when, for example, Z is at the middle.

I have active  :

#define USE_XMIN_PLUG
//#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
//#define USE_XMAX_PLUG
#define USE_YMAX_PLUG
#define USE_ZMAX_PLUG

Homing is ok, X & Y endstop stop the travel head. Bl-touch is OK.

When i launch G1 Z travel over the Z-Max endstop,  he don't stop bed.

best

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/2156120/Archive.zip)

hexane360, 2018-07-02:
Can you retry this with the latest firmware? I think this is a duplicate of #11158 which was fixed by reverting #11013.

J-charles-C, 2018-07-02:
I,

With the last, and #define USE_ZMAX_PLUG, it's ok. Z Travel is not interrupt by endstoP;

**when bed is at the middle travel**
 Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_max: open
Recv: z_min: open
Recv: z_max: open

**When bed is on z_max endstop :**

Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_max: open
Recv: z_min: open
Recv: z_max: TRIGGERED

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/2156248/Archive.zip)

best

hexane360, 2018-07-02:
Again, I believe this issue has recently been fixed. Download the latest bugfix-2.0.x, copy over your configurations, and reflash the printer.

J-charles-C, 2018-07-02:
Good evening,

 my second answer was obtained with the latest version downloaded just now.

There is something else to un comment except #define USE_ZMAX_PLUG ?

best

thinkyhead, 2018-07-03:
> There is something else to un comment except `#define USE_ZMAX_PLUG` ?

The `Z_HOME_DIR` setting must be 1 for a max endstop, or -1 for a min endstop.

J-charles-C, 2018-07-03:
I,

I have choose homing on Z_Min. this setting is ok.

J-charles-C, 2018-07-03:
If,

If i comment  //#define MAX_SOFTWARE_ENDSTOP_Z

Endstop dont't interrupt Z-Travel

If i un-comment #define ENDSTOPS_ALWAYS_ON_DEFAULT

Endstop interrup Z travel but, after, i can't move it with G1 command ; i must do a G28 Z before

best

hexane360, 2018-07-04:
When you say "i can't move it with G1 command", does that include commands both towards and away from the endstop? For example, if you run a `G1 Z200` which triggers the endstop, does a `G1 Z0` work?

thinkyhead, 2018-07-04:
> Endstop interrupt Z travel but, after, i can't move it with `G1` command

`M121` will disable endstops at any time. `M120` will turn them back on (and clear the hit state).

J-charles-C, 2018-07-04:
> When you say "i can't move it with G1 command", does that include commands both towards and away from the endstop? For example, if you run a G1 Z200 which triggers the endstop, does a G1 Z0 work?

No, I must execute an G28z before.

>MAX_SOFTWARE_ENDSTOP_Z is used to prevent movement past Z_MAX_POS and has nothing to do with endstop switches.

It's already ok but i want change software endstop for optical.

hexane360, 2018-07-14:
I think I have a new idea about this. Can you upload "Marlin/src/module/endstops.cpp" (or just your whole Marlin source)?

J-charles-C, 2018-07-15:
I,

 Also, try disabling software endstops completely and then uncomment #define ENDSTOPS_ALWAYS_ON_DEFAULT.

My commented Jul 3, 2018 :

If i comment //#define MAX_SOFTWARE_ENDSTOP_Z

Endstop don't interrupt Z-Travel

If i un-comment #define ENDSTOPS_ALWAYS_ON_DEFAULT

Endstop interrupt Z travel but, after, i can't move it with G1 command ; i must do a G28 Z before

best

J-charles-C, 2018-07-15:
My Marlin is drop on the first post (archive.zip)

hexane360, 2018-07-15:
The zip you uploaded is just the configuration headers. I'd like to see the full source.

workheart, 2018-08-27:
You say you have zmax but you are homing to -1
#define Z_HOME_DIR -1 
Should be 1 to home up

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead