MarlinFirmware/Marlin #11182: Error Compiling for Board Sanguino

Splic3dBread, 2018-07-02:
I'm trying to update my CR-10s firmware and Arduino is giving me "Error Compiling for Board Sanguino". Keep in mind that this is simply when compiling and not uploading so it's not a problem with the bootloader installation.

Here's what I have verified and things I've tried...

Reinstalling the Arduino IDE and deleting all local files
I've inserted [this](https://raw.githubusercontent.com/Lauszus/Sanguino/master/package_lauszus_sanguino_index.json) link to preferences
Sanguino selected for board
Latest version of Sanguino (1.0.2)
Prgrammer AVRISP mkII selected
Atmega1284 or Atmega1284P (16 Mhz) selected for processor
Creality Melzi Board selected for board
Reprap Discount Full Graphic Smart Controller selected for LCD
U8glib library included in configuration.h
I was able to dig up some obscure case of marlin not compiling because the baud rate was higher than the max baud rate set in the sanguino's board.txt but unfortunately changing that did not help. I even tried uploading an older version of marlin for the CR-10, the original source code infact and it still gave me the same error!

I'm absolutely stumped. It's so frustrating because I effectively have a bricked printer but there's nothing actually wrong with it!

Splic3dBread, 2018-07-02:
I was able to narrow it down to an LCD issue. For some reason all the marlin configs I'm looking at have the reprap discount full graphic display selected but I needed to use the one specifically for the cr-10, I didn't even notice it was there. Is the CR-10 stock display option new to Marlin 1.1.8?

Bob-the-Kuhn, 2018-07-03:
FYI - the CR10 display was added in September 2017.

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead