MarlinFirmware/Marlin #11184: EEPROM and LCD Bed Leveling cannot be enabled together.

Splic3dBread, 2018-07-03:
Trying to get Marlin 1.1.8 working on cr-10 had been a 12 hour straight nightmare. Scraping everything and starting over has fixed some of my issues even though the configuration files were almost identical (files were somehow currupted?), but now there's a new problem. Going down the configuration.h and compiling with each change to figure out what's causing errors has shown me that having EEPROM settings enabled along with LCD Bed Leveling + Mesh Bed Leveling creates a "Error Compiling for Board Sanguino Error". Btw, everytime one of these "mystery errors" with no conceivable explanation occur that's the output, nothing else whatsoever. Using LCD Bed Leveling + Manual Probing + Any other bed leveling works fine with EEPROM. I tried compiling with ramps1.4 as the board and it compiled just fine.

Splic3dBread, 2018-07-03:
Of course right after posting this I figured out what has been causing all my problems, the CR-10 board doesn't have enough storage space. Sorry for the spam, I wish I could delete issues....

AnHardt, 2018-07-03:
**You** at least could close it.