MarlinFirmware/Marlin #11185: Enable bed and nozzle heating simultaneously when starting print

stibinator, 2018-07-03:
Is there a reason why the bed is heated first and then the nozzle when starting a print, rather tan them both heating at once? It takes a lot longer, especially if the nozzle is already hot, because it cools down while the bed is heating and then has to heat all over again.

Steps to reproduce:
Print from SD

Expected:
Bed and nozzle heat to required temps, simultaneously.

Observed:
bed ehats, then nozzle heats

thinkyhead, 2018-07-03:
The heating commands `M109` and `M190` wait, while the commands `M104` and `M140` don't wait. Hosts and slicers decide which of these commands to use.

closed 2018-07-03 by thinkyhead

stibinator, 2018-07-03:
Ok, thanks. I'll go make the feature request for Cura instead.