MarlinFirmware/Marlin #11188: Q: Lost link to MCU list

boelle, 2018-07-03:
Hi, i lost a link that listed different MCU's, how fast they run and how much memory etc each has

i cant remember if its in the wiki or if it was in the form of an issue, but google was far from my friend on this one

best i could find was this one: https://github.com/MarlinFirmware/Marlin/issues/7076

but that was not the one i was after

Sineos, 2018-07-03:
This one? https://reprap.org/wiki/Comparison_of_Electronics

boelle, 2018-07-03:
nope...

it was not a electronics list... it was a list of MCU's.... Arduino Mega.... Due.... Teensy

tig33r, 2018-07-03:
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

Current HALs section

boelle, 2018-07-03:
yep, that was the one, link saved :-D

closed 2018-07-03 by boelle