MarlinFirmware/Marlin #11189: [2.0.x] Ender -3 BLTouch - hotend goes up after probing

lestzx, 2018-07-03:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2159249/Configuration.zip)
I have an issue with Marlin setup. After probing the z height, the nozzle goes up (also with Auto Home) and it starts printing to high. I use 3DTouch and Petsfang on my Ender 3 printer.

Thank you for your help

MikeLud, 2018-07-03:
@lestzx 
You need to adjusts the below setting in Configuration.h file, try a setting like -1.8 this will lower the nozzle by 0.2mm. Mine is set to -1.6
`#define Z_PROBE_OFFSET_FROM_EXTRUDER -2   // Z offset: -below +above  [the nozzle]`

lestzx, 2018-07-03:
It doesn't change nothing. After probing it goes to Z-20 every time

seandop, 2018-08-12:
I'm experiencing this issue, also. TMC2130 drivers with MKS Gen L board. BLtouch v2.

thinkyhead, 2018-11-05:
Assuming this is solved.

closed 2018-11-05 by thinkyhead