MarlinFirmware/Marlin #11190: Q: just a thought on UBL

boelle, 2018-07-03:
since i will upgrade to arduino due and either Ramps 1.7 or Ramps FD V2.2 i wondered if the extra space could be used to fine a finer mesh? ie more points pr axis?

one of my printers will have a 500x500 area and would benefit from it to keep the same mesh density

just an idea thou ok if its' not possible at all, it will of course require more space in eeprom, but since i will hand build the electronics i can just pick an eeprom with enough space

Roxy-3D, 2018-07-03:
The mesh resolution is purely a function of available RAM memory.   15x15 should provide enough resolution.   I only use 10x10 on my 400x400mm printer and that works well.

But with that said...    The 15x15 number can be bumped up to 20x20 or 30x30 very easily if you have a 32-bit platform (that has more RAM memory).

There would be a few declarations that have to change.   We would need to shift a few declarations like:
```
  uint16_t not_done[16];
```
To be something more like:
```
  uint32_t not_done[32];
```
The changes are very straight forward and obvious.     Most likely 15x15 is going to work fine on a 500mm x 500mm bed.   But if it doesn't...   We can go to 30x30.

boelle, 2018-07-03:
are the declarations that needs change all in the same file?

it will still be another month before i can start to test it, just finished the frame and need to save up for the whole electronics part

boelle, 2018-07-03:
oki will start out with 15x15 and see where i end up, but again it will be a month at least before i have hardware etc so that i can test it

but will close this one so it does not clutter the list more than need be, but i will save the link to issue so i can find it and report back

closed 2018-07-03 by boelle