MarlinFirmware/Marlin #11192: G-Code selection for new features

TheLongAndOnly, 2018-07-03:
I'm considering to implement a new feature which allows automated alignment of a multi-stepper Z-axis by using a Z-bed probe for 2.0.x. This would be beneficial for printers like e.g. Prusa i3 (dual) and SparkCube (triple).

I'm wondering how I should select a proper G-Code to trigger such a feature?

ManuelMcLure, 2018-07-03:
This might best be an option to either `G29` or `G28`. An argument could also be made for overloading `G33` .

thinkyhead, 2018-07-04:
I'm not sure overloading `G28` is exactly right, in terms of matching the meaning of `G28`, which is used merely to establish a known position using endstops and (optionally) a Z probe. I would suggest using `G34` instead, since [this code is available](http://reprap.org/wiki/G-code) and because codes in this general range apply to probing and calibration.

Roxy-3D, 2018-07-04:
Agreed with https://github.com/MarlinFirmware/Marlin/issues/11192#issuecomment-402335203

But @TheLongAndOnly  it really doesn't matter right now...  Take the simple approach and just grab any unused GCode number.     Grab G34 and get the code working.

Once it is working, we can circle back and figure out if it should be combined with some other functionality or be a stand alone number by itself.

TheLongAndOnly, 2018-07-04:
Ok, I will start working using G34. One additional question: I have a pending PR #11062 which I'm depending on (and a second #11132 which is mainly required for the setup with RADDS extension board). I cannot tell how fast I can get this to work, but what is your recommendation in case the PRs are not integrated? Can I somehow mark up the dependencies (or just through the comments?)

Roxy-3D, 2018-07-04:
Are the Triple Z-Steppers needed for this new work?  If not...  It would be best to not have those changes in the Pull Request.

If they are needed...   Just keep going forward with what you have.

TheLongAndOnly, 2018-07-04:
I would need to create a variant for dual z-steppers and an addition for triple then. I will give it a try and have 2 additional branches for that.

Roxy-3D, 2018-07-04:
A better solution would be to make the code sensitive to some thing like:
```
 #define N_Z_STEPPER_MOTORS 3 
```
If you made the code conditionally compiled in based on that #define value, that is more in tune with how the Marlin code base handles different hardware options.

TheLongAndOnly, 2018-07-04:
I'm anyway going to do it this way. But I could have options in which don't work if enabled.

closed 2018-11-13 by TheLongAndOnly

TheLongAndOnly, 2018-12-25:
> I'm trying to find information on how G34 works to align dual Z steppers using the probe. Can anyone give me a brief explanation as to how it functions?

You need to define two xy coordinates to probe your z-steppers. Ensure they are as close as possible to z-stepper mounts and are not twisted, otherwise the function will mis-align them further and stop.You need to run G28 before G34.If your probe position distance to your stepper is bigger, it could make sense to set the amplification. Example: G34 will probe both points, calculate the offset an the move the side which is closer to reduce the offset. It will move by offset*amplification and then run the next iteration. You can see, that amplification will speed up the process, as 1.0 will not lead to a full correction at the probe position. Iteration also stops, if the offset is below a defined value.

Aepx, 2019-03-24:
Can't find anything new on this so I'm gonna ask here.
How about making G34 just a leveling option and not a zeroing function?
So set the limit switches with a positive offset from the bed, say 10mm, then run a G34, the switches are triggered one at a time, calculating the offset as any dual Z homing does, then use the probe as the actual zeroing?

TheLongAndOnly, 2019-03-24:
I finally started to work in the Documentation. I'm sorry, I've been to busy.

G34 is really meant to be an automatic function to align the bed with the printing plane. It does not take care about end stops and relies on G28 for it.

But it triggers G28 automatically after completion.