MarlinFirmware/Marlin #11199: Compilation error if #define EXTRUDER_AUTO_FAN_SPEED not equal 255

tig33r, 2018-07-04:
I found exactly same issue some time ago but closed https://github.com/MarlinFirmware/Marlin/issues/11141 with reference to commit which should fix it https://github.com/MarlinFirmware/Marlin/pull/11165. Please correct me if I am wrong but it should works now properly.

Log

`Arduino:1.9.0-beta (Windows 10), Płytka:"Arduino Due (Programming Port)"

In file included from sketch\src\HAL\HAL_DUE\../../inc/MarlinConfig.h:37:0,

                 from sketch\src\HAL\HAL_DUE\Servo_Due.cpp:43:

sketch\src\HAL\HAL_DUE\../../inc/SanityCheck.h:1382:5: error: non-constant condition for static assertion

     static_assert(GET_TIMER(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);

     ^

sketch\src\HAL\HAL_DUE\../../inc/SanityCheck.h:1382:5: error: the value of 'g_APinDescription' is not usable in a constant expression

In file included from sketch\src\HAL\HAL_DUE\../../inc/../HAL/HAL_DUE/HAL.h:36:0,

                 from sketch\src\HAL\HAL_DUE\../../inc/MarlinConfig.h:28,

                 from sketch\src\HAL\HAL_DUE\Servo_Due.cpp:43:

C:\Users\tig33r\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino/Arduino.h:186:29: note: 'g_APinDescription' was not declared 'constexpr'

 extern const PinDescription g_APinDescription[] ;

                             ^

In file included from sketch\src\HAL\HAL_DUE\../../inc/MarlinConfig.h:37:0,

                 from sketch\src\HAL\HAL_DUE\persistent_store_impl.cpp:5:

sketch\src\HAL\HAL_DUE\../../inc/SanityCheck.h:1382:5: error: non-constant condition for static assertion

     static_assert(GET_TIMER(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);

     ^

sketch\src\HAL\HAL_DUE\../../inc/SanityCheck.h:1382:5: error: the value of 'g_APinDescription' is not usable in a constant expression

In file included from sketch\src\HAL\HAL_DUE\../../inc/../HAL/HAL_DUE/HAL.h:36:0,

                 from sketch\src\HAL\HAL_DUE\../../inc/MarlinConfig.h:28,

                 from sketch\src\HAL\HAL_DUE\persistent_store_impl.cpp:5:

C:\Users\tig33r\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino/Arduino.h:186:29: note: 'g_APinDescription' was not declared 'constexpr'

 extern const PinDescription g_APinDescription[] ;

                             ^

In file included from sketch\src\gcode\host\../../inc/MarlinConfig.h:37:0,

                 from sketch\src\gcode\host\../gcode.h:251,

                 from sketch\src\gcode\host\M118.cpp:23:

sketch\src\gcode\host\../../inc/SanityCheck.h:1382:5: error: non-constant condition for static assertion

     static_assert(GET_TIMER(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);

     ^

sketch\src\gcode\host\../../inc/SanityCheck.h:1382:5: error: the value of 'g_APinDescription' is not usable in a constant expression

In file included from sketch\src\gcode\host\../../inc/../HAL/HAL_DUE/HAL.h:36:0,

[... 1198 lines cut ...]

In file included from sketch\src\feature\../inc/MarlinConfig.h:37:0,

                 from sketch\src\feature\baricuda.cpp:23:

sketch\src\feature\../inc/SanityCheck.h:1382:5: error: non-constant condition for static assertion

     static_assert(GET_TIMER(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);

     ^

sketch\src\feature\../inc/SanityCheck.h:1382:5: error: the value of 'g_APinDescription' is not usable in a constant expression

In file included from sketch\src\feature\../inc/../HAL/HAL_DUE/HAL.h:36:0,

                 from sketch\src\feature\../inc/MarlinConfig.h:28,

                 from sketch\src\feature\baricuda.cpp:23:

C:\Users\tig33r\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino/Arduino.h:186:29: note: 'g_APinDescription' was not declared 'constexpr'

 extern const PinDescription g_APinDescription[] ;

                             ^
`

ejtagle, 2018-07-04:
@tig33r : Replace fastio_Due.h by the attached one and it should work. The fix implemented by #11165 is correct, but does not satisfy the static assertions used to verify compilation

[fastio_Due.zip](https://github.com/MarlinFirmware/Marlin/files/2164329/fastio_Due.zip)

tig33r, 2018-07-05:
Seems like issue solved with newest bugfix. Thanks!

closed 2018-07-05 by tig33r