MarlinFirmware/Marlin #11201: Question: Which 32 bit board, is MKS considered a cancer by Marlin?

gloomyandy, 2018-07-04:
I'm considering buying a 32bit board to hopefully be able to contribute to the Marlin 32bit development effort. But this of course raises the question of which board to get. I'm based in the UK and after a bit of shopping around it seems that the following are easily available to me:
MKS SBASE V1.3
RADDS 1.5 + Due
Smoothirboard 5XC
Duet WiFi
Of the above by far the lowest cost is the MKS SBASE. However I've seen a few comments about the quality of this hardware and on the Smoothieware support list a number of comments about MKS and the approach they take to open source. Having seen the issues that Marlin has had with some producers of printers and the attitude to open source, I thought it might be worth getting the views of the current developers (or others) about...
a) MKS
b) The quality of the MKS SBASE boards
c) If MKS is not considered a good choice, which of the above boards should I consider?
I'd prefer not to be the only one trying to get whatever board I have to work, so ideally I'd like to use something that is already being used by the devs. I'm a pretty experienced software developer (30+ years) and have a reasonable knowledge of hardware (and have various tools like scopes, logic analysers etc.) and have used PIC/Atmel/ARM processes in embedded systems of various sorts. In terms of what I want I'm pretty flexible, my only requirements are:
1. I'd like to run it at 24V
2. I'd like it to have a display and operate stand alone. 
3. My printer is a pretty standard Cartesian i3 style system (made by Hictop) with a single extruder.  
Thanks for any help you could provide. I did spend some time searching these topics, but couldn't find these questions being addressed before. If I missed something please feel free to direct me there and close this issue.

forkoz, 2018-07-04:
MKS sbase seemed like a drop in for for my mks gen. The gen really didn't give me any issues hardware wise. It was either sBase or TMCs, time will tell if I chose wrong or not. First sbase I bought used and it came with blown parts so I'm waiting on the replacement. That doesn't seem like MKS fault, just shady seller and me being cheap. Looking over the board HW it seems ok and the main bits would be easily replaceable, regulator, mosfets, etc. 

Issues with this setup I've noticed (before even booting the board, lol):

1. Endstop interrupts don't work even though interrupt code is in marlin (winterrupts.cpp)
2. Beeper code don't exist yet so I don't know if the reprap LCD will make noises again every press, hope not. You have to flip the LCD cables for the reprap generic LCD and its built in SD slot won't work without rewiring (IDC). A touch or whatever LCD you buy it might not matter at all.
3. You need breakout boards to use different drivers otherwise you're stuck with the stuttering DRVs. This is where most of the layer shifting people came from it seems. I hope not to see it with my setup. On the flip side a lot of boards don't support swapping drivers at all.

I too have a cartesian and want to print faster with higher stepping. Supposedly 32bit won't help us much. I looked at the other boards and they either cost too much or were missing features. Look in boards.h for other boards, there honestly aren't a lot of 32 bit choices.

From what I've read the problem with MKS is that they take from the community and don't give back. Their boards are open but they pretty much copied smoothie and left smoothie supporting their customers. Obviously smoothie didn't want a clone maker profiting off their design and leaving them with headaches and board quirks for them to fix.

p3p, 2018-07-04:
@gloomyandy The LPC176x platform supports the LPC176x mcu (with smoothie bootloader) not a specific board, but is tested on the Re-ARM and MKS-SBase (because I have them) other boards would need a pins file created and may have unforeseen issues. 

I'm not sure I'm in a position to recommend a board, I use a Re-ARM in my printer atm (it helps as I'm a LPC176x maintainer :wink: ) but the DUE platform should in theory have better support as it is at least an Arduino platform (I haven't used one though)

@forkoz Endstop interrupts needs pins that support interrupts and smoothieboard and its clones out of the many possible choices (46 of them interrupt compatible) uses ones that don't so unfortunately that's a hardware limitation. 

Making peripherals (LCD ect) work with the LPC176x has been a struggle, it is not an Arduino platform we have to write all the code ourselves and attempt to use Arduino libraries by pretending to be an Arduino platform, but its not going too badly for a beta, now I have an SBase I can at least test on it.

There's no reason DRV8825 drivers won't work properly, they just need a longer step pulse (2us) than some others but that can be configured. 

> want to print faster with higher stepping. Supposedly 32bit won't help us much.

LPC176x as it stands should hit 60-70KHz before even needing to double step in comparison AVR will get about 10KHz, so I'm not sure it's no help, though mechanical limitations (extrusion) usually hit first before even maxing out the AVR.

Bob-the-Kuhn, 2018-07-05:
I wouldn't base any decisions on increased microstep capabilities.

Microstepping's only advantage is it results in quieter stepper operation.

Microstepping does NOT increase stepper/axis resolution nor accuracy.

It does reduce the amount of power available.

----

I haven't seriously looked at the LV8729.   Why do you need breakout boards for the LV8729?

gloomyandy, 2018-07-05:
Thanks for the feedback so far! I'm tempted to try the MKS board as it is easy to get and has a smaller form factor than some of the other options. Though I must admit I'm a little concerned about possible problems with the DRV8825 and 24V. 

@p3p what LCD options are currently working (if any), I was thinking of getting a RepRapDiscount Full Graphic Smart Controller to use with this, I also have a LCD 2004 Graphic Smart Display Controller (20x4 line) that I could use.

Though I could be convinced to go RADDS/Due route as that seems to offer a lot of flexibility to experiment with different stepper drivers etc.

teemuatlut, 2018-07-05:
> Microstepping does NOT increase stepper/axis resolution nor accuracy.

Why would it not do that?

> It does reduce the amount of power available.

That's a common misunderstanding and is not true.
The driver power is determined by the current in the coils, which are determined by the microstepping table. The microstepping setting/configuration sets the jump in steps we take on this table and as long as we end up in the same slot, the coil current is the same.

tig33r, 2018-07-05:
RepRapDiscount Full Graphic Smart Controller is working good including sd card and even sd detect pin (of course you have proper connected octopus cable ;)).

Bob-the-Kuhn, 2018-07-05:
I've seen resolution, accuracy, power, ... vehemently argued both ways.  

My memory isn't the best anymore so I did a little looking at what appear to be technical articles.  A lot of the popular arguments are based on apples/oranges comparisons using less than fully defined terms. Some of the technical papers disagree on the accuracy effects.

If you're running full stepper steps (enough microsteps to move the stepper one or more native magnetic detents) then the available torque to move the stepper/axis is unaffected.

The "fun" is the starting & stopping phases.  

The stepper doesn't start to move until enough current is present (enough microsteps issued) to overcome system inertia & friction.  

Where a stepper stops depends on current, system inertia & friction.

Add to the above the current/torque vs. position curve for a stepper isn't a perfect sinusoid.

**IF** I understand things properly, net result is there **IS** an improvement in absolute positional accuracy **but** the final position depends on more than the number of microsteps.

GMagician, 2018-07-05:
@Bob-the-Kuhn you are right, resuming: more steps give, theoretical, more precision, less noise and no differences in torque (when moving), but since final position depends on "current applied to phases", final position may not be what expected. And this depends on many factors.

gloomyandy, 2018-07-05:
Is there a discussion somewhere which explains why the pins that the MKS SBASE would normally use for the SD card do not work with Marlin (do they work with smoothiware?). I'm looking forward to getting to grips with this new hardware.

p3p, 2018-07-05:
@gloomyandy It's a work in progress, those pins are shared with the onboard sd-card (SPI bus) but the USB Mass Storage driver needs exclusive control. I am working on improving things but hobby time is limited and often taken up by investigating issues rather than coding.

forkoz, 2018-07-05:
So does the USB interfere with the internal SD or only the internal/external SD interfere or all 3? The jumpers in the middle supposedly disconnect internal SD SPI access. Yea, looking on this board there are definitely not enough pins. Just the cost is so huge between smoothie and MKS, must be the difference of having to come up with your own design. 

Has marlin done anything with the ethernet port? Maybe it could support serial over TCP/IP

p3p, 2018-07-05:
The USB stack uses interrupts and needs to access the onboard card during them so it just can't share the SPI port as it currently is, same reason that Marlin can't use the card for gcode, I've been working on updating that.

> Has marlin done anything with the ethernet port? Maybe it could support serial over TCP/IP

Having a hardware Ethernet port and integrating an Ethernet stack are two very different things :wink: it's a possibility, especially now I have a board with the port on it, although on the Re-ARM most people use the Ethernet header pins for other things. My feeling on the matter are generally the hardware resources (and mine) are better spent on other things.

closed 2018-11-22 by thinkyhead

tidus350, 2018-12-18:
Problems ive run into with mks sbase 1.3. No support for delta printers so if you have a delta dont buy ive been down over a month and have tried both old and new firmware as well as marlin 2.0 so far i havent been able to even get it to calibrate half the time it slams my print head into the bed the other 1/2 i get a false read from probe saying probe was activated before move. Misreading mm i know my height from bed to tip of nozzel is exactly 310 mm mks firmware reading at 446.64 mm so i assume steps are off in firmware and no way to adjust them. Could adjust in marlin but again changing speed didnt effect and there are issues with the g33 auto calibrate it doesnt work period. Misreading between the 3.5 inch screen that comes with it doesnt help either and it locks up more often then not. So far i have found one person that has been able to get the board to work and its been printing like crap for him. This is just my personal opinion but if i had it to do over again i would have gone with the mks sbase and never will again buy any of their products.

Noudicus, 2019-04-07:
I am using a mks-sgen with separate drivers on my Delta, running the latest Marlin 2.0.x beta. I calibrated the printer manually and am using the 3d-touch for bed leveling. 
The only problem I had was that bilinear bedleveling seems to have a ?bug?. With 3 point bedleveling things seems to work as espected. 
I would prefer a MKS-sgen over a MKS-sbase as it does not use fixed drivers.

forkoz, 2019-04-07:
That wasn't available back then. I don't think sGen was available till very recently either. There are also the various SKR boards now.

Noudicus, 2019-04-07:
Wow I have overlooked the SKR, that's insanely cheap. Good tip, thnx

Roxy-3D, 2019-04-07:
> Wow I have overlooked the SKR, that's insanely cheap. Good tip, thnx

I wish the SKR had support of 6 Stepper Drivers.   Alas...   It only supports 5.