MarlinFirmware/Marlin #11203: lcd viki2 + rambo 1.3 l

labotecno, 2018-07-04:
hello every one  ! 

i try to upgrate one airwolf hd2x , i have one lcd graphic viki2 , i get it compliled but i have blank screen , someone can to help me to geting this working ? 
thank you very much
[config.zip](https://github.com/MarlinFirmware/Marlin/files/2164320/config.zip)

labotecno, 2018-07-05:
no body can to help me ?

GMagician, 2018-07-05:
Have you tryed with bugfix?

labotecno, 2018-07-05:
yes i try with bugfix and i get a problem with i dont get on the scope delay10s , the only how compiled good is with firmware 1.1.8 but i get blank screen . its not a connection problem because i get it working with this firmware 
[Marlin-Development.zip](https://github.com/MarlinFirmware/Marlin/files/2167710/Marlin-Development.zip)

fiveangle, 2018-07-05:
Your "working" config doesn't have viki2 defined:

`//#define VIKI2`

labotecno, 2018-07-06:
wait this one working , but i need the last one marlin working with viki2 someone can to help me ????
[firmwareviki2.zip](https://github.com/MarlinFirmware/Marlin/files/2168864/firmwareviki2.zip)

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead