MarlinFirmware/Marlin #11204: out of the blue compile error

boelle, 2018-07-05:
to test some noise issues i turned both bed and hotend in to bang-bang mode 

now i turned it back to PID mode and i got this compile error:

sketch\src\HAL\HAL_LPC1768\include\digipot_mcp4451_I2C_routines.c:27:0: error: expected '=', ',', ';', 'asm' or '__attribute__' at end of input

 

 ^

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

and yes the only change in config was changing these 2 lines

#define PIDTEMP
#define PIDTEMPBED

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/2165852/Configurations.zip)

the zip file contains a copy of the pin file for ramps as i have some custom changes in it

boelle, 2018-07-05:
decided to try todays snapshot of bugfix 2.0.x, with the same 2 config files and pin file as in the above zip i now get these compile errors:

> In file included from c:\users\boher\appdata\local\temp\arduino_build_558883\sketch\src\inc\MarlinConfigPre.h:31:0,
> 
>                  from c:\users\boher\appdata\local\temp\arduino_build_558883\sketch\src\inc\marlinconfig.h:26,
> 
>                  from sketch\src\gcode\bedlevel\G42.cpp:23:
> 
> c:\users\boher\appdata\local\temp\arduino_build_558883\sketch\configuration.h:1:1: error: 'b' does not name a type
> 
>  b/**
> 
>  ^
> 
> In file included from c:\users\boher\appdata\local\temp\arduino_build_558883\sketch\src\gcode\gcode.h:252:0,
> 
>                  from sketch\src\gcode\bedlevel\G42.cpp:27:
> 
> c:\users\boher\appdata\local\temp\arduino_build_558883\sketch\src\gcode\parser.h: In static member function 'static float GCodeParser::value_float()':
> 
> c:\users\boher\appdata\local\temp\arduino_build_558883\sketch\src\gcode\parser.h:189:51: error: 'strtod' was not declared in this scope
> 
>            const float ret = strtod(value_ptr, NULL);
> 
>                                                    ^
> 
> c:\users\boher\appdata\local\temp\arduino_build_558883\sketch\src\gcode\parser.h:195:36: error: 'strtod' was not declared in this scope
> 
>        return strtod(value_ptr, NULL);
> 
>                                     ^
> 
> c:\users\boher\appdata\local\temp\arduino_build_558883\sketch\src\gcode\parser.h: In static member function 'static int32_t GCodeParser::value_long()':
> 
> c:\users\boher\appdata\local\temp\arduino_build_558883\sketch\src\gcode\parser.h:201:85: error: 'strtol' was not declared in this scope
> 
>    inline static int32_t value_long() { return value_ptr ? strtol(value_ptr, NULL, 10) : 0L; }
> 
>                                                                                      ^
> 
> c:\users\boher\appdata\local\temp\arduino_build_558883\sketch\src\gcode\parser.h: In static member function 'static uint32_t GCodeParser::value_ulong()':
> 
> c:\users\boher\appdata\local\temp\arduino_build_558883\sketch\src\gcode\parser.h:202:88: error: 'strtoul' was not declared in this scope
> 
>    inline static uint32_t value_ulong() { return value_ptr ? strtoul(value_ptr, NULL, 10) : 0UL; }
> 
>                                                                                         ^
> 
> In file included from sketch\src\inc/MarlinConfigPre.h:31:0,
> 
>                  from sketch\src\inc/MarlinConfig.h:26,
> 
>                  from sketch\src\Marlin.h:25,
> 
>                  from sketch\src\Marlin.cpp:31:
> 
> c:\users\boher\appdata\local\temp\arduino_build_558883\sketch\configuration.h:1:1: error: 'b' does not name a type
> 
>  b/**
> 
>  ^
> 
> In file included from sketch\src\gcode/gcode.h:252:0,
> 
>                  from sketch\src\Marlin.cpp:49:
> 
> sketch\src\gcode/parser.h: In static member function 'static float GCodeParser::value_float()':
> 
> sketch\src\gcode/parser.h:189:51: error: 'strtod' was not declared in this scope
> 
[... 63 lines cut ...]
> In file included from c:\users\boher\appdata\local\temp\arduino_build_558883\sketch\src\inc\MarlinConfigPre.h:31:0,
> 
>                  from c:\users\boher\appdata\local\temp\arduino_build_558883\sketch\src\inc\marlinconfig.h:26,
> 
>                  from sketch\src\HAL\HAL_AVR\HAL.cpp:29:
> 
> c:\users\boher\appdata\local\temp\arduino_build_558883\sketch\configuration.h:1:1: error: 'b' does not name a type
> 
>  b/**
> 
>  ^
> 
> In file included from c:\users\boher\appdata\local\temp\arduino_build_558883\sketch\src\inc\MarlinConfigPre.h:31:0,
> 
>                  from c:\users\boher\appdata\local\temp\arduino_build_558883\sketch\src\inc\marlinconfig.h:26,
> 
>                  from sketch\src\HAL\HAL_AVR\HAL_spi_AVR.cpp:40:
> 
> c:\users\boher\appdata\local\temp\arduino_build_558883\sketch\configuration.h:1:1: error: 'b' does not name a type
> 
>  b/**
> 
>  ^
> 
> Error compiling for board Arduino/Genuino Mega or Mega 2560.

AnHardt, 2018-07-05:
```
b/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
```
Why did you add this b???

boelle, 2018-07-05:
hmmm... i can promise i did not add it, not on purpose at least

let me check if i can find that stupid B and remove it

boelle, 2018-07-05:
free laughs on me

closed 2018-07-05 by boelle