MarlinFirmware/Marlin #11207: Unable to compile Marlin 2.0.0 with ULTI_CONTROLLER LCD Option

ReProgram8a9ce45a, 2018-07-06:
### Description
Can't compile with the latest 2.0 bugfix and ULTI_CONTROLLER turned on. Compiler indicates U8glib.h is missing.

### Steps to Reproduce
Enable ULTI_CONTROLLER under LCD controller and attempt to compile

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.
sketch\src\lcd\dogm\u8g_dev_ssd1306_sh1106_128x64_I2C.cpp:72:20: fatal error: U8glib.h: No such file or directory

 #include <U8glib.h>

                    ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

melvinisken, 2018-07-06:
Did you install the U8glib library? That's a must.

ReProgram8a9ce45a, 2018-07-06:
Thanks, I've installed 1.19 now I'm getting this:

Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
Build options changed, rebuilding all
sketch\src\lcd\dogm\u8g_dev_uc1610_dogxl160.c:43:17: error: variable 'u8g_dev_uc1610_dogxl160_init_seq' must be const in order to be put into read-only section by means of '__attribute__((progmem))'

 u8g_pgm_uint8_t u8g_dev_uc1610_dogxl160_init_seq[] = {

                 ^

sketch\src\lcd\dogm\u8g_dev_uc1610_dogxl160.c:70:17: error: variable 'u8g_dev_uc1610_dogxl160_stop_seq' must be const in order to be put into read-only section by means of '__attribute__((progmem))'

 u8g_pgm_uint8_t u8g_dev_uc1610_dogxl160_stop_seq[] = {

                 ^

sketch\src\lcd\dogm\u8g_dev_uc1610_dogxl160.c:81:17: error: variable 'u8g_dev_uc1610_dogxl160_data_start' must be const in order to be put into read-only section by means of '__attribute__((progmem))'

 u8g_pgm_uint8_t u8g_dev_uc1610_dogxl160_data_start[] = {

                 ^

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

Tried it with the older version as well and getting the same results. Any ideas on what I should do next?

ReProgram8a9ce45a, 2018-07-06:
Nevermind, had some extra files in the build directory that were not supposed to be there. It works now. Thanks for all the help:)

closed 2018-07-06 by ReProgram8a9ce45a