MarlinFirmware/Marlin #11208: Filament Runout Sensor has low voltage

jalanjarosz, 2018-07-06:
Marlin 1.1.8 and filament runout sensor on Anet A8, and using stock 4 line LCD 2004

pins_ANET_10.h, I assigned PIN 27 as:
#define FIL_RUNOUT_PIN 27

Configuration.h, I enabled:
#define FILAMENT_RUNOUT_SENSOR
#define NOZZLE_PARK_FEATURE
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
#define FIL_RUNOUT_INVERTING false // set to true to invert the logic of the sensor.
#define ENDSTOPPULLUP_FIL_RUNOUT // Uncomment to use internal pullup for filament runout pins if the sensor is defined.
#define FILAMENT_RUNOUT_SCRIPT "M600"
#endif

Configuration_adv.h, I enabled:
#define ADVANCED_PAUSE_FEATURE

Soldered switch to LCD Panel pins 1 (ground) and 3 and confirmed switch is working with logic sensor using COM and NC terminals.

Also enabled M43 support and tried to test pin with constant results of:
SENDING:M43 P27
PIN: 27 Port: A4 FIL_RUNOUT_PIN Input = 0

I've found that the runout logic is working, however my switch is not triggering properly.

Switch is wired to LCD Pin1 (ground) and Pin3 = FW pin 27

PIN: 27 Port: A4 FIL_RUNOUT_PIN Input = 0

Mainboard LED (near reset button) labeled D1 is dimly lit.

When I put multi-meter in logic mode, one lead on ground and second lead on T52, the D1 LED is brightly lit and my M43 P27 results raises input = 1. I remove filament in switch, and M43 P27 = 0. 

After resetting switch (filament loaded) I tested voltage across P27 and Ground at the switch and I'm getting 1.74v, appears to be too low to trigger a "1" state.

How can the pullup be adjusted so I can get the proper trigger state?

gloomyandy, 2018-07-06:
The pin has an LED wired to it on the board which will be pulling the voltage down even with the pull up enabled. Rather than wire the switch to pull the pin to ground, wire it to pull it to +5v and turn off the pull up option. You may need to invert the detection logic depending on how the switch is wired.

ruggb, 2018-07-13:
you need a schematic of the sensor. There was another post here re endstops. They are not all the same circuit wise. I advised him to enable pullups in Marlin since his output was in series with an LED which made the input to the micro about 1/2 what it should be and more susceptible to noise. My endstops do not need a pullup as there is one on board. It also has a capacitor to filter noise which his did not. 
Lacking a schematic, for your device, you can try that.