MarlinFirmware/Marlin #11210: NO_VOLUMETRICS causes compile failure

paulr4, 2018-07-06:
### Description

Uncommenting the NO_VOLUMETRICS definition in Configuration_adv.h causes the following error:

ultralcd.cpp:3638: error: 'class Planner' has no member named 'filament_size'

GMagician, 2018-07-06:
Compiled 1.1, 1.1.x bugfix and 2.0.x bugfix...with no issues
Can you say version you are using and post your configuration files?

paulr4, 2018-07-07:
Yes of course, here they are.
It's the latest version, 1.1.8 according to Github, although the config file says CONFIGURATION_H_VERSION = 010107.
[ConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/2172400/ConfigFiles.zip)

thinkyhead, 2018-07-07:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

paulr4, 2018-07-07:
I just tested bugfix-1.1.x, it fails in the same way.

thinkyhead, 2018-07-07:
Unable to reproduce. The code in `ultralcd.cpp` correctly omits compilation of the volumetric-oriented menu items when the option is enabled.

I even updated and used your configs, and compilation was 100% successful. You must have compiled 1.1.8 again by mistake.

closed 2018-07-07 by thinkyhead

paulr4, 2018-07-07:
Strange, I think you must be right. Apologies for wasting your time.